pbatard/rufus #2119: Reader not recognized

kolbai, 2022-12-21:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus refuses to work with my onboard card reader. While it says that device is not functioning, I can write and delete files, format SD cards. I have tried multiple cards.

Log
---
```
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Enterprise LTSB, 64-bit (Build 17763.107)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found card reader device 'SDHC Card'
Could not open \\.\PhysicalDrive2: [0x0000001F] A device attached to the system is not functioning.
Device eliminated because it appears to contain no media
0 devices found
```

pbatard, 2022-12-22:
Please press <kbd>Alt</kbd>-<kbd>.</kbd> in Rufus to enable enumeration debugging.

Then, with an SD card in the drive, send me your full log.

I have to point out however that `[0x0000001F] A device attached to the system is not functioning` is an error that Windows returns when Rufus is asking Windows to open the disk, so, regardless of what other applications report, there isn't much I can do in Rufus if, when using the __native__ Windows APIs, Windows returns an error when performing basic operations.

Especially, because writing files works at the __volume__ level whereas Rufus needs to work as the __disk__ level, it is completely possible for volume operations to work and disk operations to fail.

For instance, you may have an antivirus or security solution that only allows whilelisted applications (such as Windows' disk manager) to access media at the disk level, in which case you will get an error like the on your report, and, obviously, Rufus cannot work around gaining access to a device if the system or a security solution explicitly denies access. Or, in case your SD card uses a custom driver, it may even be a problem with the driver itself.

kolbai, 2022-12-23:
I have restarted my laptop and now Rufus recognizes this card reader.

[image]

> there isn't much I can do in Rufus if, when using the native Windows APIs, Windows returns an error when performing basic operations.

Maybe it would be possible to run some [additional tests](https://learn.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_volume_is_clustered)?

> Especially, because writing files works at the volume level whereas Rufus needs to work as the disk level, it is completely possible for volume operations to work and disk operations to fail.

I am not sure if that is relevant, but when Rufus was not able to detect this card reader, ATTO Disk Benchmark could do its tests just fine and it uses Direct I/O. 

> Then, with an SD card in the drive, send me your full log.
```
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Enterprise LTSB, 64-bit (Build 17763.107)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Processing Hub 1:
  Hub[1] = '\\?\usb#root_hub30#4&1b12b7e5&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[154]: USB\VID_8087&PID_0A2B\5&18054507&0&6
Processing 'PCISTOR' device:
  Hardware ID: 'RSPCIESTOR\GenDisk'
Found card reader device 'SDHC Card'
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskNVMe____Samsung_SSD_970_1B2Q'
Found non-USB non-removable device 'Samsung SSD 970 PRO 512GB' => Eliminated
1 device found
Windows volumes from this device:
● \\?\Volume{e9ec7d77-8b76-11ea-a75a-a44cc813e8e3}\ @4194304
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No
Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No
Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No
Found card reader device 'SDHC Card'
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3880, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 29.7 GB (31910789120 bytes)
  Start Sector: 8192, Boot: No
```

Here is what AIDA finds about this card reader:
```
[ Memory technology devices / Realtek PCIE CardReader ]

Device Properties:
  Driver Description                        Realtek PCIE CardReader
  Driver Date                               2017-10-19
  Driver Version                            10.0.16299.21304
  Driver Provider                           Realtek Semiconductor Corp.
  INF File                                  oem24.inf
  INF Section                               RTS525A.Inst.NTamd64
  Hardware ID                               PCI\VEN_10EC&DEV_525A&SUBSYS_07B01028&REV_01
  Location Information                      PCI bus 3, device 0, function 0
  PCI Device                                Realtek PCI-E Card Reader

[ Realtek PCI-E Card Reader ]

Device Properties:
  Device Description                        Realtek PCI-E Card Reader
  Bus Type                                  PCI Express 2.0 x1
  Bus / Device / Function                   3 / 0 / 0
  Device ID                                 10EC-525A
  Subsystem ID                              1028-07B0
  Device Class                              FF00
  Revision                                  01
  Fast Back-to-Back Transactions            Not Supported
```

pbatard, 2022-12-23:
> I have restarted my laptop and now Rufus recognizes this card reader.

Then it's clearly an error that is external to Rufus.

> Maybe it would be possible to run some [additional tests](https://learn.microsoft.com/en-us/windows/win32/api/winioctl/ni-winioctl-ioctl_volume_is_clustered)?

This won't change the outcome.

Again, if a __native Windows API__ reports an error when Rufus queries it, and, in this case, that's a simple call to [`CreateFile()`](https://learn.microsoft.com/en-us/windows/win32/api/fileapi/nf-fileapi-createfilea) with the name of the disk device (`\\.\PhysicalDrive2`) and static parameters that, if they were incorrect first wouldn't have Windows return this specific error code, and second, would fail consistently even if you reboot your PC, then it means that the problem is external to Rufus.

I can confidently say that any software that issues the same call to open `\\.\PhysicalDrive2` will fail in the same manner, because it's really Windows itself that is reporting an issue ("Hey, you asked me to open `\\.\PhysicalDrive2`, but I cannot access it internally myself, so I am reporting an error").

> ATTO Disk Benchmark could do its tests just fine and it uses Direct I/O.

Then I am pretty sure that ATTO Disk Benchmark did not try to open `\\.\PhysicalDrive2` using `CreateFile()`, because it would have failed with the same error then.

The cost/benefit of sinking more time into what is clearly a pure Windows issue (Windows internally is unable to access the device) as well as a transient issue (since it doesn't persist between reboot) is just not worth it. As such, I will close this issue.

closed 2022-12-23 by pbatard

kolbai, 2023-01-18:
This issue keeps repeating. So, I have checked how Raspberry Pi Imager interacts with my SD card reader. Apparently, it uses PNPDeviceID instead of DeviceID when calling CreateFile function.
Rufus:
`"\\.\PhysicalDrive2"`
Raspberry Pi Imager:
`"\\?\pcistor#disk&ven_rsper&prod_rts5208lun0&rev_1.00#0000#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}"`

I have tried intercepting CreateFile calls from Rufus and manually correcting lpFileName. That allowed Rufus to recognize my SD card reader.

pbatard, 2023-01-18:
Then it looks like the driver from your card reader is buggy and intermittently fails to tell Windows that it should resolve `\\.\PhysicalDrive2` to `\\?\pcistor#disk&ven_rsper&prod_rts5208lun0&rev_1.00#0000#{53f56307-b6bf-11d0-94f2-00a0c91efb8b}`.

Basically if a driver answers the `IOCTL_VOLUME_GET_VOLUME_DISK_EXTENTS` or `IOCTL_STORAGE_GET_DEVICE_NUMBER` and returns a drive number, then the same driver is expected to properly tell Windows how it should set up its disk aliases so that `\\.\PhysicalDrive#` (where `#` is the number returned by the above calls) will match the physical disk that underlines the storage media.

Because this is Windows, there are of course multiple aliases to a physical volume, but, if a driver does its job properly, then all of them should work equally, and if one doesn't, or only works occasionally, then it means that the driver has a bug.

Especially, you are the first person in 10 years of using `\\.\PhysicalDrive#` to access disk drives in Rufus to report that the disk alias we use may not work compared to a different disk alias, and it is clear that the Windows native mass storage drivers developed by Microsoft have no such issue, so I'm afraid that, if your expectation is that we are going to change the model in which we access storage media on account that the Realtek RTS5208 SD card driver is buggy, that's not going to happen...

Instead, I would suggest that you open an issue with Realtek to report the problem and ask them to fix their driver.

kolbai, 2023-01-18:
The latest driver for this reader is [10.0.22621.21360](https://catalog.s.download.windowsupdate.com/c/msdownload/update/driver/drvs/2022/12/c19ca26f-1aba-484f-af65-30ed8e953bdd_140eac713c136a1df28fcf5f06f221ae82458c64.cab) (2022-10-13) and it does not fix this issue.

I don’t want to restart Windows each time I need to burn an image file to SD card. I have tried many popular alternatives to Rufus, but they all failed to interact with my SD card reader when it is in a “locked state” (incorrectly reported as not functioning). The reason for this is that they all try to open SD card by calling a CreateFile function and passing `\\.\PhysicalDrive2` as a file name. Raspberry Pi Imager does it differently, but fails later on. I have also found a WinDiskFlash application (I have no affiliation to it), which manages to burn image files while reader is “locked”.

>if your expectation is that we are going to change the model in which we access storage media on account that the Realtek RTS5208 SD card driver is buggy, that's not going to happen...

I don't expect anything. I have reported the issue, I have found its cause and I showed that other similar utilities handle it differently. It is up to you to choose whether your utility will try to get a device handle in alternative way when the main method fails. You have decided not to do that and that's about it.

pbatard, 2023-01-18:
> You have decided not to do that and that's about it.

Please understand that our reasoning for not doing that is that we're not working in isolation here, and if we were to add exceptions for every buggy software that exists outside of Rufus, and requires a workaround, we'd never see the end of it and spend most of our development time handling corner cases that are expected to work but don't, instead of spending it on features and fixes that are going to help a lot more people.

Plus, the issue with workarounds is that, as your report seem to indicate, you may end up with needing every other application that may run into the issue to also add such workaround (what if I add a workaround for your card reader, but then you find software that suits you better than Rufus but does also uses `\\.\PhysicalDrive2` to access the drive?), which is why, for where we stand, and for the sake of all users, we believe the effort is better spent getting Realtek to fix their driver, thereby fixing it for both users or Rufus as well as users of other applications that access the drive in a manner similar to Rufus, than hoping that this or that application may add a fallback for an error that should never occur when accessing a drive.

And I can definitely understand why it is frustrating, because, unlike most Open Source software, Realtek does not let you interact with their developers directly, thus forcing you to invest a lot more effort in, first, getting your problem heard by the relevant people, and second, addressed properly.