pbatard/rufus #2120: ERROR: unable to assign a drive letter

Almaar72, 2022-12-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here> Attached you can find a screenshot. You can see the ERROR message. 
[image]

Log
---
```
<FULL LOG>
```Rufus x86 v3.21.1949
Windows version: Windows 10 Home, 64-bit (Build 19044.2364)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Detected installed Windows Language Pack for 0x0409 (English (English))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1950, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x7A18C285
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 14.9 GB (16041979904 bytes)
  Start Sector: 2408, Boot: No

pbatard, 2023-01-05:
That log is not a full log, so I can't really help you because you've just not provided enough data.

I would suggest to try <kbd>Alt</kbd>-<kbd>Z</kbd> to zero your drive and try again, but, if you want to re-open this issue and actually get help, please provide the full log from Rufus of what happens when you try to create the media, and not just a log from opening Rufus __after__ you created the media and closed Rufus.

closed 2023-01-05 by pbatard

mokmonster, 2023-03-29:
Easiest answer - make sure your USB drive is empty