pbatard/rufus #2121: feature request: add option to Disable BitLocker automatic device encryption

lwlwlwlw, 2022-12-24:
Can you please add option to Disable BitLocker automatic device encryption?
It's anoying that bitlocker encryption is turned on all all drivers by default on my new HP notebook with 12gen cpu.

This can be done by adding registry key as below.

https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-bitlocker

Disable BitLocker automatic device encryption
OEMs can choose to disable device encryption and instead implement their own encryption technology on a device. To disable BitLocker automatic device encryption, you can use an Unattend file and set [PreventDeviceEncryption](https://learn.microsoft.com/en-us/windows-hardware/customize/desktop/unattend/microsoft-windows-securestartup-filterdriver-preventdeviceencryption) to True. Alternately, you can update this registry key: HKEY_LOCAL_MACHINE\SYSTEM\CurrentControlSet\Control\BitLocker Value: PreventDeviceEncryption equal to True (1).

Thank you.

pbatard, 2023-01-03:
Thanks for the suggestion.

I've been pondering whether or not to add the feature you request, as the idea behind the Windows User Experience dialog of Rufus is to remove the annoyances that either prevent Windows from being installed altogether (Secure Boot, TPM, online account) or interrupt the streamlining of the installation (prompt for various annoying tracking options that any privacy-minded users would choose to disable by default). So my issue is that BitLocker does not fall into the normal category of what I designed the Windows User Experience of Rufus to deal with...

However, even if I wouldn't recommend disabling disk encryption when available, I can see some reasoning behind leaving the __choice__ of turning it on or off into the hands of the end-user, so I am planning to add the option in the next version of Rufus.

closed 2023-01-10 by pbatard

TheW0LVERIN3, 2023-01-12:
> Thanks for the suggestion.
>
> I've been pondering whether or not to add the feature you request, as the idea behind the Windows User Experience dialog of Rufus is to remove the annoyances that either [...]

I am glad to see that the feature is coming, the main reason I want to disable it is because the automatic "device encryption" stores the encryption key via tpm and so if I boot windows from a usb on my laptop it erases the key that is stored by the local windows install on my laptop. This makes me have to look up the key using my microsoft account, I have no idea what a user that isn't logged in and didn't back up their key would do. Almost all new pc's especially laptops have this device encryption enabled, it isn't like bitlocker which you have to enable manually, it gets automatically enabled, even in windows home. The only way to prevent it I have found is to boot into boot options and choosing the device instead of the the windows uefi boot option for the device that automatically get created, but this becomes difficult when rebooting and sometimes I miss the window to do it and when I want to go back to my local install I have to type in the encryption key again. 
Also if a windows install on a usb encrypts itself automatically and stores the key in tpm, when you boot the usb on another pc it will ask for the  encryption key. Now users could disable it themselves but that requires all files to be decrypted which is a lot of reading and writing to usb flash storage for no reason. Basically "device encryption" that automatically stores the encryption key in tpm is not good for user experience, portability and the flash storage when having to disable it. I recommend disabling it as a default option.
Also this isn't the same as full fledged bitlocker which doesn't store anything in tpm, etc. While "device encryption uses the bitlocker tech it isn't as secure or as useful as full bitlocker. Though bitlocker is technically not available on windows home, only "device encryption is available on home edition.

TheW0LVERIN3, 2023-03-28:
@pbatard I recently checked out rufus 3.22 and found out that this new option is not there when creating a windows to go drive. 
I believe it would be very helpful for windows to go installations because automatic device encryption is not the same as normal bitlocker and instead automatically encrypts and stores the key in tpm which automatically decrypts it every time you boot up. It's not only less secure than regular bitlocker, it's also tied to the hardware because it will ask for the encryption key if you boot it up on a different machine(different tpm). It never tells you to back up that encryption key so you're fresh outta luck if you didn't log in with a microsoft account(which automatically backs up the key). 

But to me the biggest problem is that it overwrites the key already stored in tpm by the windows installation on the internal drive of whatever system you booted it up on. 
This means you have to reenter the key every time you use the windows to go drive and reboot, or shut down usb and boot back into usb(unless you specifically interrupt the boot and select the usb, not the boot entry corresponding to the windows on the usb but the usb itself)

I already explained this above in a previous comment but considering the issue was closed you probably didn't see that hence the ping.
This feature would personally be a lifesaver for me because I'm tired of putting in my automatic encryption key from some microsoft website every time the usb reboots and I didn't push f2 in time.

bryanray, 2023-03-29:
i had this option.

3.22.2009 portable,

i just updated

TheW0LVERIN3, 2023-03-29:
> i had this option.
> 
> 3.22.2009 portable,
> 
> i just updated

On a windows to go installation?

pineapple63, 2023-03-29:
@TheW0LVERIN3 
For me, the option shows when i create a normal Windows 11 installer USB, but it does NOT show when i create a Windows to Go drive with the same Windows 11 ISO

With automatic device encryption (which seems to be using bitlocker behind the scenes, bitlocker will save the keys to the TPM by default if a TPM is present, i know this because i once manually turned on bitlocker on my desktop, and it didn’t require me to set a password or anything, it was saving the key to the TPM), I discovered that if you do NOT link a Microsoft account, the encryption will require “acrivation” (by linking a Microsoft account, or on the Pro version, you can open Bitlocker settings (not device encryption), and activate the encryption after choosing where to save the recovery key (it gives 3 options, save to a Microsoft Account, save to the computer (or a USB or network drive), or print it, and you can choose more than one)
The above is how i found out it really is using bitlocker behind the scenes

So i guess that for those that don’t link a Microsoft account, it temporary saves the key onto the drive itself (until the encryption is either turned off or activated)

pineapple63, 2023-03-30:
@TheW0LVERIN3 
I have a feeling that the way Rufus turns off the automatic encryption is by putting an entry in the `unattend.xml` file that is created when selecting the customisation options (I haven’t had chance to confirm this myself, but i suspect it’s done like that so that the Windows store version can also have this option)
And I'm not 100% sure about this, but i think that entry will only be checked for when doing a regular install

Although there is potentially a way to disable it for creating Windows to go drives, one issue is that due to Microsoft being restrictive about what windows store apps can and cannot do, the option would only be available for the non windows store version (assuming that what i am thinking of IS even possible)

Unfortunately I don’t have a way to check for myself, but i wonder how far back this issue goes (if Windows 10 version 1909 does not have this issue then i think i can guess what the root cause of the issue is (it may be that newer Windows 10 versions may have had some code removed that deals with Windows to Go installations, which causes automatic encryption to enable, i can only guess as to the cause of the issue))

JonnyTech, 2023-03-30:
> I have a feeling that the way Rufus turns off the automatic encryption is by putting an entry in the `unattend.xml` file

It is indeed, as per this commit: https://github.com/pbatard/rufus/commit/1ce1f47a8bba6e4d2a39d4a6fb49817fa869d855

```fprintf(fd, "      <PreventDeviceEncryption>true</PreventDeviceEncryption>\n");```

It has been available by Microsoft since at least  Windows 10 v1709.

pbatard, 2023-03-30:
> I have a feeling that the way Rufus turns off the automatic encryption (...)

The nice thing about this project being Open Source is that you don't have to have a feeling, but you can [find that out for yourself](https://github.com/pbatard/rufus/blob/master/src/wue.c#L202-L209) (provided you can interpret the code, but it's not really rocket science here).

> And I'm not 100% sure about this, but i think that entry will only be checked for when doing a regular install

Not really. Unattend works fine whether you create a Windows To Go drive or a regular install drive, and we use it for both cases.

The issue here is that my understanding was that this setting was not useful for Windows To Go, since we always apply the image to an unencrypted drive, so, whereas [we toggle the display of that setting for regular install](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L1576-L1597), we [don't do it for Windows To Go](https://github.com/pbatard/rufus/blob/master/src/rufus.c#L1576-L1597).

So, technically, BitLocker is always disabled for Windows To Go, and I'm still puzzled as to whether what @TheW0LVERIN3 wants will actually work, and, more importantly, how actually beneficial it will be for users. Please bear in mind that I do **NOT** design Rufus for specific use cases, and my take of this whole thing is that, if you really want to sort our your BitLocker woes with Windows To Go, it's up to **you** to do it, because I think that this is an uncommon enough issue to have to want an application to take care of it (and present the option to all WTG users who will then be under the impression that, if the option is not checked, then it must mean that the drive created by Rufus should be BitLocker encrypted, which will create confusion).

So, my take on this is:
- If you want to fiddle with BitLocker and Windows To Go, just edit the unattend.xml once Rufus is done creating the drive and add the `PreventDeviceEncryption` setting yourself. Especially, this should help you validate that what you are asking the application to do actually works as you expect it to, which I have seen no evidence of so far.

You may also want to read [this FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want) with the understanding that what you are requesting is really a Power-User/Syadmin feature and one that seems a bit to specific to a single individual's situation to be worth adding to the application.

Now, if I do get reports from multiple other users about this issue, along with confirmation that adding  `PreventDeviceEncryption` does actually solves the issue, I may reconsider, but please understand that I prefer concentrating on features and fixes that will be useful for a majority of users, and this doesn't look like one of them.

commit 1ce1f47a (2023-01-10): [wue] add option to Disable BitLocker automatic device encryption

* See https://learn.microsoft.com/en-us/windows-hardware/design/device-experiences/oem-bitlocker#disable-bitlocker-automatic-device-encryption
* Closes #2121