pbatard/rufus #2122: When trying to install Rufus, it is registering as open within task manager under details but nothing is showing on the screen.

brewdog90gaming, 2022-12-26:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2022-12-26:
#2096, #1860, #1597 and others.

You are probably using Sophos Antivirus or some other __poorly designed__ security solution, such as the ones listed [here](https://github.com/pbatard/rufus/wiki/FAQ#incompatible-software) (non-exhaustive list).

brewdog90gaming, 2022-12-26:
> #2096, #1860, #1597 and others.
> 
> You are probably using Sophos Antivirus or some other **poorly designed** security solution, such as the ones listed [here](https://github.com/pbatard/rufus/wiki/FAQ#incompatible-software) (non-exhaustive list).

No Antivirus is running

brewdog90gaming, 2022-12-26:
is there a way I can force the process to happen via command prompt? like listed with linux?

pbatard, 2022-12-26:
Doesn't have to be an antivirus.

At any rate, you may want to follow the advice I provided in these other issues, such as [this one](https://github.com/pbatard/rufus/issues/2096#issuecomment-1329496486) and post the output from DebugView.

Also, what happens if you make sure that you have no drives connected?

brewdog90gaming, 2022-12-26:
[DESKTOP-07T35TK.LOG](https://github.com/pbatard/rufus/files/10304649/DESKTOP-07T35TK.LOG)
here is the debug when trying to open rufus. What I get when trying to open when no other devices are connected is the same. shows in task manager that its running but nothing on the screen.

pbatard, 2022-12-26:
Since we're past init, this looks like an issue during enumeration. You probably have a device that is very slow to enumerate. I would advise you to wait a while (e.g. more than 1 hour) to see what happens.

If Rufus still doesn't open, then the only option I see, since you are the only person outside of people using Sophos and other problematic software to report such an issue (meaning I really have no precedent to identify what could be causing your freezout), will be to recompile and debug Rufus yourself to see exactly where it appears to freeze in Visuall Studio 2022:

1. If needed, install Visual Studio 2022 (the Community Edition is completely [free to install and run](https://visualstudio.microsoft.com/vs/community/)). Make sure you install the C/C++ development tools.
2. Launch Visual Studio and select _Clone a repository_
3. Use `https://github.com/pbatard/rufus.git` as the _Repository Location_ and any directory you like for _Path_
4. Once the project is cloned, make sure the 2 dropdowns at the top are set to _Debug_ and _x86_ and press _Local Windows Debugger_ on the same line
5. Pause the execution when Rufus appears to get stuck (`0` in the screenshot below).
6. Open the the "Call stack" tab at the bottom (`1` in the screenshot below).
7. In the call stack, click on the highest line (that still pertains to the application) and it will tell exactly where in the code Rufus is stuck (`2` in the screenshot below).

[image]

pbatard, 2023-01-10:
Any update on this?

I'm afraid that since I can't replicate this in any environment, and it doesn't look like many people are reporting this issue, the only way to move forward with this issue is if you or someone experiencing the same problem try to debug it, as described above, on the machine where they experience it.

brewdog90gaming, 2023-01-11:
Hi,

Waiting a while seems to have worked

pbatard, 2023-01-11:
OK, I will close this issue then, as there's really nothing more I can do unless someone who experiences the problem can go through the steps I highlighted above.

closed 2023-01-11 by pbatard