pbatard/rufus #2123: Error accessing USB when using ver 3.21. Version 3.20 is OK

cinlung, 2022-12-28:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I just downloaded Rufus 3.21 using the version checker from Rufus 3.20. After downloaded, I tried to flash Win 10 21H2 to my USB drive and it said at the beginning that USB Drive was not accessible. After checking in Disk Management, it seemed that Rufus managed to format the drive then the process made an error on writing process.

I made sure the drive was OK using HP USB Formatter and even windows formatter and tried to write some data in it. Then I reverted back to Rufus Ver 3.20 and it works ok.

Log
---
```
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22621.963)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'ST500DM002-1ER14C' => Eliminated
Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
Found non-USB removable device 'ST1000DM014-2FR10D' => Eliminated
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 943, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {D7402B3F-7018-427B-B704-7DA4DFFD3D67}
Max parts: 128, Start Offset: 17408, Usable = 7759428096 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {683C835B-F571-434C-943D-F2739C930F4F}
  Size: 7.2 GB (7758396928 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD-ISO'
  Detected: Windows 10 ISO (Build 19041.1288)
  Size: 4.3 GB (Projected)
  Note: File on disk is larger than reported ISO size by 570 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows-10Nov2021-21H2.iso (4.3 GB)
Could not open temporary output file 'D:\My Files\Installer\Utilities\USB FD Utilties\Rufus\rufus.ini~'

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive4 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Write error [0x00000005] Access is denied.
Retrying in 5 seconds...
Write error [0x00000005] Access is denied.
Could not reset partitions
Re-mounted volume as G: after error

Found non-USB removable device 'ST500DM002-1ER14C' => Eliminated
Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
Found non-USB removable device 'ST1000DM014-2FR10D' => Eliminated
1 device found
No volume information for drive 0x84
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 943, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {D7402B3F-7018-427B-B704-7DA4DFFD3D67}
Max parts: 128, Start Offset: 17408, Usable = 7759428096 bytes
Found non-USB removable device 'ST500DM002-1ER14C' => Eliminated
Found USB 2.0 device 'TOSHIBA TransMemory USB Device' (0930:6544)
Found non-USB removable device 'ST1000DM014-2FR10D' => Eliminated
1 device found
No volume information for drive 0x84
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 943, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {D7402B3F-7018-427B-B704-7DA4DFFD3D67}
Max parts: 128, Start Offset: 17408, Usable = 7759428096 bytes

```

pbatard, 2022-12-28:
https://github.com/pbatard/rufus/wiki/FAQ#error-0x00000005-access-is-denied

Note that these issues may be __transcient__ meaning that, unless you have tried repeatedly with both 3.20 and 3.21, you can't actually conclude that 3.21 has an issue that 3.20 hasn't (especially as I can vouch not to have changed any code that could result in such an issue).

It is also very possible that you have a security solution that may whitelist applications for access to USB drives, and consider that Rufus 3.20 and Rufus 3.21 are different applications (since the executable will have a different name).

So, if Rufus 3.20 is whitelisted but not Rufus 3.21, you would get the kind of systematic access denied errors you see (but they have to be systematic - if they only happen once in a while, it's something else).

At any rate, as explained in the FAQ, access denied is not something that can be fixed in Rufus, because it's an external application or Windows actively preventing access to the drive (or it could be a hardware error as well), so there's really nothing I can do, even if, from your perspective, it _seems to look_ like a regression.

cinlung, 2022-12-29:
Hi Pete

 

First, thanks for making great program like Rufus. As for the security that whitelist the app, you might be correct, I checked windows defender log, it seems it has prevented rufus 3.21 to access the USB.

I will try later.

Thank you

 

https://github.com/pbatard/rufus/wiki/FAQ#error-0x00000005-access-is-denied

pbatard, 2022-12-29:
OK. I will close this issue then.

closed 2022-12-29 by pbatard