pbatard/rufus #2124: Feature request: verify checksums embedded into the .iso image (isomd5sum)

gvegidy, 2022-12-28:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
The tool [isomd5sum](https://github.com/rhinstaller/isomd5sum) allows to embed a MD5 checksum into the iso file itself. Using this technique allows to automatically detect corrupted images, for example due to download or mirror issues. Since no separate files or cross checking against websites is required of the user, this check can run fully automatic without user interaction and just warns the user if the checksum doesn't match.

These checksums are only meant to detect accidentally corrupted images, not malicious tampering with the image. So MD5 is not chosen for it's (broken) cryptographic properties, but just as an enhanced CRC.

isomd5sum was developed more than 10 years ago by Redhat and is used in all Fedora, Redhat Enterprise Linux and all it's derivatives like Rocky and Alma. 

I'm one of the developers of [System Rescue](https://www.system-rescue.org/) and we recently added implanted checksums to our .iso-Images. This is in addition to regular checksum files and PGP signatures. We recommend using Rufus to create bootable USB devices for users running Windows.

It would be nice if a future version of Rufus could check if such a checksum is embedded into the .iso and then do a check automatically. The user should just be notified of the issue if there is a checksum available and it doesn't match. So a opportunistic verification.

I know of the checksum feature of Rufus. But it has the downside that the user has to manually compare the checksums, with the result that many users don't do this. When we compare the download statistics of the SystemRescue .iso images and the separate checksum files, we see that not many users do check against the separate checksum files or PGP signatures. The embedded checksums and automatic checking of them would at least add protection against accidental corruption for users not doing the recommended proper checks manually.

`isomd5sum` is written in C and released under `GPL-2.0-or-later`. It doesn't need any external libraries. So maybe parts of the source could be adapted to Windows and then embedded into Rufus.

Log
---
I changed a .iso image so the embedded checksum doesn't match anymore:
```
$ checkisomd5 broken.iso 
Press [Esc] to abort check.

The media check is complete, the result is: FAIL.

It is not recommended to use this media.
```
This broken image is then written to USB media without complaints by Rufus:
[rufus.log](https://github.com/pbatard/rufus/files/10315734/rufus.log)

JonnyTech, 2022-12-28:
I use checksums (from confirmed secure locations) to not only verify the integrity of downloaded files, but also to ensure that they have not been tampered. This is offtopic for Rufus, but how can one confirm the latter? Someone can modify the iso and checksum then share it as genuine. Or is this use case outside the scope of `isomd5sum`?

gvegidy, 2022-12-28:
> I use checksums (from confirmed secure locations) to not only verify the integrity of downloaded files, but also to ensure that they have not been tampered. This is offtopic for Rufus, but how can one confirm the latter? Someone can modify the iso and checksum then share it as genuine. 

Using isomd5sum you can't verify that a file is genuine and hasn't been intentionally tampered with by a bad actor. isomd5sum is just to protect against accidental corruption, for example during download or mirroring.

> Or is this use case outside the scope of `isomd5sum`?

Exactly.

pbatard, 2022-12-28:
> allows to embed a MD5 checksum into the iso file itself

That's fine and all, but I am not aware of a de-facto convention of where to look for the sums.
As a matter of fact, I just downloaded `systemrescue-9.06-amd64.iso`, since you say that you added the MD5's there, expecting to find a `.txt` file at the root of the ISO with the individual sums for each file, as this is what distros (such as Debian or Ubuntu) normally do, but I found no such thing.

As you can expect, __if__ I am to add this feature (which I am not convinced is worth the cost) I am not interested in adding dozens of ways to support this or that method of embedding MD5sums because it looks to me, and I'm afraid this is all too common in the Linux world, that the Red Hat people chose to reinvent the wheel, instead of just base their MD5 validator on what other prominent distros did, __as they should have done if they GENUINELY wanted to help end users__.

Trying to support persistence (with Debian, Ubuntu disagreeing on how this should be implemented, instead of _communicating_ with one another to push towards a common universal __compromise__ that can satisfy the end-users) is already painful enough as it is, and I sure don't want to run into yet another _"You added n methods of supporting MD5 validation, but I'm using a different method, so I will expect you to support n+1"_.

In short, I'll be much more willing to add automated md5sum validation when Linuxfolks make a concerted effort to solve a __common__ problem they have, and agree on a single method, which is really what you and other folks who want to benefit from it should really be pressuring them into. There's absolutely zero reason why Red Hat could not use an `md5sum.txt` at the root of their ISO like other distros do.

Furthermore, I will assert that whoever devised this method didn't really consider the implication of using an ISO-9660 specific way of storing the data, which means that, if you apply file system transposition (i.e. copying the content of the ISO to a different file system), you completely lose the ability to validate the MD5 on the transposed media, whereas the Debian/Ubuntu method, of having the MD5's in the text file does not suffer from what I consider to be a huge drawback.

Thus, whereas I do have some plans to add checksum validation __in the long term__, I am afraid that I would very much like the ill-devised _not-invented-here_ Red Hat method to fail, because there was really zero reason for Red Hat to introduce fragmentation, and fragmentation does not ever benefit the end users. As such, I am not planning to ever support Red Hat's method of only supporting MD5 validation through custom ISO-9660 file system trickery, and I have to dismiss this request.

Now, if you were to ditch the Red Hat way and switch to using an `md5sum.txt` at the root of the media, like Debian, Ubuntu and others do (which would help towards making the `md5sum.txt` way the de-facto one thereby benefiting end users), I guess I would be more inclined to try to support this request, though I have to point out that, even if the feature gets added to Rufus, it is unlikely to be before a couple of years...

closed 2022-12-28 by pbatard

gvegidy, 2022-12-29:
> As you can expect, **if** I am to add this feature (which I am not convinced is worth the cost) I am not interested in adding dozens of ways to support this or that method of embedding MD5sums [...]

I don't think this is "Red Hat people choosing to reinvent the wheel" here:
1. According to the copyright dates, the first release from Redhat of this was in 2001. It looks to me like Debian started their `md5sum.txt` thing around the same time. It first appeared in Debian 2.2. So it looks to me more like two different solutions for a similar problem that were developed at the same time than one rejecting what the others did and doing their own thing.

2. Checksumming the individual files that are contained on the .iso is quite a different thing than to checksum the whole .iso itself. The difference is especially important for bootable CDs, because they have the Eltorio/ISO-Hybrid images embedded that make them bootable. These images don't show up as files and thus aren't checksummed by the `md5sum.txt` solution. Also often things like the volume label are also important to make a image usable, so it should be checksummed too.

> Furthermore, I will assert that whoever devised this method didn't really consider the implication of using an ISO-9660 specific way of storing the data, which means that, if you apply file system [...]

When looking at this from the side of using a ISO-Image, like on optical media, on a Baseboard management controller, on a VM or on USB-media created with DD, I think it makes much more sense to verify the whole ISO-Image including all the boot images and configuration.

Only when you transpose a image to a different filesystem checksumming the individual files becomes an advantage. So I can see why you prefer this method.

> Now, if you were to ditch the Red Hat way and switch to using an `md5sum.txt` at the root of the media, like Debian, Ubuntu and others do (which would help towards making the `md5sum.txt` way the [...]

If Rufus gained support for automatically validating `md5sum.txt`, we'll add a `md5sum.txt` to SystemRescue in addition to the embedded isomd5sum.

pbatard, 2022-12-29:
> I don't think this is "Red Hat people choosing to reinvent the wheel" here:

Well, it's definitely a case of Linuxfolks not trying to work together to find a common solution to a common problems they have.

>  The difference is especially important for bootable CDs, because they have the Eltorio/ISO-Hybrid images embedded that make them bootable. These images don't show up as files.

And Rufus doesn't extract El-Torito content in the first place, meaning that there's little point for use trying to validate that content. As a matter of fact, it would make little sense to have Rufus report that the El-Torito part failed validation if everything else didn't, as it won't impair boot.

> Also often things like the volume label are also important to make a image usable, so it should be checksummed too.

That's another of my big issues with how distro maintainers master their images.

__A volume label should NEVER EVER be an integral part of making an image bootable, because you should not expect an ISO-9660 or UDF volume label to survive file system transposition!__

In the modern world of UEFI, supporting File System Transposition is crucial if you want to help users create bootable media, because it does away with needing to use a third party utility like Rufus or even `dd`, and it can be absolutely __crucial__ if you want Windows users to actually boot media, rather than ditch it altogether because they believe that it was not created properly.

For background info about UEFI File System Transposition and many examples of why not supporting it creates real problems for users, see my recent GRUB patch at https://lists.gnu.org/archive/html/grub-devel/2022-11/msg00121.html.

Here again, you should know that, if the volume label does not fit with the FAT32 specifications (11 uppercase characters max, with many restrictions on allowed chars), which, apart for Arch releases, is almost always the case, Rufus will have to alter it anyway so that boot doesn't break, so having a label that fails validation does not change anything.

In short, the only content that actually matters to Rufus, and that we actually need to validate, is the content that we transpose. Therefore, we really don't need more than an `md5sum.txt`. And considering how unlikely it is that you would not get errors in the transposable content when you have errors elsewhere (and how if one does experience boot issues, the first thing they should do is try to validate the MD5/SHA of the whole ISO, from the sum that's usually provided on the download website, rather than use a sum that's embedded in the media itself), I don't really see Red Hat's solution as a smart one. Plus, if you really are going to consider very localised errors, then you must also consider that case where it's the area where you stored the MD5 that got corrupted (not sure if Red Hat added a checksum for the checksum there... but what if it's that checksum that gets corrupted?).

In my view, the Red Hat solution is/was the wrong approach to the problem and is unlikely to benefit users as much as Red Hat imagines. So I'm not going to promote it in Rufus. Besides, I'd still much rather have Red Hat understand while file system transposition is important... [which they still don't seem to currently do, much to the dismay of their users (and not just the ones that use Rufus to create bootable media](https://github.com/rhinstaller/anaconda/pull/3529#issuecomment-922031795).

> If Rufus gained support for automatically validating `md5sum.txt`, we'll add a `md5sum.txt` to SystemRescue in addition to the embedded isomd5sum.

As I said, this _may_ happen, but not for some time...

gvegidy, 2022-12-29:
> Here again, you should know that, if the volume label does not fit with the FAT32 specifications (11 uppercase characters max, with many restrictions on allowed chars), which, apart for Arch [...]

SystemRescue is based on Arch Linux. It relies on a special filesystem label to find the next stage. But we are well aware of the FAT32 label limitations and thus ensure that the label is valid for FAT32. We use `RESCUEXYZ` with XYZ being the version number as label.

> > If Rufus gained support for automatically validating `md5sum.txt`, we'll add a `md5sum.txt` to SystemRescue in addition to the embedded isomd5sum.
> 
> As I said, this _may_ happen, but not for some time...

Wouldn't it then make sense to change the name of the issue (remove the `isomd5sum`) and then reopen it?

pbatard, 2022-12-29:
> We use `RESCUEXYZ` with XYZ being the version number as label.

That's fine, but not enough for users who just extract the ISO and copy it to a FAT32 volume that could be labelled anything.

For better File System Transposition support, you want to do away with looking for a label, but instead search for a specific file on all available volumes. This is what recent Debian ISOs have moved towards, and also what the `grub-mkrescue` patch I submitted does. Typically, you want to have something unique enough to your media in a `.disk/` directory that will get copied over during transposition, that you can then have the bootloader (e.g. GRUB) search for to identify the volume you are trying to boot. This way, volume labels don't matter.

Note however that, unfortunately, the people who started using `.disk/` failed to take into consideration that if a Linux user is going to use `cp -r * <target>` to duplicate the content, dot directories like `.disk/` may be omitted from the copy... But since I didn't want to reinvent the wheel from what Debian did, I also went with `.disk/` in the GRUB proposal, so that it can eventually become a de-facto standard...

> Wouldn't it then make sense to change the name of the issue (remove the `isomd5sum`) and then reopen it?

I'm not going to have a _deferred_ issue left open that I know I'm unlikely to add for at least a year or two... if at all. I already have [enough of those](https://github.com/pbatard/rufus/issues?q=is%3Aissue+is%3Aopen+label%3Adeferred), and checksum validation in one way or another is something that is on my radar, whether there's an issue open for it or not.

gvegidy, 2022-12-29:
> > We use `RESCUEXYZ` with XYZ being the version number as label.
> 
> That's fine, but not enough for users who just extract the ISO and copy it to a FAT32 volume that could be labelled anything.

Just extracting the ISO and copying the files isn't going to yield a fully supported boot media. You can't boot traditional BIOS this way and I don't think it is worthwhile to invest effort to make this a fully supported method, since we already have several of those for each OS ( https://www.system-rescue.org/Installing-SystemRescue-on-a-USB-memory-stick/ ). These all create fully-featured boot media, usable on UEFI and traditional BIOS.

> For better File System Transposition support, you want to do away with looking for a label, but instead search for a specific file on all available volumes.

This would mean you have to mount all available filesystems on the system. This may be ok for regular Linux distros, but it is a bad idea for a rescue use case. There you have to expect defective disks, broken filesystems and so on. Trying to mount a filesystem, even read-only, from a defective disk can stall your boot for several minutes or even result in not booting at all.

> I'm not going to have a _deferred_ issue left open that I know I'm unlikely to add for at least a year or two... if at all. I already have [enough of [...]

Would you consider pull requests for such a feature? Then it could make sense to leave the issue open to make it easy for possible contributors to find this issue with all the discussion about the proper method and so on.

pbatard, 2022-12-29:
> Just extracting the ISO and copying the files isn't going to yield a fully supported boot media. You can't boot traditional BIOS this way

That's why I made sure to mention UEFI. File System Transposition __must__ be a de facto way of creating bootable media for UEFI, because the whole point of UEFI was do away with boot loaders that work at the block level to instead use ones that work at the file system level.

Please read the second part of [my UEFI File System Transposition patch to the GRUB mailing list](https://lists.gnu.org/archive/html/grub-devel/2022-11/msg00121.html) to understand the context and why distro maintainers should concern themselves with UEFI File System Transposition if they want their users, and especially ones who create bootable media on Windows, to actually try to boot their media and not dismiss them wholesale before they even attempt to try to boot it.

> This would mean you have to mount all available filesystems on the system.

No. This means you need to have the bootable volume you are looking for mounted... which you do regardless of whether you use volume lookup or file lookup.

For the GRUB stage, it doesn't change a thing, because all you do is look for a file on volumes __that are already mounted__ and for Linux, it shouldn't change a thing, because again all you'll do is look for a file on volumes __that should already be mounted by the kernel__.

Please understand that the process I described is not _"mount everything and search for a file"_ but _"search for a file in what has __already__ been mounted"_. There's no extra mount commands invoked anywhere.

> Would you consider pull requests for such a feature?

People can create pull requests whenever they want. There's no need to have an associated issue.

pbatard, 2022-12-29:
Just to clarify the above, because I'm pretty sure you're going to want to comment on that, once the volume has been identified by the base bootloader, the same process you used to mount it in Linux can still apply (i.e. you should be able to resolve that boot volume to a volume label or some other id that you can pass to the kernel to mount it, if you need to do so).

Or, yes, you can try to mount everything and see if you can locate the file, which isn't a big deal. I'm trying to locate the source where Debian do it, since I didn't invent this method, but saw that it was already being used to great effect.

pbatard, 2022-12-29:
https://salsa.debian.org/installer-team/cdrom-detect/-/blob/master/debian/cdrom-detect.postinst is how Debian do it. You'll see they are looking for a `.disk/info` file, which isn't super specific but is actually good enough to locate what is expected to be a single boot media (because quite frankly, going the extra mile to to try to resolve the "right" boot media for someone who plugged 2 boot media for the same distro on their computer is a waste of time).

In this case, they do indeed try to mount USB and optical media to locate the file. But it doesn't look to me like they have had many complaints from the usage scenario you describe with broken CD media or media that take a long time to init themselves (which, I would assert, is likely to be an issue later on regardless anyway, so there's not much to gain in trying to avoid it early), and I still maintain that it's the better approach if you want to help your end users in the long run.