pbatard/rufus #2125: Rufus damaged my USB making it a read only device!

Kalinixenko, 2022-12-31:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
 I was trying to create a bootable device of windows 10, I had an iso image downloaded from the windows page. But it didn't work for any reason. Then I am not sure how or when my USB becomes "protected against writing" due tu the use of Rufus, I suspect that it was when Rufus tried to format my USB by second time with another type of file NTFS instead of fat32. But at the end now my device is damaged and my pc can't format or repair it. ><

Log
---
```
<FULL LOG>
```

pbatard, 2022-12-31:
https://github.com/pbatard/rufus/wiki/FAQ#rufus-cant-detectformat-my-drive-and-reports-that-it-is-read-only-or-contains-no-media and https://github.com/pbatard/rufus/wiki/FAQ#help-rufus-damaged-my-flash-drive.

closed 2022-12-31 by pbatard

Kalinixenko, 2023-01-20:
in your FAQ the solution that you told me of select non-bootable in order to eliminate the UEFI partition doesn't work, it only shows the same error message 
[image]
and I tried with fat32 and ex fat
so... it was not easy ._.

and in the other solution when you say software don't damage usb devices, I disagree with you because blocking it against writing is actually a damage, despite that it is not a hardware damage... my usb's are all from Hewlett Packard so they are expensive ones, and for me not be a surprise that as you said, a great huge of money would be involved in a kind of conspiration to "accelerate the economy", it is possible of course!

pbatard, 2023-01-20:
> I disagree with you because blocking it against writing is actually a damage, despite that it is not a hardware damage...

Nowhere in the FAQ do I say that a device that was turned read-only isn't damaged, or that I don't consider it damaged. [On the contrary](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Rufus_cant_detectformat_my_drive_and_reports_that_it_is_readonly_or_contains_no_media).

Please don't construct a straw man.

If your device turned read-only, it was most certainly damaged. But again, I have to stress out that this damage cannot have been caused by software, unless you have been repeatedly writing the same device over and over. It's just natural wear and tear + low life expectancy of flash memory cell + probability.