pbatard/rufus #2126: Make Rufus compatible for Linux

ramvikrams, 2023-01-02:
It's not an issue but an enhancement I wanted if we could make `Rufus` compatible on `Linux`.
I want Rufus on Linux so I would also like to work on this if anyone could guide me

pbatard, 2023-01-02:
I'll counter by _"Make people who think the checklist does not apply to them realise that the checklist very much applies to them, and that, if they had followed the check list, they would have found [this VERY RELEVANT FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#do-you-plan-to-port-rufus-to-linuxmac-ossome-other-os)"_

Can I log an `enhancement` request against some humans? 😄

closed 2023-01-02 by pbatard

ramvikrams, 2023-01-02:
No, I was saying that I could help for making it comptible on Linux

pbatard, 2023-01-02:
I'm not going to work on this. At all. So if you want to try to port Rufus to Linux, you can go ahead, but don't expect this to be a collaborative work, on account that, for the reasons I have listed in the FAQ, I don't believe that the people who want to port Rufus to Linux have a realistic idea of the work it will actually entail. in short, unless you are planning to spend __months__ rewriting system features that are provided by default on Windows, the best you're gonna get is a utility that is missing a lot of what makes Rufus "Rufus", and that, because of this, I will kindly ask you not to call Rufus, so as not to deceive people who will be expecting things like _Windows To Go_ support or Windows 11 hardware requirements removal...

ramvikrams, 2023-01-02:
> I'm not going to work on this. At all. So if you want to try to port Rufus to Linux, you can go ahead, but don't expect this to be a collaborative work, on account that, for the reasons I have listed in the FAQ

Ok but what I was saying that I can go ahead with that but what I thought was getting some advices in between but your second argument was good, but still I feel that should not stop us for making it portable to Linux

pbatard, 2023-01-02:
The project is Open Source. You can create any derivative software you want. Just don't count on much help from me, especially as past experience has shown that people who are eager to port Rufus to Linux or other OSes soon come to realise, when they actually try to get to it, that it's a lot more than they can chew, especially if they are unfamiliar with pure C development and Windows APIs (in short it's not just a UI slapped on top of easily duplicatable OS calls. Instead, it's pretty much the opposite, with the UI being by far the easiest thing to port, because, unlike the rest, you are expected to find equivalent on whatever OS you are using).

In short, I just don't have the time to invest with people (there have been a few in the past) who say that they want to port Rufus to this or that OS, but seem to have failed to grasp the amount of work it'd take them, and therefore are bound to abandon the project altogether. As a matter of fact, the only people that I deem likely to succeed in porting Rufus to another OS are the ones who won't need any external help (or approval) in the first place and can simply sort what they need by reading and understanding the current source. Because if they do just that, I suspect that they will both have the competence required for the task and be tenacious (or foolish) enough to persevere with what they set to accomplish...

ramvikrams, 2023-01-02:
I'll start them, if I make  progress I'll keep you updated via email.

hannescam, 2023-01-11:
any progress on it?