pbatard/rufus #2127: Gentoo installcd copied in ISO mode no longer boots in UEFI mode

benkohler, 2023-01-03:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm part of the Gentoo Linux release team, and we began getting reports that our Minimal InstallCD image is no longer booting in UEFI mode when copied via rufus in ISO image mode.  When booting BIOS/legacy it still works, but in UEFI mode it shows:

```
Welcome to GRUB!

error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
error: unknown filesystem.
Entering rescue mode...
grub rescue>
```

I believe the breaking change is on our end, when our ISO build tool changed how it prepares the ISO.

Previously we had an isolinux boot image for BIOS and a grub boot image (via grub-mkstandalone) for UEFI, and we called mkisofs to generate an ISO with those boot images embedded.  Now we are calling grub-mkrescue to build the ISO, which will use grub for both BIOS & UEFI boot.

You can see our old call here: https://github.com/gentoo/catalyst/blob/catalyst-3.0-stable/targets/support/create-iso.sh#L258
It was roughly: 

```
mkisofs -J -R -l ${mkisofs_zisofs_opts} -V "${clst_iso_volume_id}" -o "${1}" -b isolinux/isolinux.bin -c isolinux/boot.cat -no-emul-boot -boot-load-size 4 -boot-info-table -eltorito-alt-boot -eltorito-platform efi -b gentoo.efimg -no-emul-boot -z "${clst_target_path}"/
```

The new ISO creation command is defined here: https://github.com/gentoo/catalyst/blob/master/targets/support/create-iso.sh#L203 and just did a test build and here is the exact grub-mkrescue call and its output:
```
grub-mkrescue -joliet -iso-level 3 -o /var/tmp/catalyst/builds/default/install-amd64-minimal-latest.iso /var/tmp/catalyst/builds/default/livecd-stage2-amd64-latest
xorriso 1.5.4 : RockRidge filesystem manipulator, libburnia project.

Drive current: -outdev 'stdio:/var/tmp/catalyst/builds/default/install-amd64-minimal-latest.iso'
Media current: stdio file, overwriteable
Media status : is blank
Media summary: 0 sessions, 0 data blocks, 0 data, 2101g free
Added to ISO image: directory '/'='/tmp/grub.PeDzRO'
xorriso : UPDATE :     882 files added in 1 seconds
Added to ISO image: directory '/'='/var/tmp/catalyst/builds/default/livecd-stage2-amd64-latest'
xorriso : UPDATE :     894 files added in 1 seconds
xorriso : NOTE : Copying to System Area: 512 bytes from file '/usr/lib/grub/i386-pc/boot_hybrid.img'
xorriso : UPDATE :  26.14% done
xorriso : UPDATE : Thank you for being patient. Working since 1 seconds.
xorriso : UPDATE : Thank you for being patient. Working since 2 seconds.
ISO image produced: 216413 sectors
Written to medium : 216413 sectors at LBA 0
Writing to 'stdio:/var/tmp/catalyst/builds/default/install-amd64-minimal-latest.iso' completed successfully.
```

[... 1054 lines cut ...]
Extracting: E:\image.squashfs (390.3 MB)
Extracting: E:\livecd (0 bytes)
Extracting: E:\mach_kernel (0 bytes)
Extracting: E:\EFI\BOOT\BOOT.EFI (from '/efi.img', 112 KB)
Extracting: E:\EFI\BOOT\BOOTIA32.EFI (from '/efi.img', 112 KB)
Extracting: E:\EFI\BOOT\BOOTX64.EFI (from '/efi.img', 144 KB)
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 3.0 device ' USB  SanDisk 3.2Gen1 USB Device' (0781:5591)
Using 'autorun.inf' label for drive E: 'ISOIMAGE'
1 device found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7480, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001C76AE
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 57.3 GB (61529391104 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2023-01-03:
Hi Ben,

First of all, let me start that I always appreciate when distro maintainers report breakage, because we really all want the same thing here: Ensure that users can always create bootable media from a distro with a utility they are comfortable with, and therefore resolve any issue that is found during the media creation process.

I certainly want people to continue to be able to use Rufus to install Gentoo too!

> Now we are calling grub-mkrescue to build the ISO

Then you want to apply [this patch series that I sent to the GRUB mailing list](https://lists.gnu.org/archive/html/grub-devel/2022-11/threads.html#00121) (_Add support for EFI file system transposition_), because I believe that this is your issue.

For what is worth, this is something that I have been trying to pre-empt with the GRUB folks ever since I became aware of the problem early last year (and as a matter of fact, the series I am linking to is a re-sent of the same series I submitted [quite a few month earlier](https://lists.gnu.org/archive/html/grub-devel/2022-06/threads.html#00024)), and that took [quite a lot of convincing to get the GRUB folks to consider integrating](https://lists.gnu.org/archive/html/grub-devel/2022-11/msg00007.html).

Hopefully, you can apply this patchset to your source tree and that will solve your issue.

I also have some [notes on how to test this without having to generate a full blow distribution ISO](https://gist.github.com/pbatard/0deddbd71eefc35a3ed0b08e12a9e7e3), that I jolted down for the GRUB maintainers.

If that does help solve your issue, please don't hesitate to spread the word that EFI File System Transposition is a crucial means of creating UEFI bootable media for users (since it can make Rufus or external utilities including `dd` entirely redundant), especially in a world where trying to enforce `dd` with ISOHybrid does leave Windows users stranded (see the many examples in my cover letter for the series), and needs to be considered as a viable alternative that should be tested always by distro maintainers (which apparently you are doing, so props to Gentoo for doing that!).

benkohler, 2023-01-03:
Thanks for the quick followup! I'm going to apply the grub patches and do a test build, I'll let you know the result and if it's success then I'll talk to our grub maintainer about taking up the patches until upstream does.

As I started reading your patch description, it all started to sound very familiar... then I remembered that a few years ago, I added similar logic to our ISO build system for rufus/unetbootin compatibility (forgive me if I got any technical details or terminology wrong):
https://github.com/gentoo/catalyst/commit/8a70e1ec82d1dbe29b8edf69514a5073ecc1ed71

When we moved to using grub-mkrescue, it greatly simplified our ISO generation, BUT we lost that logic to duplicate /boot/efi contents!  So it makes sense that now grub-mkrescue needs to handle that, I'm really surprised it doesn't already.

pbatard, 2023-01-03:
Well, the GRUB patchset should take care of ensuring that `/efi/boot/` is populated with the UEFI bootloaders, so, yes, that is one of the elements that you need to take care of to ensure File System Transposition support.

Basically, and this is what I mention in the patchset, you want your boot process to honour the 3 following requirements:
* Your boot process should support content residing on a FAT file systems (rather than just ISO9660/UDF or some non UEFI mandated file system). Bonus if you also support NTFS, which has become to be enabled by most manufacturer on newer UEFI firmwares (at least for x86_64) to alleviate the issue of >4GB files, such as large squashfs images and such. Not that this means that, besides the GRUB patches, you may still have to sort out your `initrd` or equivalent to properly support said filesystem(s).
* The file system for your image (ISO9660 or UDF) should should include UEFI bootloaders at the location that is mandated by the UEFI specs for disk boot (`/efi/boot/`), regardless of whether (as is the case with `xorriso` generated media) you may be able to "shove" them into a monolithic `efi.img` that'll translate to a separate ESP when using `dd`. This part will be automatically sorted out for you if using the GRUB patchset.
* Your boot process should __not__ be dependent on locating the boot media by volume label, as, in its intended form, UEFI File System Transposition should be achievable to extracting the ISO content to a FAT or NTFS volume created by the user that may have any label they choose... or no label at all. Now, the only thing the GRUB patchset guarantees is that this will be the case for GRUB, but once you have booted the kernel, it is the job of the distro maintainers to make sure that they can locate the media without relying on a label. I have to point out however that, for people using Rufus, this is not an issue, as Rufus tries to sort out label handling for the user (as well as label issues that arise from the FAT limitations of forcing uppercase and no more than 11 characters, whereas ISO9660/UDF allow long labels in mixed case), which is why most distros that do rely on seek-by-label should still work. I would encourage you however to take a note from Debian if needed, who have swicthed to seek-by-content (see [here](https://salsa.debian.org/installer-team/cdrom-detect/-/blob/master/debian/cdrom-detect.postinst) by searching for the presence of a `.disk/info` file (which I have to say isn't a terribly unique way of identifying content, which is part of the reason why I went with a `.disk/<UUID> file in the GRUB patch proposal).

If you can honour the 3 requirements above, then, no matter the OS, users who are trying to create a disk-based boot media from your ISO should be able to do so without using anything but the native tools that are guaranteed to be present on their OS (where `dd` is __not__ even  a guarantee).

benkohler, 2023-01-04:
With your grub patches, I do now see /efi/boot/ populated on the resulting ISO and on the USB drive created by rufus, but it still fails to boot, with the same grub errors.  I do see one difference though, I am now able to "ls (hd0,msdos1)/" from the grub rescue shell and read the files on the usb stick-- so the minimal grub image has fat support now.

And now with a few grub commands, I can get it booting:
```
set root=(hd0,msdos1)
set prefix=/boot/grub
insmod normal
normal
```

This gets me to the expected grub menu where I can select a Gentoo kernel and boot successfully.

Any idea where the missing piece is now?

I have a copy of the new ISO (built with your grub patches) here: https://drive.google.com/file/d/15xIiuqVgIz9puKA-qeV4j_JHsgBDNVBV/view?usp=share_link

pbatard, 2023-01-04:
The `.disk/` directory is missing from your image. So, yeah, you're going to have trouble booting because the search for the bootable device will fail.

I'm not sure why that is the case because, the way I patched `util/grub-mkrescue.c`'s `main()` is that, as soon as you have a EFI source dir (`if (source_dirs[GRUB_INSTALL_PLATFORM_###_EFI])  ...`) you should get a `.disk/YYYY-DD-MM-HH-MM-SS-00.uuid` file created and added to the generated ISO, and it shouldn't matter if you also have BIOS content there since it's a plain `if` and not an `else if`.

Can you please first validate that the patchset has been properly applied, and that you do have the following lines in your `util/grub-mkrescue.c`:
```C
      /* Create a '.disk/<TIMEBASED_UUID>.uuid' file that can be used to locate the boot media. */
      diskdir = grub_util_path_concat (2, iso9660_dir, ".disk");
      grub_install_mkdir_p (diskdir);
      iso_uuid_file = xasprintf ("%s.uuid", iso_uuid);
      diskdir_uuid = grub_util_path_concat (2, diskdir, iso_uuid_file);
      f = grub_util_fopen (diskdir_uuid, "wb");
      fclose (f);
```

Also be very mindful that, unless you ran grub install, you may still invoke your system's default unpatched binaries, which, even if you do invoke the newly built `grub-mkrescue` could also explain why you are experiencing the issue.

I suggest that you follow the steps I described from https://gist.github.com/pbatard/0deddbd71eefc35a3ed0b08e12a9e7e3 to validate that you can generate a minimal super grub disk image that includes the `.disk/` content (NB: It shouldn't matter if you use `--with-platform=efi --target=amd64-pe` for configure as long as GRUB can autodetect an EFI platform). Then, once you have that, maybe add a `printf()` statement in the line I highlighted above and try to generate your ISO, to validate that the code that does setup the `.disk/` directory does get invoked.

>  but from where, if the ISO embedded boot images are gone?

The embedded boot image is not gone. The patchset __duplicates__ the `/efi/boot/` content from the image onto the ISO9660 file system. Ideally, it should be xorriso's  job to do that, or place the binaries in a fashion that avoids duplication (which I guess could be achieved by enforcing a FAT cluster size of 2048 for the image, since that's the also the block size for ISO9660 and with some trickery, it should be possible to have a FAT table as well as an ISO9660 index pointing to the same sectors), but considering the urgency of having a working solution, and how losing a few MB on a ISO that is most likely to be around 500 MB or larger is hardly an issue, duplicating is what we are going with for now.

> I notice that partciular grub image has increased in size. Before it was 257KB, now it is 297KB.

I do mention that in my notes. We're adding extra file system and partition modules, so, yeah, this will come as an extra few KB (and since we are using the same bootloaders on the ISO fs as the one from the image, this will increase the image size). There really was no point trying to devise a solution where the (ISOHybrid-as-dd-disk-boot) image and (File-system-transposed-disk-boot) ISO9660 bootloaders would be tailored for one specific role, with removal of the parts that only applied for the other role. Better to create GRUB bootloaders that fit both roles, and avoid having to indentify weird issues down the line of _"It boots when I use dd, but not when I file transpose"_ that boil down to a different `bootx64.efi` being used.

benkohler, 2023-01-04:
Aha, thanks for catching that.  I had to mangle the patches a bit after copy/pasting from the ML archive and somehow that one chunk of patch 3 was not being applied (not erroring either).  

I manually re-rolled that patch, verified that the "/* Create a '.disk/<TIMEBASED_UUID>.uuid' file.." code block exists now, and have rebuilt grub.  Sure enough, the next ISO created with grub-mkrescue and passed on to rufus, boots perfectly.

I believe we can close this issue out unless you want to use it as reference until the patches (hopefully) land upstream at grub.

pbatard, 2023-01-04:
Happy to hear that you sorted it out. I'm going to close the issue, since people who do a search should be able to locate it, and, considering how far and in between GRUB releases are (which, sadly, forces distro maintainers to perform a lot of unwarranted extra work), I wouldn't actually bet my life on the next GRUB release being sure to happen within the next 12 months...

closed 2023-01-04 by pbatard