pbatard/rufus #2128: Windows installation failure - error regarding parsing of unattend.xml

belaircomputerguy, 2023-01-03:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'm using RUFUS to create an unattended Windows 11 installer.  When asked which custom features I want to enable(default username, remove 4GB ram check, etc), I've selected all of them.  RUFUS creates the bootable media and I begin the installation. The installer completes and the system restarts - all normal so far.  However as Windows is loading, it goes through 'starting services', then 'getting devices ready', then 'getting ready'.  After a while, I'm getting a dialogue box which reads: 

**'Windows could not parse or process the unattend answer file for pass [specialize].  The settings specified in the answer file cannot be applied.  The error was detected while processing settings for component [Microsoft-Windows-Deployment].'**

If I press enter, the machine restarts and I get a dialogue which reads:

**'The computer restarted unexpectedly or encountered an unexpected error.  Windows installation cannot proceed.'**

I'm unsure of where I might find logging for this.  I found no readable logs in the Logging folder on my bootable media.

pbatard, 2023-01-03:
> I'm unsure of where I might find logging for this

Please re-read the explicit item from the checklist which you checked, and that says:

>  I clicked the 'Log' button or pressed Ctrl-L in Rufus, and copy/pasted the log into the line that says <FULL LOG> below.

There __is__ a log button in Rufus (left of `START`) or you can press kbd>Ctrl</kbd>-<kbd>L</kbd>. That's how you access the log from Rufus.

I will need to see your full log, so please recreate the media and provide the full log (after testing that you still get the same issue), as I cannot help on this issue unless I see your log.

belaircomputerguy, 2023-01-04:
oh man, super sorry about that.  I don't know why I wasn't thinking of this.  For some reason I was thinking the rufus log would be useless to you as it would be on the machine I used it on to create the media and the trouble I'm having is on the target machine.  What an idiot.  I'll get it posted up in the AM.

belaircomputerguy, 2023-01-04:
OK, well, that's weird - it worked that time.  I re-flashed that installer 3 times.  First with all options of customization checked, the second with only the bypass network req and bypass tpm check, neither of which worked.  Then after I posted this issue here, I flashed the third time with no customization and unsurprisingly, it installed.  Since that machine was busy installing, I re-flashed with all options(to generate the log) and ran the installer on another machine and it worked as intended.  Wonder if it was the machine and not the installer.  That machine which was having issues is a junky HP with Windows 11 home in S mode, whereas the second machine(successful) is a Dell which did NOT have s mode originally.  So bizarre.  Thanks for your patience, we can close this one.

closed 2023-01-04 by belaircomputerguy

pwmsdn, 2023-01-17:
I get this same issue on a dell laptop running in home S mode. Must be something specific to computers originally shipped and still set to S mode.

pbatard, 2023-01-17:
> Must be something specific to computers originally shipped and still set to S mode.

S mode does restricts and remove a lot of stuff, so that wouldn't be surprising. Still I would expect that, if Microsoft did a proper job, it'd just ignore options that don't apply to S instead of aborting the install with an error, especially as one may want to use a single `unattend.xml` for multiple versions of Windows.

I don't expect that there's an `if Windows.mode = S` condition that one can insert into `unattend.xml`, so, unless there exists a flag to ignore `unattend.xml` errors (which I haven't found yet) I guess the only option I'll have for S users, since it's impossible for Rufus to know the kind of mode Windows will be installed in in advance (because it depends both on the platform and/or on the choice selected by the user during install), is to just recreate the boot media and remove options until it works...