pbatard/rufus #2130: "Error: could not retrieve architectures from server" when using the ISO download feature

pbatard, 2023-01-06:
As happened in the past (pbatard/Fido/issues/41), it appears that Microsoft have once again altered their ISO download service to make it more difficult for third party applications to use them.

As a result, you will be getting the following error:
[image]

We are currently investigating it.
See also pbatard/Fido/issues/52.

pbatard, 2023-01-06:
How nice of Microsoft of adding this __new__ data in their error message (which is basically what we now get instead of the expected ISO download links):

> Some users, entities and locations are banned from using this service. For this reason, leveraging anonymous or location hiding technologies when connecting to this service is not generally allowed.

This sure wasn't part of the bland error message they returned previously.
Why do I get the feeling that the _entities_ Microsoft mentions explicitly refers to Rufus/Fido?...

Evernow, 2023-01-07:
> Just a note that if you need to download the latest Windows 11 ISO, you can do so by going to https://www.microsoft.com/software-download/ and then clicking Windows 11.
> 
> You will however only be able to download the latest Windows 11 ISO or the latest Windows 10 ISO (but for Windows 10, you will have to change your `User-Agent` if you are accessing the website from Windows).

FYI got the message while on Microsoft's site aswell using Firefox and Edge: 
[image]

pbatard, 2023-01-07:
@Evernow, this may be because you are in a country that is embargoed by Microsoft.

I have no issue downloading ISOs through Firefox or Edge, and Microsoft are using the same error message for _"banned entities"_ (which seems to include Fido) and _"banned locations"_ (which would include Russia or any IP that Microsoft classified as part of a banned location).

In short, if you do see the error even with the browser, then your issue is a different one even if the error message is the same (and there is nothing Fido will be able to do for you, because no amount of scripting can fake an IP address).

Evernow, 2023-01-07:
> 

> @Evernow, this may be because you are in a country that is embargoed by Microsoft.
>
> I have no issue downloading ISOs through Firefox or Edge, and Microsoft are using the same error message for [...]

That's interesting, because I'm in Boston, MA, USA. But yes it does seem dependent on IP address. One given to me by Comcast (ISP) gives the error, using my phone (T-Mobile provided IP) seems to work fine.

pbatard, 2023-01-07:
Oh yeah, another thing I ran into is that if you poke their download servers a bit too much, Microsoft may deem your IP _persona non grata_ for a while (usually one day or two), and I would suspect they use the same message for banned users...

pbatard, 2023-01-08:
This should now be fixed.

closed 2023-01-08 by pbatard

hearnia2k, 2023-01-09:
I just tried to download an iso with Rufus, and it worked, but not initially, and wanted to just put something in here in case others bump into it.  It seems that now the script tries to access "vlscppe.microsoft.com", which was not used before.  This happened to be blocked by my PiHole, and Rufus showed no error in the GUI, just never moved to the language selection prompt, and instead closed the dwnload selection window.

Perhaps an enhancement could be made in Rufus for when a DNS resolution occurs?

From my log, a few minutes ago:
```
Downloading https://github.com/pbatard/Fido/releases/download/v1.39/Fido.ps1.lzma
File length: 18.1 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Exception calling "ShowDialog" with "0" argument(s): "The remote name could not be resolved: 'vlscppe.microsoft.com'"
At C:\Users\hearn\AppData\Local\Temp\{F28562B7-B230-477C-AF04-095F9ADBD7CD}.ps1:1198 char:1
+ $XMLForm.ShowDialog() | Out-Null
+ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : CmdletInvocationException
Exited download script with code: 1
```

pbatard, 2023-01-09:
Agreed that exceptions for the query of vlscppe.microsoft.com should be better handled so I have added just that in the latest release. If you use Rufus, it will make its way to the application once I update the link to use version 1.40.