pbatard/rufus #2133: Download option works under Windows 10, but not Windows 8.1

c627627, 2023-01-13:
I know Windows 8.1 is about to reach End of Life in ten days, but it hasn't yet ;)

I just wanted to let you know that on my multi-boot PC, the option to DOWNLOAD older Windows 11 ISO through Rufus works under Windows 10, but does not work under Windows 8.1 if you wanted to test it...

pbatard, 2023-01-13:
Please provide a log. It is not optional. And __YES__ it very much applies if you want me to consider looking into this issue.

By choosing to ignoring the check list, you are pretty much guaranteeing that I am not going to look into it.

c627627, 2023-01-14:
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 8.1 Pro with Media Center, 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.1 device 'Micron Crucial X8 SSD USB Device' (0634:5600)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 3.0 device 'USB DISK 3.0 USB Device' (13FE:6700)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3773, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xD4988282
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 28.9 GB (31037915136 bytes)
  Start Sector: 8064, Boot: No
Downloading https://github.com/pbatard/Fido/releases/download/v1.40/Fido.ps1.lzma
File length: 18.2 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ?
Script signature is valid ?
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=58f79a0d-3228-457f-9882-73b7e556d833
Error: The underlying connection was closed: An unexpected error occurred on a receive.
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=58f79a0d-3228-457f-9882-73b7e556d833
Error: The underlying connection was closed: An unexpected error occurred on a receive.
Exited download script with code: 3
[image]

c627627, 2023-01-14:
I have a multi boot. When I reboot into Windows 10, and try the same, there is no error.
So it is Windows 8.1 related.

pbatard, 2023-01-14:
Are you sure your Windows 8.1 machine doesn't have a firewall that filters https://vlscppe.microsoft.com?

What happens if you try to navigate to https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=58f79a0d-3228-457f-9882-73b7e556d833 with a browser on Windows 8.1?

c627627, 2023-01-14:
You seem to be right, as I get a blank page for that link on Windows 8.1.

pbatard, 2023-01-14:
No, the page is expected to be blank, so as long as you don't see an error, it means that you were able to access it. If you try the same link with Windows 10, you should see a blank page too, and my expectation was that, if you had a firewall, you would get an error instead.

Now, since I don't have any Windows 8.1 platforms readily available to test, I have just ran a test with Windows 7, and it appears I can replicate your issue there, so it seems there is something going on with older versions of Windows indeed.

I'll try to look into it, but don't expect anything fast.

c627627, 2023-01-14:
Personally I can reboot into Windows 10 any time, but just wanted to contribute to the community by letting you know about a potential bug.
Good luck.

mrkapqa, 2023-01-21:
onmy windows 7 computers id would also no longer function,
running download script for windows 11 or 10 would yield
"error the underlying connection was closed: an unexpected error occured on send"

pbatard, 2023-01-23:
Looks like the issue is that Windows 7 and Windows 8.1 try to use an obsolete/incompatible version of TLS by default to access https sites, and the `vlscppe.microsoft.com` server refuses to reply in that case:

https://blog.darrenjrobinson.com/powershell-the-underlying-connection-was-closed-an-unexpected-error-occurred-on-a-send/

Adding the line
```
[Net.ServicePointManager]::SecurityProtocol = [Net.SecurityProtocolType]::Tls -bor [Net.SecurityProtocolType]::Tls11 -bor [Net.SecurityProtocolType]::Tls12
```
before we query  `vlscppe.microsoft.com` seems to remove the error...

mrkapqa, 2023-01-23:
on my computer running iwndows 7 the problem was solved with this fix here, now i can download again via rufus

https://www.linkedin.com/pulse/error-systemnetwebexception-underlying-connection-send-chima-duru?trk=articles_directory

pbatard, 2023-01-27:
I believe it should be fixed, but I am tempbanned by Microsoft again so I cannot properly test it right now...

pbatard, 2023-01-27:
Got to test Windows 8.1, and I could download ISOs, so I am considering this issue fixed.

closed 2023-01-27 by pbatard