pbatard/rufus #2135: format was unexpectedly slow like full format even though the setting was quick format

infr-automation, 2023-01-15:
Rufus x86 v3.22.1958 (Alpha)
Windows version: Windows 11 Pro, 64-bit (Build 22621.1105)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'General USB Flash Disk USB Device' (1F75:0902)
Found USB device 'Multi Flash Reader' (????:????)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'SDXC Card' (0BDA:0316)
2 devices found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1888, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00586B7F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 14.5 GB (15533586432 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 30341084, Boot: No
2 devices found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 15157, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: exFAT (0x07)
  Detected File System: exFAT
  Size: 116.1 GB (124658909184 bytes)
  Start Sector: 32768, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.06
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kubuntu 22.04.1 LTS amd64'
  Size: 4.6 GB (Projected)
  Has a >64 chars filename
  Uses: GRUB2
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system
  limitations. Because of this, some features from this image may not work...
Using image: jammy-desktop-amd64.iso (4.6 GB)

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Opened \\.\PhysicalDrive4 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 2176 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 5.1 GB)
● Creating Linux Persistence Partition (offset: 5490342400, size: 111.0 GB)
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Notice: Using physical device to access partition data
Formatting (ext3)...
1818623 possible inodes out of 29097983 blocks (block size = 4096)
1454899 blocks (5.0%) reserved for the super user
Creating 888 inode sets: [1 marker = 11.1 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 65536 journal blocks: [1 marker = 819.2 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Formatting (Large FAT32)...
Opened \\?\Volume{a1616c07-94bf-11ed-b62a-d8bbc1782bd3} for exclusive write access
Size : 5.1 GB 10721277 sectors
Cluster size 32768 bytes, 512 bytes per sector
Volume ID is 150b:1760
32 Reserved sectors, 1309 sectors per FAT, 2 FATs
167478 Total clusters
167477 Free clusters
Clearing out 2714 sectors for reserved sectors, FATs and root cluster...
Initializing reserved sectors and FATs...
FAT #0 sector at address: 32
FAT #1 sector at address: 1341
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Setting label...
Format completed.
Opened \\.\PhysicalDrive4 for shared write access
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Grub 2.0 MBR
Writing Grub 2.0 SBR (from embedded)
Found volume \\?\Volume{a1616c07-94bf-11ed-b62a-d8bbc1782bd3}\
Notice: Volume Device Path is \Device\HarddiskVolume12
Waiting for access on \\?\Volume{a1616c07-94bf-11ed-b62a-d8bbc1782bd3}...

infr-automation, 2023-01-15:
The rest was fine just the format was much slower than expected. During the Format, Task manager was showing constant read+write at around 6MB/s with 0.3 ms latency of the microSDXC card.

The rest writing files was at 70MB/s ~

infr-automation, 2023-01-15:
The other unexpected thing was that symlinks are discarded instad of the files being copied twice. My purely logical expectations would be that the files would be copied twice when the filesystem does not support symlinks.

pbatard, 2023-01-15:
You're creating a persistent partition, and there is no "quick" way to create an `ext3` file system. This is actually a very random write intensive operation, as it needs to initialise inodes all over the place.

I suggest you try without a persistent partition. You will see that formatting and writing to the device is much quicker.
Oh, and in case you wonder, the code used by Rufus to format `ext3` partitions is the code from ext2fs, which is the same as for Linux, so it's not really a problem of Rufus using it's own version of `ext3`.

There's really nothing for me to do here (unless you want to volunteer to optimise the Windows ext2fs formatting code), as your issue has everything to do with creating an __optional__ persistent partition, which is inherently slow, so I will close your issue.

closed 2023-01-15 by pbatard

infr-automation, 2023-01-15:
Cool. Probably an ext3/4 expert will use ChatGPT and Copilot to rewrite things with more first principles logic. I saw in Android TV (nvidia shield) there exist two formatters maybe one of them is faster.