pbatard/rufus #2137: Secure boot on Microsoft Surface devices

pineapple63, 2023-01-19:
Just want to let you know about an issue i ran into, just incase anyone else does run into this

It seems that Microsoft Surface devices have 2 levels (or strictnesses) of secure boot, “Microsoft only”, or “Microsoft and third party”, the former option seems to prevent USBs created using Rufus from booting

I was reinstalling Windows 11 onto a Microsoft Surface Go 3 (rather than messing around with a recovery image), and the USB created using Rufus wasn’t booting, i had a look at the secure boot settings, it was set to “Microsoft only”, so i tried changing it to “Microsoft and third party”, and it worked, it booted to the USB

pbatard, 2023-01-19:
Thanks for the report. I've seen hints here and there that, despite the fact that our bootloaders are signed __by MIcrosoft themselves__ for Secure Boot, Surface devices may be reluctant to use them for boot indeed.

On really has to wonder what the point of having a __BLOODY UNIVERSAL__ specification for Secure Boot, that manufacturer are supposed to follow, if it's for the one entity (Microsoft) that somehow managed to weasel its way into being in charge of the whole signing process to say: _"You know what, I'm going to **ABUSE** my power as the keyholder, and deviate from the specs to make my devices more restricted. so that, even if someone goes through the specs-established process of getting their bootloaders signed, they're still not going to boot on my devices"_

So, once again, we have Microsoft bullshitting their way into **abusing** their position as the SecureBoot gatekeeper, which they've done by:
- Putting forward a completely ludicrous pretence for not signing anything GPLv3 ever... with the rather obvious implicit goal of deterring switching a Windows platform to a Linux one by making that process more difficult when Secure Boot is enabled (or, with their thankfully ill-fated attempt at once again restricting what users can do with their ARM WinRT platforms, comes by default and cannot be disabled)
- Placing unwarranted restrictive conditions on who can download Secure Boot revocation lists (for instance, and individual developer cannot produce an app that can update Secure Boot revocation, because Microsoft does not allow anybody but OS manufacturers or firmware manufacturers to redistribute said list (see the legal terms at https://uefi.org/revocationlistfile).
- Signing "special" bootloaders that are *more* Secure Boot than Secure Boot (in the same manner as some people are *more equal* than others), so that, even if you go through the hassle of having your bootloader signed by Microsoft for Secure Boot, you still get **no guarantee** that it will work on a Secure Boot enabled device. Once again, this sure does seem to be purely designed to make it more difficult for the casual user to switch their device to Linux (but Microsoft will of course spew some bullshit that it's to make their device "more secured", even though one has only to point out that it's still Microsoft that is 100% in control of signing every single bootloader for the "less secured" path, so they're going to have to explain how one process they have full vetting on is "less secure" than another process they have full vetting on).

Gotta wonder what Microsoft's next **abuse of power** when it comes to Secure Boot is going to be, because they certainly seem to be hell bent on continuing down that road.

marcosfrm, 2023-01-19:
Yeah, the same settings (using different names) do exist in Hyper-V Generation 2 virtual machines:

https://blog.matrixpost.net/wp-content/uploads/2021/11/secure_boot_ubuntu.png

marcosfrm, 2023-01-19:
> Thanks for the info Marcos. I assume that the "Microsoft Windows" template is the more _restrictive_ one while the "Microsoft UEFI Certificate Authority" is the standard UEFI Secure Boot one.

Yes.

closed 2023-01-30 by pbatard