pbatard/rufus #2140: If you are getting read errors at any stage, then this is a hardware issue, and it has nothing to do with Rufus.

muin360, 2023-01-21:
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Enterprise, 64-bit (Build 19042.631)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:1212)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1968, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x003510F6
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 15.1 GB (16190013440 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'Atlas'
  Detected: Windows 10 ISO (Build 19042.631)
  Size: 1.5 GB (Projected)
  Note: File on disk is larger than reported ISO size by 542 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Atlas_v0.5.2.iso (1.5 GB)
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 15.1 GB)
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as F: after error

Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:1212)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1968, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0038ED8C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.1 GB (16190013440 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:1212)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1968, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0038ED8C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.1 GB (16190013440 bytes)
  Start Sector: 2048, Boot: No

pbatard, 2023-01-21:
Indeed, in a similar reported issue (#1558) the user reported errors during the bad blocks check (which you were supposed to run if you followed the check list that you were asked to follow when opening the issue).

I would **strongly** suggest that you run a bad blocks check against your device.

muin360, 2023-01-21:
Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 15.1 GB)
Waiting for logical drive to reappear...
Formatting to NTFS (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Error while formatting
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
[0x0000007E] The specified module could not be found. - Retrying...
Format error: Undetermined error while formatting.
Re-mounted volume as F: after error

Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:1212)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1968, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01DC35AB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.1 GB (16190013440 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Mass Storage Device USB Device' (14CD:1212)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1968, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01DC35AB
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 15.1 GB (16190013440 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2023-01-21:
Still not seeing the report of your bad blocks check. [here](https://github.com/pbatard/rufus/issues/1799) is an example of log with a bad blocks check report:

```
(...)
Bad Blocks: Checking from block 0 to 14866 (1 block = 512 КБ)
Bad Blocks: Using offset 17838 for fake device check
Bad Blocks: Writing test pattern 0x55
Bad Blocks: Reading and comparing
Bad Blocks: Writing test pattern 0xAA
Bad Blocks: Reading and comparing
Bad Blocks: Check completed, 0 bad blocks found. (0/0/0 errors)
(...)
```

muin360, 2023-01-22:
sorry not working

pbatard, 2023-01-22:
If you can't run a bad blocks check, I'm going to assume that your USB drive is defective and close this issue.

closed 2023-01-22 by pbatard