pbatard/rufus #2142: Rufus does not open

Gamer98YT, 2023-01-21:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [X] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [X] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<When I try to insert the USB I want to use for the ISO image, it stucks on Not Responding, when I open without nothing, it works perfectly until I put the USB, if I try to open Rufus with the USB already inserted, it directly does not open and if y try to open it again says that a Rufus instance its already running. The log doesnt have any error message, because when I connect the USB it hangs up, and the log also>

Log
---
```
Rufus x86 v3.21.1949
Windows version: Windows 11 Pro, 64-bit (Build 22621.525)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
0 devices found

pbatard, 2023-01-21:
Looks similar to #2122, #2096, #1860, #1657 and most especially #1597, most of which appear to be caused by problematic security solutions.

Especially, are you using Sophos Antivirus by any chance?

Gamer98YT, 2023-01-22:
No, Im using windows defender

Gamer98YT, 2023-01-22:
Tried to disable the windows defender and still hangs up when connecting the USB

pbatard, 2023-01-23:
Then see the advice I provided in https://github.com/pbatard/rufus/issues/2122#issuecomment-1365421428

There's really nothing else I can do to help you with besides you identifying the *external* conditions that are responsible for the enumeration freezout/slow times, or debugging the application to find out where exactly it hangs

pbatard, 2023-01-30:
Since there's nothing further I can do, and this appears to be a purely environmental problem, I will close this issue.

closed 2023-01-30 by pbatard