pbatard/rufus #2143: is rufus-tool.com an official domain?

xFrann, 2023-01-21:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Hello,

When using google.com to search for rufus there's in an AD showing first before the official known website (rufus.ie) with 
the domain mentioned in the title.
I believe this might be a fake page used to lure people into downloading the software + additional malware,
as this is a spreading issue for different open source projects used by many.
if this is not the case and the domain is an official domain this would be nice to be mentioned somewhere in a list such as trusted domains.
I just wanted to check if anybody is aware about this, and if this is a fake domain if we can contribute by reporting the ad on the link below
https://support.google.com/ads/troubleshooter/4578507?hl=en

Thank you for offering this tool and for providing support for it's users.

Log
---
```
<FULL LOG>
```

pbatard, 2023-01-21:
> I believe this might be a fake page used to lure people into downloading the software + additional malware,

It is. You did the right thing in reporting it.

The **ONLY** trusted domains for the homepage are rufus.ie or akeo.ie. EVERYTHING ELSE is a scam (though of course some people may link to the release pages at https://github.com/pbatard/rufus/releases or https://www.fosshub.com/Rufus.html or https://reboot.pro/files/file/145-rufus/ which are the only 3 release pages that I update).

xFrann, 2023-01-21:
Thank you for the quick response, I'll make sure to inform any friend that uses Rufus to report the page. :)

closed 2023-01-21 by xFrann

xFrann, 2023-01-27:
Updates, this is happening for multiple software businesses, here is the FBI notice: 
https://www.ic3.gov/Media/Y2022/PSA221221
Stay safe 💪