pbatard/rufus #2145: FAT32 only in GPT mode

infr-automation, 2023-01-23:
Macrorit_PartExpert_WinPE_7.0.0.iso – 467.7 MB

Can do FAT32 only in GPT mode on a 1GB flash drive.

What keeps it from doing FAT32 in MBR UEFI-CSM or dual compatible (Alt+E) ?

Thanks.

latest release version

infr-automation, 2023-01-24:
After resume from Standby today I can select FAT32 in both MBR and GPT. The Rufus app was not even restarted it was the same instance from yesterday. Not sure what happened yesterday that didnt allow FAT32 in GPT mode.

pbatard, 2023-01-24:
No log &rarr; Can't investigate. It's as simple as this.

closed 2023-01-24 by pbatard

infr-automation, 2023-01-30:
Found USB 3.0 device 'General USB Flash Disk USB Device' (1F75:0902)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1888, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00586B7F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 14.5 GB (15533586432 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 30341084, Boot: No
Found USB 3.0 device 'General USB Flash Disk USB Device' (1F75:0902)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1888, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00586B7F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 14.5 GB (15533586432 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI System Partition (0xef)
  Detected File System: FAT12
  Size: 1 MB (1048576 bytes)
  Start Sector: 30341084, Boot: No
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'USB_STRELEC'
  Size: 4.5 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS only)
Using image: WinPE11_10_8_Sergei_Strelec_x86_x64_2023.01.30_English.iso (4.4 GB)

infr-automation, 2023-01-30:
This log is from latest MinGW build

I just clicked around the interface and it let me select FAT32 +  MBR - BIOS or UEFI (Alt+E my favorite one) didnt notice when exactly.  Probably when I was clicking the ISO file I clicked UEFI:NTFS I think and then back to the ISO not sure.

pbatard, 2023-01-30:
You're still not following the check list.

If you have an issue with `Macrorit_PartExpert_WinPE_7.0.0.iso` I expect you to provide a log of what happens when you select that image. And I expect you to use the release version, not an Alpha.

You're trying things at random right now (FreeDOS, Alpha, UEFI:NTFS, Strelec) and I can't help you, because you're going in all directions instead of, as was asked of you, concentrating on your initial issue and that issue only.

Also, be mindful that Alt-E / GPT / FAT32 is **entirely** dependent on the ISO you use (which is why I want to see what Rufus says in it's log when you select the ISO). **I can't magically make a feature that is RESTRICTED by the ISO you use magically become available**.