pbatard/rufus #2146: Error: Could not parse languages

LoopyLucy, 2023-01-25:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
See image attached
Log
---
```
<FULL LOG>
Rufus x86 v3.21.1949
Windows version: Windows 10 Pro, 64-bit (Build 19044.2545)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3775, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xABFEF36B
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Detected File System: NTFS
  Size: 28.9 GB (31050432512 bytes)
  Start Sector: 2048, Boot: No
Downloading https://github.com/pbatard/Fido/releases/download/v1.40/Fido.ps1.lzma
File length: 18.2 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=b73d1e0a-43b5-44c9-9ec8-0f91b69deb00
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=b73d1e0a-43b5-44c9-9ec8-0f91b69deb00&productEditionId=2370&sdVersion=2
Error: Could not parse languages
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=b73d1e0a-43b5-44c9-9ec8-0f91b69deb00
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=b73d1e0a-43b5-44c9-9ec8-0f91b69deb00&productEditionId=2370&sdVersion=2
Error: Could not parse languages
Exited download script with code: 3
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=d6870baf-f37b-441d-8519-e86046715ba4
Querying https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=d6870baf-f37b-441d-8519-e86046715ba4&productEditionId=2370&sdVersion=2
Error: Could not parse languages
Exited download script with code: 3
```
[image]

pbatard, 2023-01-25:
Thanks for the report. It looks like Microsoft have altered the content from their ISO download servers yet again (See #2130 as well as pbatard/Fido#52) and are returning unexpected content.

I'll look into this when I get a chance.

pbatard, 2023-01-25:
By the way, can you please validate what you get when you attempt to download the ISO by going to https://www.microsoft.com/en-GB/software-download/windows11 and trying to obtain an ISO download link?

Right now I am getting:
> Error We are unable to complete your request at this time. Some users, entities and locations are banned from using this service. For this reason, leveraging anonymous or location hiding technologies when connecting to this service is not generally allowed. If you believe that you encountered this problem in error, please try again.

So it could be that the language error has to do with a temporary ban.

In short, can you actually download a Windows 11 ISO if you try to do so in the browser from Microsoft's website?

LoopyLucy, 2023-01-25:
I am currently downloading Windows 11 English International now, I don't get any errors on the website, it is however very slow and buggy (could be Dark Reader doing that though).

JoatWiPete, 2023-01-25:
I am having the same issue, could not parse languages.
I can download an ISO without issue from the Microsoft website.
If I manually go to the [link in the logs](https://www.microsoft.com/en-US/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=d3ece462-1437-40fb-8c61-82a05a7667ba&productEditionId=2370&sdVersion=2), it's trying to ask for me to pick a product language.

See image:
[image]

pbatard, 2023-01-25:
Thanks. I guess the error I get is not related to this issue, as I was playing with Fido earlier and that's probably why I got tempbanned.

pbatard, 2023-01-25:
> If I manually go to the link in the logs, it's trying to ask for me to pick a product language.

Yes, that is correct. The Microsoft servers serve **fragments** of HTML for each query (which is also how the page from the website behaves whenever you click a button there) and we process that to try to get the info we need.

My current guess is that Microsoft altered their response content for the language in a manner that no longer matches what we expect (though I'm looking at it and not seeing anything obvious right now).

pbatard, 2023-01-25:
Ah, I think I get it. Up until now, Microsoft returned:
```html
<select id="product-languages">
(...)
</select>
```
but now it seems they dropped the closing `</select>` and we are expecting `<select>...</select>` pattern from our regular expression filter:

```
'(?s)<select id="product-languages">(.*)?</select>'
```

There seems to be additional changes too, so it's not as simple as adding the `</select>` ourselves

pbatard, 2023-01-25:
Actually what I described above was wrong, we do have the closing `</select>` but the problem Microsoft added extra stuff before, after and within the initial `<select>`.

At any rate, I believe the latest Fido update, which I just pushed, should fix this. Please try again and let me know if you still see the error. And in case you see it, please check the log to ensure that it downloaded Fido `v1.41` rather than `v1.40`. In short, you should see a line that says:
```
Downloading https://github.com/pbatard/Fido/releases/download/v1.41/Fido.ps1.lzma
```

JoatWiPete, 2023-01-25:
That works for me.
Thank you.

pbatard, 2023-01-25:
OK, I'll close this issue then.

closed 2023-01-25 by pbatard