pbatard/rufus #2148: Error: could not parse languages

Benna80, 2023-01-26:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
Hi all, i'd like to create an iso of windows 11 21H2 on a USB stick. I choose the option to download the ISO, i then click on dhe DOWNLOAD button, select the Windows version (in my case Windows 11) , then select the release 21H2 V1 (Build 22000.318), then select the only option available "Windows 11 Home/Pro/Edu" and clicking NEST i get the "Could not parse languages" so i'm not able to continue.

I also tried selecting every other option for Windows 11 but i keep getting the same error

<FULL LOG>
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 11 Pro, 64-bit (Build 22621.1105)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0410 (it-IT)
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
Found USB device 'Multiple Card Reader' (????:????)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk U3 Cruzer Micro USB Device' (0781:5406)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x11D4279F
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 14.9 GB (16027484160 bytes)
  Start Sector: 2048, Boot: Yes
Downloading https://github.com/pbatard/Fido/releases/download/v1.41/Fido.ps1.lzma
File length: 18.2 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/it-IT/software-download/
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=9d707077-4455-4796-9461-20f52ab6493a
Querying https://www.microsoft.com/it-IT/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=9d707077-4455-4796-9461-20f52ab6493a&productEditionId=2093&sdVersion=2
Error: Could not parse languages
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=9d707077-4455-4796-9461-20f52ab6493a
Querying https://www.microsoft.com/it-IT/api/controls/contentinclude/html?pageId=a8f8f489-4c7f-463a-9ca6-5cff94d8d041&host=www.microsoft.com&segments=software-download,windows11&query=&action=getskuinformationbyproductedition&sessionId=9d707077-4455-4796-9461-20f52ab6493a&productEditionId=2093&sdVersion=2
Error: Could not parse languages
Exited download script with code: 3

pbatard, 2023-01-26:
Thanks for the report. That's probably a remnant from #2146 which was fixed yesterday, where non English languages probably get different content.

The workaround for now is to change the Rufus language to English as you should find that it works then.

I'll see if I can figure out what the issue is with Italian and potentially other non English locales.

Benna80, 2023-01-26:
Nice, i've changed to English and then RUFUS went trough the preceding step so i was able to download Windows 11 21H2 in Italian.

Thanks, much apreciated

pbatard, 2023-01-27:
This should be fixed now.

closed 2023-01-27 by pbatard