pbatard/rufus #2149: Too many bad blocks error

diorisdash, 2023-01-27:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [X ] Plugging the USB into a different port.
  - [X ] Running Rufus on a different computer.
- [X ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
too many bad blocks

Log
---
```
<FULL LOG>
Rufus x86 v3.21.1949
Windows version: Windows 8.1 Home, 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'APPLE SD Card Reader USB Device' (05AC:8406)
No logical drive found (unpartitioned?)
Found UAS (USB 3.0) device 'JMicron Tech SCSI Disk Device' (152D:0578)
Device eliminated because it was detected as a Hard Drive (score 13 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
2 devices found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1019, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.8 GB (8387559424 bytes)
  Start Sector: 2048, Boot: No
Found USB 3.0 device 'APPLE SD Card Reader USB Device' (05AC:8406)
No logical drive found (unpartitioned?)
Found UAS (USB 3.0) device 'JMicron Tech SCSI Disk Device' (152D:0578)
Removing C: from the list: This is the system disk!
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
2 devices found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1019, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.8 GB (8387559424 bytes)
  Start Sector: 2048, Boot: No
2 devices found
No volume information for drive 0x82
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7764, Tracks per cylinder: 255, Sectors per track: 63
[... 437 lines cut ...]
Bad Blocks: 1268
Bad Blocks: 1269
Bad Blocks: 1270
Bad Blocks: 1271
Bad Blocks: 1272
Bad Blocks: 1273
Bad Blocks: 1274
Bad Blocks: 1275
Bad Blocks: 1276
Bad Blocks: 1277
Bad Blocks: 1278
Bad Blocks: 1279
Too many bad blocks, aborting test
Bad Blocks: Check completed, 264 bad blocks found. (0/0/264 errors)

Found USB 3.0 device 'APPLE SD Card Reader USB Device' (05AC:8406)
No logical drive found (unpartitioned?)
Found UAS (USB 3.0) device 'JMicron Tech SCSI Disk Device' (152D:0578)
Removing C: from the list: This is the system disk!
Found USB 2.0 device 'General UDisk USB Device' (048D:04D2)
2 devices found
No volume information for drive 0x83
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 1019, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

diorisdash, 2023-01-27:
sorry for the X's i got confused on the checklist

pbatard, 2023-01-27:
Your USB drive **is** defective. There is nothing Rufus can do about it.

A bad blocks check detects hardware errors. If it reports errors, it means your hardware has failed.

You will need to replace that drive.

closed 2023-01-27 by pbatard

diorisdash, 2023-01-27:
nono. just found out the USB dosen't like NTFS, it works with fat32 :(

pbatard, 2023-01-27:
Again, if you have a bad blocks report, your USB *is* defective. You may try to fool yourself otherwise, but then it's your choice to try to work with defective hardware. If you are hoping that somehow Rufus is wrong about the bad blocks, feel free to use another application, because you will find that it will report the same thing.