pbatard/rufus #2151: Disable Logging...is it possible?

jvidalw, 2023-01-30:
﻿Is it possible to disable the Log creation? 
That's all I want to know

Obviuosly, no log included, since This is not a bug report.

pbatard, 2023-01-30:
> Is it possible to disable the Log creation?

No.

The log is not for user convenience (or inconvenience). It is for me, the developer, to get a fighting chance to actually be able to troubleshoot an issue. Having a permanent log from the last session that was run can provide invaluable data, so there is no way I will ever allow users to disable it.

Furthermore, can you please explain what your issue is with having a log saved to disk? How does that impair what you are trying to accomplish?

jvidalw, 2023-01-30:
it bothers me, because it creates a folder that i have to delete every single time I use rufus.

pbatard, 2023-01-30:
> it bothers me

So you are looking into the Windows AppData folder every time you run an application? I gather that you must be bothered a lot then...

closed 2023-01-30 by pbatard

jvidalw, 2023-01-30:
that's different, those are system folders, usually containinig junk (which I routinely clean using tools like ccleaner).
IMHO it should be possible to disable logging and users could enable it only when troubleshooting. it sounds a lot more logical.

pbatard, 2023-01-30:
Not gonna happen, sorry. Also since you thought you didn't need to provide a log, I cannot tell if you are using the portable or regular version. I am guessing you really want to use the regular version instead of the portable one, because, if you read the FAQ, you will understand that the [difference between portable and regular is probably not what you think it is](https://github.com/pbatard/rufus/wiki/FAQ#user-content-What_is_the_difference_between_the_portable_and_the_regular_version).

jvidalw, 2023-01-30:
that's too bad.
well, that's all i wanted to know. thx.