pbatard/rufus #2153: W7 - Download > Error: This feature not available for this version of Windows?

mrkapqa, 2023-01-31:
﻿Hello ,

since few days Windows 7 seem no longer able to achieve the download with Rufus,

after fixing TLS prior it would have functioned for some days, but now it says

Error: This feature not available for this version of Windows.

Thought it has been fixed by now.

Thanks.

pbatard, 2023-01-31:
I deliberately disabled Windows 7 support per https://github.com/pbatard/Fido/releases/tag/v1.42

I spent about 6 hours trying to make it work for all cases (the TLS fix was not enough because a separate Windows 7 workaround resulted in the message being issued by Microsoft servers when they ban your IP generating an out of memory exception), and decided this was just no longer worth the trouble, so Windows 7 support is off the table for downloads.

Please do not expect **busy** developer to support obsolete platforms forever. You had your run with Windows 7. If you **choose** to continue to use it, and as is the case for Windows XP, [you are 100% on your own](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED).

closed 2023-01-31 by pbatard

pbatard, 2023-01-31:
Again, it's not that simple. As I explained errors need to be handled too, especially now that Microsoft is a lot more proactive about banning IPs after a few downloads) and I simply could not make that work in a reasonable timeframe. If I only go with the TLS fix, the script will crash as soon as the user is banned, with no explanation, and that's just not something I want to have in Rufus.

Please understand that there's **A LOT** more work involved in proper development than jump on the first solution that looks like it may do the job.

So, let me clear here: There will be NO reverting of the termination of Windows 7 support for ISO downloads. You had your run. Move on.

Da-Github, 2023-03-12:
> I deliberately disabled Windows 7 support per https://github.com/pbatard/Fido/releases/tag/v1.42

You did WHAT !?  I would like to vehemently protest this discontinuation of windows 7 support.  -_- 

> So, let me clear here: There will be NO reverting of the termination of Windows 7 support for ISO downloads. You had your run. Move on.

Nope not going to do that! I'm still happy with windows 7 and XP. I do still have it running on modern hardware.  I'd like to better understand the issue of banning IP's you mentioned earlier, Does Microsoft actually ban the downloading of ISO's I'm aware that there really trying to push their media creation tool (which doesn't work on windows 7 btw) but I wasn't aware of them banning IP addresses that could be irritating. any additional information would be helpful. Does the old version still work ?

Instead of dropping support for windows 7 all together would it be possible to redirect windows 7 to an Alternate host of the ISO's with a descriptive popup message ?

pbatard, 2023-03-12:
> Does Microsoft actually ban the downloading of ISO's

Yes. If you download a few too many ISOs in a short time, or if your IP is in a geolocalisation that Microsoft has decided to ban (e.g. Russia) you won't be able to download ISOs.

This is an IP ban, which means there is absolutely nothing I can do in Rufus, since we are downloading the ISOs directly from Microsoft servers, and therefore an IP ban enacted by Microsoft can't be worked around.

> Instead of dropping support for windows 7 all together would it be possible to redirect windows 7 to an Alternate host of the ISO's with a descriptive popup message ?

If you want to **infringe** copyright by duplicating the Microsoft retails ISOs, feel free to do that yourself, and then deal with law enforcement and the obligatory DMCA requests you will receive.

Also, I dropped Windows 7 support in Fido because I could no longer ensure parity with other platforms when it came to error handling.

You should also be aware that the upcoming Rufus 3.22 will also be the very last version that can run on Windows 7. Starting with Rufus 3.23, you will no longer be able to run Rufus on Windows 7.

You are of course free to try to continue to use Windows 7, or Windows XP, or Windows 3.1. But don't expect **OTHERS** to invest **THEIR TIME** to support those OSes for you.

pineapple63, 2023-03-12:
> You did WHAT !? I would like to vehemently protest this discontinuation of windows 7 support. -_-

I get the feeling you would protest every time a piece of software stops supporting an old operating system (i could probably name a few pieces of software i use which have dropped Windows 7 support, and even one which has already dropped Windows 8.1 support)

For Windows 8.1 and Windows 11 ISOs, there should be nothing stopping you downloading the ISO from Microsofts website (for Windows 11, you should be offered the option for a direct ISO download in addition to the tool), for Windows 10 you would likely have to trick Microsofts site into believing you are using Linux or Mac OS

Aokromes, 2023-03-18:
> > I deliberately disabled Windows 7 support per [pbatard/Fido@`v1.42` (release)](https://github.com/pbatard/Fido/releases/tag/v1.42)
> 
> You did WHAT !? I would like to vehemently protest this discontinuation of windows 7 support. -_-
> 
> > So, let me clear here: There will be NO reverting of the termination of Windows 7 support for ISO downloads. You had your run. Move on.
> 
> Nope not going to do that! I'm still happy with windows 7 and XP. I do still have it running on modern hardware. I'd like to better understand the issue of banning IP's you mentioned earlier, Does Microsoft actually ban the downloading of ISO's I'm aware that there really trying to push their media creation tool (which doesn't work on windows 7 btw) but I wasn't aware of them banning IP addresses that could be irritating. any additional information would be helpful. Does the old version still work ?
> 
> Instead of dropping support for windows 7 all together would it be possible to redirect windows 7 to an Alternate host of the ISO's with a descriptive popup message ?

even 14 years old cpus supports windows 10, if you use something older cpu you can get windows 10 compatible 2nd hand pcs for less than 100 euros.

pineapple63, 2023-03-18:
> even 14 years old cpus supports windows 10, if you use something older cpu you can get windows 10 compatible 2nd hand pcs for less than 100 euros.

I get the feeling its not a case of their computer not being compatible, but that they don’t want to have Windows 10 on their main PC (for one reason or another, I suspect its either a case of they don’t like the UI and so refuse to upgrade, or the telemetry that Windows 10 sends to microsoft, or both)

Aokromes, 2023-03-19:
> I get the feeling its not a case of their computer not being compatible, but that they don’t want to have Windows 10 on their main PC (for one reason or another, I suspect its either a case of they [...]

well, iirc telemetry was added into windows 7 too xd, and open shell menu allows having windows 7 like start menu :)