pbatard/rufus #2156: No Wifi adapter after install with offline account.

Vulgaris-Viridis, 2023-02-04:
Also if you want me to dedicate my time to report an issue so you can fix your own app, "please" stop with this infantile warnings and threats. People don't have to do sh!t. 

Github was made for open source collaboration not for making threats to people who are trying to help (YOU).

" 1. You *MUST* 

YOU HAVE BEEN WARNED.
-->
"

Ridiculous.

pineapple63, 2023-02-04:
The problem is most likely that the installer does not have the drivers for the WiFi card

Sometimes you can get the (correct) drivers through Windows Update (by connecting via Ethernet temporarily, you may need to use a USB to Ethernet adapter), but failing that you would need to get the drivers from the manufacturer of the computer

And if the drivers aren’t available, then unfortunately there isn’t a solution

Vulgaris-Viridis, 2023-02-04:
Thanks @pineapple63, you're probabky right. It goes without saying - no issues with the adapter on any other OS. It works fine on all previous windows and on pretty much every Linux DE there is.

Vulgaris-Viridis, 2023-02-04:
Thanks @pineapple63, you're probably right. It goes without saying - no issues with the adapter on any other OS. It works fine on all previous windows and on pretty much every Linux DE there is. 

Edit: Also worth noting, I tried every single "solution" from first few pages of search results, nothing seems to be working, I just can't get the Win to recognize adapter..

pbatard, 2023-02-04:
> People don't have to do sh!t.

Exactly. I shouldn't have to waste my time trying to **guess** the data I need to properly help you.

> open source collaboration

Yup, *collaboration* is a two way street. You give me the data I need and __then__ I try to help you.

If you break the terms of the collaboration deal and throw a fit about equating being asked to provide a minimum level of data to a threat, then don't expect much help.

> making threats to people who are trying to help (YOU).

And how exactly are you helping me (or anyone) by not even disclosing the kind of network adapter you use (hardware ID or at least brand and model) or how you created the media by providing a log?

pbatard, 2023-02-04:
> You are developing the app primarily for yourself.

Man, you do have a twisted view of how other people function. No wonder you clearly seem under the impression that the world should be at your feet. So please go back to your selfish world, and let people who are genuinely trying to help others, and have taken steps to try to ensure that people are in the best position to receive help from the get go, try to do their job.

> I've chosen to apparently waste my time by posting an error here so that you can improve your app.

No, what you've done is come here without providing enough data about your issue. You've been called on it right from the start and got angry that someone had the audacity to ask you to provide enough data to investigate your issue properly.

> So please don't waste time on me.

I won't. And to make sure that this will truly be the case moving forward, you have been banned from posting in this issue tracker.