pbatard/rufus #2159: Not showing the Windows 11 Installation setup for 22621.1194

dinosaw, 2023-02-08:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
After selecting the ISO for Windows 11 Home/Pro [22621.1194](https://learn.microsoft.com/en-us/windows/release-health/windows11-release-information), the special Windows 11 Installation dropdown is not an option. I am able to do this with an earlier build of Windows 11 without issue.

Log
-----------------
```
Rufus x86 v3.13.1730
Windows version: Windows 10 64-bit (Build 19045.2486)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Could not get Disk Extents: [0x00000001] Incorrect function.
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1919, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {CF0DD2D3-C36C-414B-B590-DCD37901BA1E}
Max parts: 128, Start Offset: 17408, Usable = 15791520256 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {9787CA38-5CF4-45D9-BDE9-3D391C428A08}
  Size: 14.7 GB (15789408768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {42D8DF37-180E-4ABC-B35E-0173431506C5}
  Size: 512 KB (524288 bytes)
  Start Sector: 30840737, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCSA_X64FRE_EN-US_DV5'
  Size: 5.7 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 22621.1194.230121-0244.NI_RELEASE_SVC_PROD3_CLIENTMULTI_X64FRE_EN-US.ISO (5.7 GB)
```

pbatard, 2023-02-08:
`22621.1194.230121-0244.NI_RELEASE_SVC_PROD3_CLIENTMULTI_X64FRE_EN-US.ISO` doesn't look like an official Microsoft release to me.

Where exactly did you download this ISO?

If it's an unofficial ISO that you did not obtain from Microsoft, please be aware that I cannot support it in Rufus.

dinosaw, 2023-02-08:
That is most likely my issue, as this was an ISO built via UUP. Strange though, because the earlier version was as well an it worked fine. But I'll go ahead and close this out. Thank you.

closed 2023-02-08 by dinosaw

pbatard, 2023-02-08:
I seem to recall that someone had an issue with UUP generated ISOs in the past, and this was due to the manner/environment with which the ISO was generated. Let me see if I can dig that issue back.

pbatard, 2023-02-08:
Can't seem to locate that issue. But I kind of remember that when the ISO was generated from a different machine, it appeared to fix the issue, so you may want to try that. Still, I'm afraid that the only ISOs I can support for the extended options are the ones that are officially provided by Microsoft, as there are just too many ways in which non-official ISOs can break things.

dinosaw, 2023-02-08:
Understood - I appreciate you looking. I'll try it on a few different systems and if I have any success then I'll report back.

dinosaw, 2023-02-08:
I'm going to leave this closed because I believe the root of the issue is still with UUP, however I did find something when trying another system.

I can't believe this, but I was using a much older release of Rufus - specifically 3.13 as noted in the logs in the OP. Very much user error, but the reason was I had confused an installed version on my system and a portable version on my system.

That said, I downloaded the portable version of the most recent Rufus available and this time it identifies the ISO as a Windows 11 image. Unfortunately there is still no `Standard Windows 11 Installation` option.

I know you don't provide support for any images that aren't official, but I did want to at least mention this to see if you had any ideas and if anyone else has similar issues. No need to reply back, as I'll keep this as closed for now.

Below are the logs from two different images built using UUPdump - an 22621.1105 Home/Pro build, and a 22621.1105 Pro build.

```
# 1105 - Multi
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Enterprise, 64-bit (Build 19045.2486)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1919, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {CF0DD2D3-C36C-414B-B590-DCD37901BA1E}
Max parts: 128, Start Offset: 17408, Usable = 15791520256 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {9787CA38-5CF4-45D9-BDE9-3D391C428A08}
  Size: 14.7 GB (15789408768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {42D8DF37-180E-4ABC-B35E-0173431506C5}
  Size: 512 KB (524288 bytes)
  Start Sector: 30840737, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CCSA_X64FRE_EN-US_DV5'
  Detected: Windows 11 ISO (Build 22621.1105)      << - - - - - - - - - -
  Size: 5.6 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 22621.1105.230106-2203.NI_RELEASE_SVC_PROD1_CLIENTMULTI_X64FRE_EN-US.ISO (5.6 GB)
```
```
# 1105 - Pro
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Enterprise, 64-bit (Build 19045.2486)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1919, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {CF0DD2D3-C36C-414B-B590-DCD37901BA1E}
Max parts: 128, Start Offset: 17408, Usable = 15791520256 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: NTFS
  ID: {9787CA38-5CF4-45D9-BDE9-3D391C428A08}
  Size: 14.7 GB (15789408768 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  Detected File System: FAT12
  ID: {42D8DF37-180E-4ABC-B35E-0173431506C5}
  Size: 512 KB (524288 bytes)
  Start Sector: 30840737, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'CPRA_X64FRE_EN-US_DV5'
  Detected: Windows 11 ISO (Build 22621.1105)     << - - - - - - - - - -
  Size: 5.6 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 22621.1105.230106-2203.NI_RELEASE_SVC_PROD1_CLIENTPRO_OEMRET_X64FRE_EN-US.ISO (5.6 GB)
```

pbatard, 2023-02-08:
> `Rufus x86 v3.13.1730`

Ah shoot, I totally missed that from your log.

> Unfortunately there is still no `Standard Windows 11 Installation` option.

Because the options have been moved: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_I_no_longer_see_the_option_to_bypass_TPMSecure_Boot_with_Windows_11

dinosaw, 2023-02-08:
> Because the options have been moved: https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_I_no_longer_see_the_option_to_bypass_TPMSecure_Boot_with_Windows_11

Good grief. All sorts of user error in this issue. I apologize for taking up your time. FWIW I did take note of that in the FAQ and was hitting Start to check if the options became available, but that step over time disappeared with me trying to troubleshoot all the other stuff that I forgot I needed to do that, heh.

Anyway, it does indeed show all good now. Sorry again, but your help is much appreciated.