pbatard/rufus #2160: [Feature Request] Prevent standby / hibernate / sleep mode

oakwhiz, 2023-02-09:
Was just bit by this recently because I forgot to use PowerToys to temporarily disable sleep. It can be somewhat error prone to manually disable sleep because you might either forget to re-enable sleep again, or if you use the timed disable option, you might not provide enough time for the I/O speeds (could be network I/O when retrieving the disk image) and size of the disk images in use. It would be easier if Rufus had the ability to inhibit sleep for the full duration of any in-progress operation because that would be one less thing for me to consciously have to think about.

oakwhiz, 2023-02-09:
Continuation of issue #1519

pbatard, 2023-02-09:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want

Not enough people are going to benefit from this. Not worth adding it.

closed 2023-02-09 by pbatard