pbatard/rufus #2163: Enhancement: special warning when flashing big disks

pakal, 2023-02-13:
So, after a day of flashing different images for Raspberry Pi, I misclicked and formatted my 2TB veracrypt-encrypted external drive, which was, as one would expect, listed as "removable device" by Windows (the USB key itself was badly inserted into PC).

Since we automatically ignore recurring "warning" popups after some times, I thought : maybe, like BalenaEtcher, Rufus should display a special, graphic, warning, when someone tries to write a disk over 256 Go or so ? Since most USB keys are below that size, and all external drives are now above ?

It will never prevent all the kinds of horrors that can happen at the end of a long day, but might save distracted people like me ^^'

Thaaaanks

pbatard, 2023-02-13:
All latest versions of Rufus lists disks by order of increasing size, so, quite frankly, unless you were trying to format a drive that was larger than 2 TB, I have trouble explaining how you managed to format the wrong drive by mistake, as the default selection from Rufus should have been any other drive but your 2 TB veracrypt one.

Also, Rufus already uses large size [as a means to treat drives that are larger than 800  GB as fixed drives](https://github.com/pbatard/rufus/blob/master/src/smart.c#L452-L455), so as not to list them by default, so in most circumstances, your drive should also have been unlisted by default.

In short, while I am not directly accusing you of anything here, I am far from having enough data to draw the same conclusion as you do, and I have to point out that, whenever someone manages to erase a drive by mistake, they tend to tweak their report of what really happened to make it look as if Rufus was the sole culprit of the issue, whereas the actual situation is far from being as clear cut as what they are trying to represent, especially as:

1. They might not report the they enabled the listing of fixed drives/HDDs through the advanced options, which is something that is never enabled by default and therefore that needs to be actively set by the user.
2. They might not report the they changed the default selection, that did pick the drive they really wanted to format (on account that drives are listed by increasing size).

In short, despite it being explicitly requested, the fact that you chose not to provide a log, with both your drives present, that would let me find out whether the detection issue is as clear cut as you are presenting it, or if there were other factors at play, forces me to **guess** what might have really happened, and unfortunately for you, since you did withhold critical information as to what Rufus actually did detect and list by default, I'm having a hard time seeing anything that I should add on top of the features that already exist in Rufus to prevent just the kind of mistake that happened. Especially, my issue here is that, no matter what I do, at the end of the day, the user will always be left with the **choice** to format whichever removable drive they want, which will always lead to data loss, and users trying to point the finger at Rufus for what was a pure user mistake.

In short, there's only so much hand holding I'm going to add in Rufus, and if you end up erasing a drive by mistake because you ignore explicit warnings, enable advance options or change the default selection, then, unless you do present information that demonstrates otherwise, I'm going to conclude that that mistake was entirely on you.

closed 2023-02-13 by pbatard

pakal, 2023-02-13:
Hello, thanks for the detailed explanation, 

I'm really *not* pointing the finger at Rufus, I mentioned that my usb key unfortunately wasn't mounted (my micro-sd adapter is cheap and faulty), and yes by force of habit I ignored the pre-flash warning. My bad. ^^'

Since the enhancement I thought about is already implemented (with a threshold of 800Go), it means it's more of a problem with my 2To external disk instead (I hadn't checked the "list usb hard drives" option); maybe this setup, with a partition encrypted by veracrypt, is presented differently by Windows.

Here is a ticket with a proper Log so : 
https://github.com/pbatard/rufus/issues/2164

And thanks again for the software, since win32disk-imager is now too outdated, it's a life saver