pbatard/rufus #2164: SanDisk Extreme Portable 2TB is listed by default

pakal, 2023-02-13:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------

My external USB-C SSD (SanDisk Extreme Portable 2TB SSD 1050MB/s SDSSDE61-2T00-G25) is listed by default in Rufus, despite being over 800Go, so here is the log in case some exception or new logic should be added ^^

Please note that it contains a single partition, encrypted in whole by veracrypt.

Log
---
```
Rufus x86 v3.21.1949
Windows version: Windows 11 Pro, 64-bit (Build 22621.1105)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.1) device 'SanDisk Extreme 55AE SCSI Disk Device' (0781:55AE)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Could not get device number for device \\.\P:: Windows error code 0x00000001 (FormatMessage error code 0x00003AFC)
Ignoring volume '\\?\Volume{9a85e63a-6296-11ed-89d0-106fd9dcde06}' because it has more than one extent (RAID?)...
Could not get Disk Extents: Windows error code 0x00000001
Could not get Disk Extents: Windows error code 0x00000001
Notice: Volume Device Path is \Device\HarddiskVolume17
Waiting for access on \\?\Volume{ff7e3530-0000-0000-0000-100000000000}...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
NOTE: Could not identify the process(es) or service(s) accessing \Device\HarddiskVolume17
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Could not get device number for device \\.\P:: Windows error code 0x00000001
1 device found
Ignoring volume '\\?\Volume{9a85e63a-6296-11ed-89d0-106fd9dcde06}' because it has more than one extent (RAID?)...
Could not get Disk Extents: Windows error code 0x00000001
Could not get Disk Extents: Windows error code 0x00000001
No volume information for drive 0x83
Disk type: FIXED, Disk size: 2To, Sector size: 512 bytes
Cylinders: 243197, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xFF7E3530
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 (0x06)
  Detected File System: (Unrecognized)
  Size: 1.8 TB (2000363192320 bytes)
  Start Sector: 2048, Boot: No
```
[image]

pbatard, 2023-02-13:
Thanks. Any chance you could also send me the log with your disk plugged in after pressing <kbd>Alt</kbd><kbd>.</kbd>?

This will enable enumeration debug, and you should see a lot of extra data in the log when listing the disk.

pakal, 2023-02-13:
Sure, here is the fuller log (I'm on French keyboard but "alt+;" seems to have worked)

```
Rufus x86 v3.21.1949
Windows version: Windows 11 Pro, 64-bit (Build 22621.1105)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x040C (fr-FR)
Will use default UI locale 0x040C
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'fr-FR'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found UAS (USB 3.1) device 'SanDisk Extreme 55AE SCSI Disk Device' (0781:55AE)
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Could not get device number for device \\.\P:: Windows error code 0x00000001 (FormatMessage error code 0x00003AFC)
Ignoring volume '\\?\Volume{9a85e63a-6296-11ed-89d0-106fd9dcde06}' because it has more than one extent (RAID?)...
Could not get Disk Extents: Windows error code 0x00000001
Could not get Disk Extents: Windows error code 0x00000001
Notice: Volume Device Path is \Device\HarddiskVolume17
Waiting for access on \\?\Volume{ff7e3530-0000-0000-0000-100000000000}...
Warning: Could not obtain exclusive rights. Retrying with write sharing enabled...
NOTE: Could not identify the process(es) or service(s) accessing \Device\HarddiskVolume17
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Could not get device number for device \\.\P:: Windows error code 0x00000001
1 device found
Ignoring volume '\\?\Volume{9a85e63a-6296-11ed-89d0-106fd9dcde06}' because it has more than one extent (RAID?)...
Could not get Disk Extents: Windows error code 0x00000001
Could not get Disk Extents: Windows error code 0x00000001
No volume information for drive 0x83
Disk type: FIXED, Disk size: 2To, Sector size: 512 bytes
Cylinders: 243197, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xFF7E3530
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 (0x06)
  Detected File System: (Unrecognized)
  Size: 1.8 TB (2000363192320 bytes)
  Start Sector: 2048, Boot: No
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#vid_2109&pid_2811#8&1199c2a1&0&11#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[220]: USB\VID_046D&PID_0825\200901010001
  Found ID[152]: USB\VID_045E&PID_077D\9&102DED21&0&3
  Found ID[224]: USB\VID_04F2&PID_0402\9&102DED21&0&4
Processing Hub 3:
Processing Hub 4:
Processing Hub 5:
  Hub[2] = '\\?\usb#root_hub30#9&12a12108&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[232]: USB\VID_051D&PID_0002\3B1446X13526__
  Found ID[033]: USB\VID_0781&PID_55AE\MSFT3032323335324E343032323937
Processing Hub 6:
Processing Hub 7:
  Hub[3] = '\\?\usb#root_hub30#7&27ed1eec&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[210]: USB\VID_2109&PID_8110\8&1199C2A1&0&5
  Found ID[254]: USB\VID_0E8D&PID_0608\000000000
  Found ID[233]: USB\VID_2109&PID_2811\8&1199C2A1&0&11
  Found ID[027]: USB\VID_0B05&PID_19AF\9876543210
Processing IDs belonging to 'UASPSTOR':
  USB\VID_0781&PID_55AE\MSFT3032323335324E343032323937
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA__________HDWE160FS2A'
Found non-USB non-removable device 'TOSHIBA HDWE160' => Eliminated
Processing 'SCSI' device:
  Hardware ID: ''
Found non-USB non-removable device 'KINGSTON SKC3000S1024G' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskTOSHIBA__________HDWE160FS2A'
Found non-USB non-removable device 'TOSHIBA HDWE160' => Eliminated
Processing 'SCSI' device:
[... 25 lines cut ...]
SearchProcess: Could not open process 32472: Access Denied
SearchProcess: Could not open process 24968: Access Denied
SearchProcess: Could not open process 26168: Access Denied
NOTE: Could not identify the process(es) or service(s) accessing \Device\HarddiskVolume17
Ignoring drive '\\.\D:' as it spans multiple disks (RAID?)
Could not get device number for device \\.\P:: Windows error code 0x00000001
1 device found
Windows volumes from this device:
● \\?\Volume{ff7e3530-0000-0000-0000-100000000000}\ @1048576
Ignoring volume '\\?\Volume{9a85e63a-6296-11ed-89d0-106fd9dcde06}' because it has more than one extent (RAID?)...
Could not get Disk Extents: Windows error code 0x00000001
Could not get Disk Extents: Windows error code 0x00000001
No volume information for drive 0x83
Disk type: FIXED, Disk size: 2To, Sector size: 512 bytes
Cylinders: 243197, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xFF7E3530
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT16 (0x06)
  Detected File System: (Unrecognized)
  Size: 1.8 TB (2000363192320 bytes)
  Start Sector: 2048, Boot: No

```

pbatard, 2023-02-13:
Thanks. I now have a better idea why your device is listed by default.

The main reason is that, because SanDisk are known as mostly a purveyor of flash drives, we adjust the score whenever we see SanDisk reported by the device. So, our score (positive means device Rufus will not list by default, negative means it will) goes:

* `3` (`0 + 3`) on account that the drive reports itself as Fixed
* `13` (`3 + 10`) on account that drive is larger than 800 GB
* `-2` (`13 - 15`) on account that the drive ID contains `SANDISK`
* `-7` (`-2 - 5`) on account that the VID is `0781` which is a know VID for SanDisk

And thus, with a negative final score, your device will appear by default.

I guess I need to reevaluate how far I should skew the score for SanDisk devices, now that it has been reported that they produce other devices than common USB Flash drives. There are multiple ways we can go about it, and I'm probably going to use a mix of those.

I'll get this sorted out for the next release.

commit 637f8338 (2023-02-13): [core] fix default listing of large SanDisk SSD devices

* Closes #2164
* Also add breakdown of score computation when device enumeration debug is active
* Also fix a minor Code Analysis warning in msapi_utf8.h
commit b163b3df (2023-02-13): [core] fix default listing of large SanDisk SSD devices

* Closes #2164
* Also add breakdown of score computation when device enumeration debug is active
* Also fix a minor Code Analysis warning in msapi_utf8.h