pbatard/rufus #2168: Windows user experience window not shown

acuna-public, 2023-02-15:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Windows user experience not shown in Windows 7 even with MBR and official MSDN image. Thanks.

Log
---
```
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 8 ГБ, Sector size: 512 bytes
Cylinders: 976, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0057DC8A
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Detected File System: (Unrecognized)
  Size: 7.5 GB (8026849280 bytes)
  Start Sector: 2048, Boot: Yes
Unable to send request: This system's SSL library is too old to be able to access this website.
Checking for Rufus updates...
Checking release channel...
Unable to send request: This system's SSL library is too old to be able to access this website.
Updates: Unable to access version data
```

pineapple63, 2023-02-15:
https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

closed 2023-02-15 by pbatard

acuna-public, 2023-02-16:
Guys, who have an issues with Windows user experience window just create an image with Rufus and add `AutoUnattend.xml` file on the root of your flash drive:

	<unattend xmlns="urn:schemas-microsoft-com:unattend">
		<settings pass="windowsPE">
			<component name="Microsoft-Windows-Setup" processorArchitecture="amd64" publicKeyToken="31bf3856ad364e35" language="neutral" versionScope="nonSxS" xmlns:wcm="http://schemas.microsoft.com/WMIConfig/2002/State" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
				<RunSynchronous>
					<RunSynchronousCommand wcm:action="add">
						<Order>1</Order>
						<Path>reg add HKLM\System\Setup\LabConfig /v BypassTPMCheck /t reg_dword /d 0x00000001 /f</Path>
					</RunSynchronousCommand>
					<RunSynchronousCommand wcm:action="add">
						<Order>2</Order>
						<Path>reg add HKLM\System\Setup\LabConfig /v BypassSecureBootCheck /t reg_dword /d 0x00000001 /f</Path>
					</RunSynchronousCommand>
					<RunSynchronousCommand wcm:action="add">
						<Order>3</Order>
						<Path>reg add HKLM\System\Setup\LabConfig /v BypassRAMCheck /t reg_dword /d 0x00000001 /f</Path>
					</RunSynchronousCommand>
					<RunSynchronousCommand wcm:action="add">
						<Order>5</Order>
						<Path>reg add HKLM\System\Setup\LabConfig /v BypassCPUCheck /t reg_dword /d 0x00000001 /f</Path>
					</RunSynchronousCommand>
					<RunSynchronousCommand wcm:action="add">
						<Order>4</Order>
						<Path>reg add HKLM\System\Setup\LabConfig /v BypassStorageCheck /t reg_dword /d 0x00000001 /f</Path>
					</RunSynchronousCommand>
				</RunSynchronous>
				<UserData>
					<ProductKey>
					<Key></Key>
					</ProductKey>
				</UserData>
			</component>
			<component name="Microsoft-Windows-Shell-Setup" processorArchitecture="amd64" publicKeyToken="31bf3856ad364e35" language="neutral" versionScope="nonSxS" xmlns:wcm="http://schemas.microsoft.com/WMIConfig/2002/State" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
				<OOBE>
					<HideOnlineAccountScreens>true</HideOnlineAccountScreens>
					<HideWirelessSetupInOOBE>true</HideWirelessSetupInOOBE>
					<ProtectYourPC>3</ProtectYourPC>
				</OOBE>
			</component>
		</settings>
	</unattend>

It will skip CPU, RAM and TPM check and pypass online login window, something which Windows user experience window do.