pbatard/rufus #2169: Windows to Go Updates issue

syneoa, 2023-02-15:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Not able to perform typical cumulative Windows 11 updates once I boot into OS created on a SSD drive with rufus W2GO option. Interestingly enough, other updates works fine, eg. drivers, .net, bios, etc but for example latest Windows 11 February  update can't install. Tried with downloading msu and running manually but issue persist. KB5022845 Install error - 0x80070002

Log
---
```
<FULL LOG>
```

pineapple63, 2023-02-15:
I would guess that this is most likely a Windows issue, not a Rufus issue

And considering that Microsoft removed official support for Windows to Go a while ago (before Windows 11 was even released), and also, as far as i know, Home and Pro editions have never officially supported Windows to Go, i very much doubt you would get support from Microsoft regarding this

syneoa, 2023-02-15:
Yes I understand this. To bad since it was really nice feature. I have tried with Enterprise versions but I guess this doesn't matter.
Maybe a suggestion only to add a warning in Rufus that W2GO can't be serviceable anymore before creating W2GO drive just to avoid this question being repeated.

pbatard, 2023-02-15:
Yes, it's a pure Microsoft issue.

Rufus creates Windows To Go drives using the **official** Microsoft documented way of doing so at https://docs.microsoft.com/en-us/previous-versions/windows/it-pro/windows-8.1-and-8/jj721578(v=ws.11).

So if you create a Windows To Go drive without using Rufus at all, you will have the same issue, which means that this is indeed a pure Microsoft issue.

I'm afraid I am not planning to look into working around this issue, and at this stage, I'm not seeing enough repeated questions about this to warrant a FAQ entry, so I will close this issue.

closed 2023-02-15 by pbatard

syneoa, 2023-03-29:
Happy to report some success for anyone that may stumble to this thread looking for same answer:

I have used latest Rufus which is v3.22p at the time of this writing and LTSC version of Windows 10, en-us_windows_10_enterprise_ltsc_2021_x64_dvd_d289cf96.iso with hash info C90A6DF8997BF49E56B9673982F3E80745058723A707AEF8F22998AE6479597D

Updates works! So now I have Windows To Go with latest March/23 update installed up and running on a usb m2 ssd and will receive updates up until the end of support for this release which will be in Jan 12, 2027. Fantastic!

venkeythemonkey, 2023-04-29:
hi @syneoa , Thanks for commenting that LTSC works fine. I will install the same. Just wondering how is the performance of ssd when its plugged through USB3?