pbatard/rufus #2173: Samsung SSD 970 EVO Plus USB Device - Report as per log

bigcrimping, 2023-02-16:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Reporting a HD type not picked up by the app, log says to report it.

Log
---
```
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19045.2486)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Samsung SSD 970 EVO Plus USB Device' (0BDA:9210)
NOTE: This device is a USB 3.1 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 28 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'Samsung SSD 970 EVO Plus USB Device' (0BDA:9210)
NOTE: This device is a USB 3.1 device operating at lower speed...
Device eliminated because it was detected as a Hard Drive (score 28 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
```

pbatard, 2023-02-16:
In the context of Rufus, this should be detected as a "Hard Drive" (Hard Drives including HDDs as well as SSDs and anything of a large enough size with the FIXED disk attribute).

It would be way too dangerous to start listing Samsung SSDs by default in Rufus.

As such, I am not planning to alter anything, and if you want to have this device listed in Rufus and use it, you will need to use the option under _Show advanced drive properties_.

closed 2023-02-16 by pbatard

bigcrimping, 2023-02-16:
Understood, I used the function show advanced drive properties I just saw the message in the log to report it and love the SW so much I would provide the feedback. Cheers for the quick response.