pbatard/rufus #2174: unable to download windows 10 iso

meo1412, 2023-02-17:
As noted by the title, i try to use rufus to download an older Iso to fix my version, but it stumbled in that last bit with error in underlying connection. Here is the logfile:
Rufus x86 v3.21.1949 (Portable)
Windows version: Windows 10 Enterprise, 64-bit (Build 17134.2208)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'ASMT 2235 USB Device' (174C:235C)
Device eliminated because it was detected as a Hard Drive (score 6 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Downloading https://github.com/pbatard/Fido/releases/download/v1.43/Fido.ps1.lzma
File length: 18.1 KB
Successfully downloaded 'Fido.ps1.lzma'
Download signature is valid ✓
Script signature is valid ✓
Please Wait...
Querying https://www.microsoft.com/en-US/software-download/
Querying https://vlscppe.microsoft.com/tags?org_id=y6jn8c31&session_id=9c9ae28f-48f3-4920-8806-5aa497c6fcf9
Error: The underlying connection was closed: An unexpected error occurred on a send.
Exited download script with code: 3
Saved C:\Users\username\Desktop\rufus.log

pbatard, 2023-02-17:
This looks a bit similar to #2133. However, whereas #2133 was quickly identified as an issue that had to do with trying to use an obsolete version of SSL/TLS, and that could easily be replicated on any Windows 8 or earlier machine, this one is very specific to your environment.

The fact that you are using Windows 10 Enterprise leads me to think that you may have some corporate restrictions installed with regards to the secure protocols that can be used to access SSL websites like the Microsoft one (please bear in mind that the fact that you can access them in a browser means nothing, as Rufus needs to use the native Windows/PowerShell APIs to do so, and these are a lot more limited than the ones that are handled by a modern browser).

Or it could just be that your corporate environment is restricting what PowerShell scripts can access (since Rufus uses PowerShell behind the scenes to download Windows ISOs).

Unfortunately again, without being able to replicate your issue, it's going to be very difficult to troubleshoot it, let alone address it, especially as many people are running Rufus on Windows 10 and don't have that issue)...

What I can say however is that, as opposed to https://www.microsoft.com, https://vlscppe.microsoft.com appears to use an version of the SSL/TLS protocol that is more restricted or more stringent, so I'm not entirely surprised that on nonstandard systems, this may produce errors.

Your best best then is to go to https://www.microsoft.com/en-US/software-download and try to download the latest Windows 10 ISO manually (after switching the `User-Agent` from your browser).

meo1412, 2023-02-18:
Thanks a lot for your help, in my case, this is actually my personal PC , and I was updating from windows7 ultimate and since i got the options I go with Enterprise version, I definitely modified a lot of group policy since 20h2 break a lot of my data and program for whatever reason, I will try to turn them off and see if i can use Rufus again

pbatard, 2023-02-20:
I'm going to close this issue for now.

closed 2023-02-20 by pbatard