pbatard/rufus #2178: Add remove restrictions about unsupported CPUs for Windows 11

seba2282, 2023-02-20:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

Hello, I only wanted give suggestion because it is needed options especially for old devices.
Can you add  remove restrictions about unsupported CPUs for Windows 11 for rufus?

Best Regards

pbatard, 2023-02-20:
Not planning to, as I have yet to see a Windows 10 compatible CPU that would fail a Windows 11 clean installation.

Do you have an example of a CPU where you can clean install Windows 10 but not Windows 11?

seba2282, 2023-02-20:
I have example amd athlon II 640 x64 bit x4
Is old example but still only need reg key to skip or put install wim/esd from windows 11 to windows 10 files. But then i can't still skip Microsoft account. But that small problems, because i use him. 

But i see in new versions rufus i don't see exchanced installation windows 11. Only standard. In some older version Rufus i saw that option. But now not. And i don't know it is bug only to me or because i use old motherboard what have only bios legacy. 
Sorry for more tell under topic here.

pbatard, 2023-02-20:
> i see in new versions rufus i don't see exchanced installation windows 11

That's because you haven't seen [this from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_I_no_longer_see_the_option_to_bypass_TPMSecure_Boot_with_Windows_11). The options have been moved.

Also please test a _clean_ install on athlon 640 rather than an in-place upgrade, as the in-place upgrade is a lot more restrictive than clean install.

At this stage, I'm still not convinced I want to do anything for old CPUs.

seba2282, 2023-02-20:
Then how work media creation tool bat what made one user on GitHub? He make iso with all restrictions out for windows 11. I see Rufus has options to remove all restrictions without CPU restrictions. And interesting fact. Unpacked iso windows 10 and mount iso windows 11. I copied install wim/esd from windows 11 to windows 10 and from files on disc i made iso bootable by imgburn and all work but original you can't skip online account Microsoft how is putted by Microsoft. But install work without problems. 

About that options to moved what interesting me in Rufus, because i wanted make pendrive from original iso i try find in informations what you gave. :)

seba2282, 2023-02-20:
If i understood good, i can other restrictions to put remove after put start button too? Mean remove Microsoft accounts, remove that ram and etc? Sorry my asks. But it is few new for me.

pbatard, 2023-02-20:
> Then how work media creation tool bat what made one user on GitHub?

This is not a matter of whether it's possible to add the feature but whether the developer of the software thinks it's worth adding it.

There are plenty of similar features I could add to Rufus but that I have chosen not to, because I want to keep the scope focused and I estimate that in the long run adding said features will be more problematic for users than it will be beneficial. Considering that your is the first example I get of a CPU that allegedly will be rejected from a clean install of Windows 11, but not from Windows 10, and that, as I struggle with trying to get rid of the Windows 7 legacy, I am well placed to know that trying to support hardware that has been declared "obsolete" by Microsoft way past its expiration date is **never** a good investment, regardless of that _others_ choose to do in their software, I am not really seeing any incentive to allow people to install Windows 11 on CPUs that Microsoft has declared obsolete in Rufus.

In short, there are some decisions from Microsoft I disagree with (requirement for Secure Boot or TPM and, in some cases, a minimum amount of RAM) and there are some decisions from Microsoft I strongly agree with (requirement for a CPU that is not too old, or having a limited shelf life for each version of Windows, regardless of how many people want to continue using it).

>  i can other restrictions to put remove after put start button too?

Yes. Just try it.

seba2282, 2023-02-20:
Thanks for help. And i hope in some new version Rufus you look positive on my suggestion. Especially if much people report to Microsoft how example on old Intel processors works fine all if about Windows 11. And Microsoft ignore only, because is only interests with producents hardwares and dollars must agree and that is true. And believe me, more people will find rescue in example Rufus or combine in other style how unlock to can install Windows 11. In clean install i need i think add some key for cpu for can install. So i think it was perfect if Rufus made remove and that restrictions. 
One more thank you for help.

pbatard, 2023-02-20:
I'll consider it if I get more verifiable reports of CPUs, but for now, it doesn't seem that this is an issue for enough users and again, I don't think trying to support old CPUs is actually beneficial for end users in the long run.

I will close this issue for now.

closed 2023-02-20 by pbatard

seba2282, 2023-02-21:
So I let's me tell. I made replace install.wim/esd from windows 11 to windows 10 files source. I made iso from that files bootable. I made by Rufus with options for windows 11 only remove restrictions on account Microsoft when install. And work without all restrictions (cpu too is removed by trick with that how was made iso by me). So added no check CPU is good idea for future version Rufus. So please consider that seriously. And much people will tell then thank you, especially for who won't change old pc/laptop, because work good for him/her still. 
Best Regards

pineapple63, 2023-02-21:
Unless your computer has a single core CPU, I can’t see how swapping the Windows 10 install.wim/esd for a Windows 11 install.wim/esd would do anything that the SB/TPM/RAM bypass option in rufus doesn’t do

i have verified (on multiple computers) that the Windows 11 (offline) installer does NOT check that the CPU model is supported
(On the computers i used to test this, they do not meet the requirements to run Windows 11 (Unsupported CPU model), however, i can install Windows 11 using installation media without applying any bypasses)

seba2282, 2023-02-21:
https://beebom.com/how-bypass-windows-11-cpu-tpm-secure-boot-ram-online-account-requirements/

After step 6 i can please about your comment? 
It is about cpu to bypass. The same upgrade from normal turned pc. Need key. 
Replace install.wim / esd in windows 10 by install.wim/esd make resolve problems, but... That don't make skip account Microsoft and here is still need remove example by use Rufus. About what i told.

pineapple63, 2023-02-21:
Shortly after i wrote my previous comment, i started wondering if the problem might be that it is a legacy BIOS system (which would explain why swapping the install.win/esd file onto a Windows 10 installer works)
A pre-2010 CPU indicates that it could potentially be a legacy BIOS system

I am not sure if a bypass for legacy BIOS systems would be wise to implement into Rufus, since those systems would be quite old now, and there is no guarantee that Microsoft won’t change something that breaks Windows 11 running on legacy BIOS systems (tomorrow, in 6 months, or in 5 years time)

seba2282, 2023-02-21:
Microsoft don't let's because dollars needs agree. Much people after forums find put out all restrictions from windows 11 and to install on seriously old devices. In telling true, we both know, much people don't change what work good. And more people will find ways on installation windows 11. Are and official methods to put out and that watermark what come after moment to users who haven't supported devices. So seriously. 
It is good topic because i am first from 3/4 people who use old devices and i tell about possible to let's us install windows 11 too. Microsoft... He will feel after few times what he made if he will make what make. Much people want windows 11 officially and they report here work, here work. And Microsoft only ignore.

seba2282, 2023-02-21:
> I am not sure if a bypass for legacy BIOS systems would be wise to implement into Rufus

I take bios mbr without cms or how tell ntfs default + that patch for old bioses as sector bonus and start and remove restrictions and work too. But without cpu if clean by Rufus on only iso with windows 11. So clean windows 11 need only that reg CPU later. So why problems remove restrictions CPU too?

pineapple63, 2023-02-21:
Although i do agree that dropping support for older hardware was (at least partially) incentivised by money, i think there are possibly also (some) security reasons for this as well, as older hardware generally lacks a lot of security features that more modern hardware has, and older hardware also may have security vulnerabilities in it, which could be dangerous if not patched in software (assuming it is even possible to patch them in software)

A lot of people also have unrealistic expectations as to how long computers should last (realistically computers shouldn’t be expected to last more than about 5 to 8 years)
And by how long a computer should last, I’m referring to how long it should ideally be in service until it is replaced

I also consider it unreasonable to expect hardware to be supported forever, and although i may not agree with some of the decisions Microsoft made, i do agree with their decision to drop support for much older hardware, in fact, with some of the system requirements, i think they actually set them a bit too low (storage is one of them)

seba2282, 2023-02-22:
But we tell true. You make program, which use much people. Because is easy make bootable USB then. And is hot topic about removing all restrictions which are absurdly made by Microsoft. We know good too. Much people won't change their PCs/laptops if work still good. And the don't go fast after new technology, which have missions make general closed all. And much people wanted windows 11, especially if are tests on more old devices by people. So why to don't give people all complete to can install that Windows 11? I was seriously wonder over give that. Who has compatible device then good. But who haven't. In telling true. Windows 11 works better than windows 10 on my old PC. So i don't know when Microsoft go after mind to mind.