pbatard/rufus #2179: Anaconda workaround not applied to BIOS grub.cfg

marcosfrm, 2023-02-20:
'inst.stage2' ➔ 'inst.repo' applied  only to UEFI grub.cfg. Fedora 37.

```
Rufus x86 v3.21.1949
Windows version: Windows 11 Pro, 64-bit (Build 22621.1265)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.06-fedora-nonstandard
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Fedora-S-dvd-x86_64-37'
  Size: 654.6 MB (Projected)
  Uses: GRUB2
  Uses: EFI
Using image: Fedora-Server-netinst-x86_64-37-1.7.iso (665.1 MB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Downloading https://rufus.ie/files/grub-2.06-fedora-nonstandard/core.img
File length: 37.2 KB
Successfully downloaded 'core.img'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.7 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Opened \\.\PhysicalDrive1 for exclusive write access
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
Using Grub 2.0 MBR
Writing Grub 2.0 SBR (from download) ✓
Found volume \\?\Volume{a51e666f-b134-11ed-bd29-5cc9d31753dd}\
Opened \\?\Volume{a51e666f-b134-11ed-bd29-5cc9d31753dd} for exclusive write access
Writing Partition Boot Record...
Using Standard FAT32 partition boot record
Confirmed new volume has a primary FAT32 boot sector
Setting primary FAT32 boot sector for boot...
Confirmed new volume has a secondary FAT32 boot sector
Setting secondary FAT32 boot sector for boot...
Successfully remounted \\?\Volume{a51e666f-b134-11ed-bd29-5cc9d31753dd}\ as E:
Extracting files...
[... 314 lines cut ...]
Extracting: E:\boot\grub2\i386-pc\xnu_uuid_test.mod (1.9 KB)
Extracting: E:\boot\grub2\i386-pc\xzio.mod (25.0 KB)
Extracting: E:\boot\grub2\i386-pc\zfs.mod (55.2 KB)
Extracting: E:\boot\grub2\i386-pc\zfscrypt.mod (6.7 KB)
Extracting: E:\boot\grub2\i386-pc\zfsinfo.mod (7.1 KB)
Extracting: E:\boot\grub2\i386-pc\zstd.mod (96.5 KB)
Extracting: E:\images\eltorito.img (34.2 KB)
Extracting: E:\images\install.img (532.6 MB)
Extracting: E:\images\pxeboot\initrd.img (97.1 MB)
Extracting: E:\images\pxeboot\vmlinuz (12.1 MB)
Finalizing, please wait...

Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x000BB512
Drive has a Grub 2.0 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2023-02-20:
[image]

I may be wrong on this but please look at `boot/grub2/grub.cfg` and let me know if you see a `inst.stage2` string we can patch...

marcosfrm, 2023-02-20:
`\boot\grub2\grub.cfg`:

```
set default="1"

function load_video {
  insmod all_video
}

load_video
set gfxpayload=keep
insmod gzio
insmod part_gpt
insmod ext2
insmod chain

set timeout=60
### END /etc/grub.d/00_header ###

search --no-floppy --set=root -l 'F37S'

### BEGIN /etc/grub.d/10_linux ###
menuentry 'Install Fedora 37' --class fedora --class gnu-linux --class gnu --class os {
	linux /images/pxeboot/vmlinuz inst.stage2=hd:LABEL=F37S quiet
	initrd /images/pxeboot/initrd.img
}
menuentry 'Test this media & install Fedora 37' --class fedora --class gnu-linux --class gnu --class os {
	linux /images/pxeboot/vmlinuz inst.stage2=hd:LABEL=F37S rd.live.check quiet
	initrd /images/pxeboot/initrd.img
}
submenu 'Troubleshooting -->' {
	menuentry 'Install Fedora 37 in basic graphics mode' --class fedora --class gnu-linux --class gnu --class os {
		linux /images/pxeboot/vmlinuz inst.stage2=hd:LABEL=F37S nomodeset quiet
		initrd /images/pxeboot/initrd.img
	}
	menuentry 'Rescue a Fedora system' --class fedora --class gnu-linux --class gnu --class os {
		linux /images/pxeboot/vmlinuz inst.stage2=hd:LABEL=F37S inst.rescue quiet
		initrd /images/pxeboot/initrd.img
	}
	menuentry 'Boot first drive' --class fedora --class gnu-linux --class gnu --class os {
		chainloader (hd0)+1
	}
	menuentry 'Boot second drive' --class fedora --class gnu-linux --class gnu --class os {
		chainloader (hd1)+1
	}
}
```

`\EFI\BOOT\grub.cfg`:

```
set default="1"

function load_video {
  insmod efi_gop
  insmod efi_uga
  insmod video_bochs
  insmod video_cirrus
  insmod all_video
}

load_video
set gfxpayload=keep
insmod gzio
insmod part_gpt
insmod ext2

set timeout=60
### END /etc/grub.d/00_header ###

search --no-floppy --set=root -l 'F37S'

### BEGIN /etc/grub.d/10_linux ###
menuentry 'Install Fedora 37' --class fedora --class gnu-linux --class gnu --class os {
	linuxefi /images/pxeboot/vmlinuz inst.repo=hd:LABEL=F37S quiet
	initrdefi /images/pxeboot/initrd.img
}
menuentry 'Test this media & install Fedora 37' --class fedora --class gnu-linux --class gnu --class os {
	linuxefi /images/pxeboot/vmlinuz inst.repo=hd:LABEL=F37S rd.live.check quiet
	initrdefi /images/pxeboot/initrd.img
}
submenu 'Troubleshooting -->' {
	menuentry 'Install Fedora 37 in basic graphics mode' --class fedora --class gnu-linux --class gnu --class os {
		linuxefi /images/pxeboot/vmlinuz inst.repo=hd:LABEL=F37S nomodeset quiet
		initrdefi /images/pxeboot/initrd.img
	}
	menuentry 'Rescue a Fedora system' --class fedora --class gnu-linux --class gnu --class os {
		linuxefi /images/pxeboot/vmlinuz inst.repo=hd:LABEL=F37S inst.rescue quiet
		initrdefi /images/pxeboot/initrd.img
	}
}
```

pbatard, 2023-02-20:
Interestingly I saw no `inst.stage2` in `/boot/grub2/grub.cfg` from `Fedora-LXDE-Live-x86_64-37_Beta-1.5.iso` (but I have them in `/EFI/BOOT/grub.cfg`):

```
set default="1"

function load_video {
  insmod all_video
}

load_video
set gfxpayload=keep
insmod gzio
insmod part_gpt
insmod ext2
insmod chain

set timeout=60
### END /etc/grub.d/00_header ###

search --no-floppy --set=root -l 'Fedora-LXDE-Live-37_B-1-5'

### BEGIN /etc/grub.d/10_linux ###
menuentry 'Start Fedora-LXDE-Live 37_Beta' --class fedora --class gnu-linux --class gnu --class os {
	linux /images/pxeboot/vmlinuz root=live:CDLABEL=Fedora-LXDE-Live-37_B-1-5  rd.live.image quiet rhgb
	initrd /images/pxeboot/initrd.img
}
menuentry 'Test this media & start Fedora-LXDE-Live 37_Beta' --class fedora --class gnu-linux --class gnu --class os {
	linux /images/pxeboot/vmlinuz root=live:CDLABEL=Fedora-LXDE-Live-37_B-1-5  rd.live.image rd.live.check quiet
	initrd /images/pxeboot/initrd.img
}
submenu 'Troubleshooting -->' {
	menuentry 'Start Fedora-LXDE-Live 37_Beta in basic graphics mode' --class fedora --class gnu-linux --class gnu --class os {
		linux /images/pxeboot/vmlinuz root=live:CDLABEL=Fedora-LXDE-Live-37_B-1-5  rd.live.image nomodeset quiet rhgb
		initrd /images/pxeboot/initrd.img
	}
	menuentry 'Boot first drive' --class fedora --class gnu-linux --class gnu --class os {
		chainloader (hd0)+1
	}
	menuentry 'Boot second drive' --class fedora --class gnu-linux --class gnu --class os {
		chainloader (hd1)+1
	}
}
```

Let me get my hands on the actual release you use, because it's super surprising that the precise issue we're having trouble with is with something that Fedora would have added between BETA and release (i.e. these odds look suspicious).

marcosfrm, 2023-02-20:
The non-live media used by the Server variant has it.

The funny thing is: for the **netinstall** media, the workaround will break the installer. For the regular media, like `Fedora-Server-dvd-x86_64-37-1.7.iso`, or all "minimal" RHEL derivatives (like `Rocky-9.1-x86_64-minimal.iso`, `AlmaLinux-9.1-x86_64-minimal.iso`, ...), it will fix it.

These RHEL "minimal" ISOs are very popular. Hence, the workaround is beneficial after all. Good news, btw: Anaconda folks are aiming Fedora 39 to get this mess fixed: https://github.com/rhinstaller/anaconda/pull/4567.

pbatard, 2023-02-20:
> The non-live media used by the Server variant has it.

Yeah, I can see that `inst.stage2` for BIOS seems to be a netinst-only thing, because `Fedora-Workstation-Live-x86_64-37-1.7.iso` is in line with the BETA I tested and doesn't have any instance of it for BIOS.

Obviously, because I never expected netinst to be different from regular, especially when it's an option that effectively points back to the boot media (**not** the remote netinst repo), and I tried to fix breakage before the formal release of Fedora 37, I never tested with netinstall to see that Fedora had introduced a major difference there.

Oh, and the issue is that, because I only saw it as required for EFI, we only look for a line that starts with `linuxefi` in `grub.cfg`, which of course will leave BIOS unpatched

> The funny thing is: for the **netinstall** media, the workaround will break the installer. (...) Hence, the workaround is beneficial after all.

I take this to mean that Rufus is somehow doing the right thing by not patching it then?

> btw: Anaconda folks are aiming Fedora 39 to get this mess fixed: rhinstaller/anaconda#4567.

Well, it's not like, as you know, the Red Hat folks hadn't been explicitly warned that oughta take a closer look at fixing this in rhinstaller/anaconda#3529, instead of trying to shift the blame to external apps (or expect them to add workarounds for their mess), and did promise that they would _"definitely take a look"_ at it before...

I guess I'll have to plan for more BETA testing of Fedora 39, as well as testing a bunch of different ISOs from them, since they can't seem to harmonize their boot options and, as I suspected, the workaround that I need to apply in Rufus to work around this mess is continuing to be incredibly brittle (with the result that end-users failing to install RH/Fedora/CentOS will continue to blame Rufus for their woes, when it's really RH that is the cause of their issue).

pbatard, 2023-02-20:
Also, before I try to allocate time to test this, when you say that _"for the **netinstall** media, the workaround will break the installer"_ I'm hoping/assuming that this is for BIOS-only.

Or did you test EFI netinst with the `inst.stage2`&rarr;`inst.repo` workaround applied, and found that it would break install?

marcosfrm, 2023-02-20:
> > The funny thing is: for the **netinstall** media, the workaround will break the installer. (...) Hence, the workaround is beneficial after all.
> 
> I take this to mean that Rufus is somehow doing the right thing by not patching it then?

Nope, because I was wrong. I tested again `Fedora-Server-netinst-x86_64-37-1.7.iso` (in UEFI) and it works with `inst.repo`. Not sure how I assumed it did not work... sorry for the noise. Lack of coffee or whatever.

The difference is that the netinstall media does work with `inst.stage2` **too**, while `Fedora-Server-dvd-x86_64-37-1.7.iso` and all "minimal" RHEL derivatives **only** work with `inst.repo`. So Rufus has to patch the thing still, BIOS and UEFI.

I will light a candle for Anaconda devs here. 🙏

pbatard, 2023-02-20:
> The difference is that the netinstall media does work with `inst.stage2` **too**, while `Fedora-Server-dvd-x86_64-37-1.7.iso` and all "minimal" RHEL derivatives **only** work with `inst.repo`. So Rufus has to patch the thing still, BIOS and UEFI.

So I guess what we're doing right now is fine, since we do patch `Fedora-Server-dvd-x86_64-37-1.7.iso` where we can (i.e. where `grub.cfg` uses `inst.stage2`) whereas, if either `inst.stage2` or `inst.repo` works for netinstall, it doesn't matter if we only patch zero, one or two...

marcosfrm, 2023-02-21:
> Oh, and the issue is that, because I only saw it as required for EFI, we only look for a line that starts with `linuxefi` in `grub.cfg`, which of course will leave BIOS unpatched

Before F37, Syslinux was used for BIOS and the code patched it by searching for `append`, no?

https://github.com/pbatard/rufus/blob/ddcbe8ed817880a56c4420dabddaebd20b2608f8/src/iso.c#L390-L394

> So I guess what we're doing right now is fine, since we do patch `Fedora-Server-dvd-x86_64-37-1.7.iso` where we can (i.e. where `grub.cfg` uses `inst.stage2`) whereas, if either `inst.stage2` or `inst.repo` works for netinstall, it doesn't matter if we only patch zero, one or two `grub.cfg`'s there

Now Syslinux is gone in F37+... it means, for BIOS boot, only the netinstall media will work. `Fedora-Server-dvd-x86_64-37-1.7.iso` and RHEL minimal will not.

pbatard, 2023-02-21:
> Before F37, Syslinux was used for BIOS and the code patched it by searching for `append`, no?

Yup, because I never saw it for non UEFI grub, and I think RH was still using Syslinux at the time.
I'm pretty sure if I had seen it used for the Fedora 37 ISOs I tested (that switched to GRUB for BIOS) I would have altered the code, but it wasn't there for the workstation images, and you don't want to break the existing Syslinux compatibility.

> it means, for BIOS boot, only the netinstall media will work. 

What's the point of having a /boot/grub2/grub.cfg if not for BIOS then?

We're installing a GRUB bootloader for BIOS (which we actually already had to install a workaround for, since Fedora are using a nonstandard BIOS GRUB location), and I'm pretty sure I did test both BIOS and UEFI GRUB boot for Fedora **Workstation** 37 BETA to makes sure that the media would at least boot in BIOS or UEFI mode (can't vouch for the ability to actually install though, which is the point of `inst.stage2` vs `inst.repo` and what I'm asking about here).

In short, I'm trying to find out, since you said that netinstall didn't care about `inst.stage2` vs `inst.repo` and (as far as I know) Rufus does setup Fedora to boot either for BIOS or UEFI even if it uses GRUB only, whether we're getting an issue later down the line...

I guess I'm gonna run a quick test to see what's happening with Fedora 37 netinst...

marcosfrm, 2023-02-21:
> Yup, because I never saw it for non UEFI grub, and I think RH was still using Syslinux at the time. I'm pretty sure if I had seen it used for the Fedora 37 ISOs I tested (that switched to GRUB for [...]

I mean only the netinstall media will work _without patching_.

Let's see if we are in the same page: F37+ uses GRUB for BIOS. Since `/boot/grub2/grub.cfg` is not patched today (because the current code expects `linuxefi` only), for BIOS boot `inst.stage2` will be used. This is only OK for the netinstall media. The `-dvd` and RHEL `-minimal` ISOs do not work with `inst.stage2`.

pbatard, 2023-02-21:
Well, BIOS boot of F37 netinst seems to be fine and appears to find the installation source from the network, so this appears to indicate that, indeed, netinst doesn't care about `inst.stage2` vs `inst.repo`  at least for BIOS, since this is the option that would cause the _Installation Source_ to fail for Workstation if left unpacthed.

On the other hand UEFI boot of F37 netinst yields `Error setting up the repository` which seems to indicate that, as you hinted earlier, F37 netinstall should actually use `inst.stage2` and **not** `inst.repo`.

What a complete mess this is! Depending on whether you're installing netinst or workstation, you need to use either `inst.stage2` or `inst.repo`. So now I need an even more brittle workaround to detect if we have a RH netinst or local install...

pbatard, 2023-02-21:
> I mean only the netinstall media will work without patching.

From what I can see, the netinstall media **requires** the `grub.cfg` not to be patched. But the non netinstall media requires it to be.

marcosfrm, 2023-02-21:
> On the other hand UEFI boot of F37 netinst yields `Error setting up the repository` which seems to indicate that, as you hinted earlier, F37 netinstall should actually use `inst.stage2` and **not** `inst.repo`.
> 

Yep, but if you open the "Installation source" section and select "On the network" or so it will work.

pbatard, 2023-02-21:
> but if you open the "Installation source" section and select "On the network" or so it will work.

My issue with this is that it works from the get go with BIOS and `inst.stage2`, and if I _unpatch_ UEFI, then it also works by default.

Again, I can't expect Fedora users to run into an error from the get go in ISO mode, which they won't get in DD mode and not to conclude that it's Rufus' fault (when it's really Red Hat having created a complete shit show with their `inst.*` options)

marcosfrm, 2023-02-21:
IMHO the non-netinstall ISOs are far more important.

marcosfrm, 2023-02-21:
> So now I need an even more brittle workaround to detect if we have a RH netinst or local install...

There must be some directory indicating a local repository in the ISO.

pbatard, 2023-02-21:
> IMHO the non-netinstall ISOs are far more important.

AFAIK, these should work as expected, but Workstation requires to sacrifice a disk to actually get to the source repo, whereas netinstall doesn't, so all I could test is was that the Workstation installer didn't complain about not finding a source, which I guess means that **patched** BIOS and UEFI Workstation "should" be OK.

> There must be some directory indicating a local repository in the ISO.

My problem isn't that we won't be able figure a workaround for current ISOs. My problem is that, this is likely to change with subsequent releases and break everything, hence "brittle". Having to rely on an ISO structure to be a specific way to detect X vs Y is an approach that is best avoided, as it always leads to massive issues in the long run... And there's all the RH derivatives that we should ideally test.

marcosfrm, 2023-02-21:
> > IMHO the non-netinstall ISOs are far more important.
> 
> AFAIK, these should work as expected

The live ISOs use `root=live:CDLABEL`, not `inst.stage2`. No patching there.

Patch always then. Netinstall will require intervention (adjust "Installation source") until they fix Anaconda mess. Every other media will work.

https://github.com/pbatard/rufus/blob/ddcbe8ed817880a56c4420dabddaebd20b2608f8/src/rufus.c#L1255-L1266

If they deliver, once F39 is released (and assuming RHEL 10 gets the fix), the issue will disappear by itself.