pbatard/rufus #2182: Writing a Windows 2022 LTSC ISO with ExFAT works; NTFS should, but does not

doctorpangloss, 2023-02-21:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [X] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [X] Plugging the USB into a different port.
  - [X] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Thanks for taking a look at this issue. Trying to write the April 2022 update to Windows 2022 LTSC using NTFS causes the USB key to boot into a command line with the single command `wpeinit` already written instead of Setup, and running `X:\sources\setup.exe` causes the error:

```
Windows could not load required file winsetup.dll. The file may be corrupt. To install windows, restart the installation. Error code 0xC1.
```
with an Access Denied exception.

Manually creating the image using ExFAT (see https://gist.github.com/pavankjadda/0b89ad88ee6968aa292b24f3e2efc3f7) works.

It isn't clear to me what the underlying issue is. There is nothing abnormal in the logs, all the files were successfully copied.

Also tested on 2.18p and 3.20p, no change.

Log
---
```
(gathering on another machine, your patience is appreciated)
```

pbatard, 2023-02-22:
>  Trying to write the April 2022 update to Windows 2022 LTSC

Please provide the SHA-1 of the ISO you use to see if it's an official ISO that was published by Microsoft.

I hope you can appreciate that I can only support official ISOs that have been published by Microsoft. Anything else, that has been mastered in a dodgy unofficial manner, is unsupported, **precisely** because it can lead to exactly the issues you are facing.

So, unless I get the SHA-1 of your image (which you can produce in Rufus by clicking the `(✓)` __as you were advised to do in the check list__, I will have to consider that you are using an unofficial image and dismiss your request for support.

doctorpangloss, 2023-02-22:
> SHA-1 of the ISO you use to see if it's an official ISO that was published by Microsoft.

Doesn't look like it is, thanks!

closed 2023-02-22 by doctorpangloss