pbatard/rufus #2183: GRUB4DOS menu.lst not patched

marcosfrm, 2023-02-22:
Test ISO:

```
$ ls -l iso/
total 320
-rw-r--r--. 1 marcos marcos 319541 jul 18  2022 grldr
-rw-r--r--. 1 marcos marcos     77 fev 21 20:20 menu.lst

$ cat iso/menu.lst
timeout 90

title Test
kernel /foo/vmlinuz LABEL=TEST123
initrd /foo/bar.img

$ xorriso -as mkisofs -r -J -V TEST123 --boot-catalog-hide -b grldr -no-emul-boot -boot-load-size 4 -no-emul-boot -o test.iso iso
xorriso 1.5.4 : RockRidge filesystem manipulator, libburnia project.

Drive current: -outdev 'stdio:test.iso'
Media current: stdio file, overwriteable
Media status : is blank
Media summary: 0 sessions, 0 data blocks, 0 data, 7095m free
Added to ISO image: directory '/'='/home/marcos/build/iso'
xorriso : UPDATE :       2 files added in 1 seconds
xorriso : UPDATE :       2 files added in 1 seconds
ISO image produced: 342 sectors
Written to medium : 342 sectors at LBA 0
Writing to 'stdio:test.iso' completed successfully.
```

[image]

```
Rufus x86 v3.21.1949
Windows version: Windows 10 Pro, 64-bit (Build 19045.2604)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'TEST123'
  Size: 316 KB (Projected)
  Uses: Grub4DOS
Using image: test.iso (684 KB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.7 GB)
Waiting for logical drive to reappear...
Formatting to FAT (using IFS)
Using cluster size: 65536 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT16 LBA...
Set bootable USB partition as 0x80
Using Grub4DOS MBR
Writing Grub4Dos SBR
Found volume \\?\Volume{68ccf167-b18f-11ed-a1bf-dc0ea1ca85f9}\
Opened \\?\Volume{68ccf167-b18f-11ed-a1bf-dc0ea1ca85f9} for exclusive write access
Writing Partition Boot Record...
Using Standard FAT16 partition boot record
Confirmed new volume has a FAT16 boot sector
Successfully remounted \\?\Volume{68ccf167-b18f-11ed-a1bf-dc0ea1ca85f9}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: E:\grldr (312.1 KB)
Extracting: E:\menu.lst (77 bytes)
Finalizing, please wait...

Found USB 2.0 device 'Patriot Memory USB Device' (13FE:3600)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 486, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0040FFD4
Drive has a Grub4DOS Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Detected File System: FAT16
  Size: 3.7 GB (3998220288 bytes)
  Start Sector: 2048, Boot: Yes
```

`kernel /foo/vmlinuz LABEL=TEST123` is still there in `menu.lst` (new filesystem label is `NEW123`).

Feel free to ignore, as I think only I use GRUB4DOS in this way...

https://github.com/chenall/grub4dos/blob/0.4.6a/README_GRUB4DOS.txt#L237

pbatard, 2023-02-22:
Yeah, that's because I don't have `menu.lst` in the list of config files I detect for patching.

The reason for that is that I have not seen a release of an ISO that was based on Grub4DOS and that would require the label patch (but I can't say I looked very hard for those).

I guess I could add `menu.lst` detection, but I'd rather do so after seeing public Grub4DOS ISOs that use a `menu.lst` with a `LABEL` that needs fixing by default (i.e. a `LABEL` that is not FAT32 compatible), otherwise, it'd be adding a feature that, while theoretically useful, going to be limited to a very small number of people, since you'd have to forcefully change the label to be affected, and regular joes aren't expected to do that.

Or do you have a real life scenario where this was actually problematic for you? Because, while I acknowledge that users may run into the issue you describe, the scenario you describe still seems more theoretical than anything at the moment...

pbatard, 2023-03-18:
I've now added `menu.lst` patching, which you'll find in the BETA I'm about to release, but this is 100% untested.

closed 2023-03-18 by pbatard

marcosfrm, 2023-03-20:
Does not work in rufus-3.22_BETA:

```
Rufus x86 v3.22.2003 (Beta)
Windows version: Windows 10 Pro, 64-bit (Build 19045.2728)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0416 (pt-BR)
Will use default UI locale 0x0416
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'pt-BR'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 4.1GB, Sector size: 512 bytes
Cylinders: 493, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'TEST123'
  Size: 318 KB (Projected)
  Uses: Grub4DOS
Using image: test.iso (682 KB)
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.8 GB)
Waiting for logical drive to reappear...
Formatting to FAT (using IFS)
Using cluster size: 65536 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT16 LBA...
Set bootable USB partition as 0x80
Using Grub4DOS MBR
Writing Grub4Dos SBR
Found volume \\?\Volume{6d91b40a-c70f-11ed-a896-dc0ea1ca85f9}\
Opened \\?\Volume{6d91b40a-c70f-11ed-a896-dc0ea1ca85f9} for exclusive write access
Writing Partition Boot Record...
Using Standard FAT16 partition boot record
Confirmed new volume has a FAT16 boot sector
Successfully remounted \\?\Volume{6d91b40a-c70f-11ed-a896-dc0ea1ca85f9}\ as E:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: E:\boot.catalog (2 KB)
Extracting: E:\grldr (313.3 KB)
Extracting: E:\menu.lst (90 bytes)
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6545)
Using 'autorun.inf' label for drive E: 'NEW123'
1 device found
Disk type: Removable, Disk size: 4.1GB, Sector size: 512 bytes
Cylinders: 493, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0015BD10
Drive has a Grub4DOS Master Boot Record
Partition 1:
  Type: FAT16 LBA (0x0e)
  Detected File System: FAT16
  Size: 3.8 GB (4062183424 bytes)
  Start Sector: 2048, Boot: Yes
```

(`boot.catalog` is present because I used `genisoimage` this time; it does not appear to have `--boot-catalog-hide` option)

Slight new `menu.lst`:

```
timeout 90

title TEST123 Test
kernel /foo/vmlinuz LABEL=TEST123
initrd /foo/bar.img
```

It was not patched (`TEST123` -> `NEW123`).

pbatard, 2023-03-22:
Now reverted in 365449fcce7ea693feffd53dbd0e6ad9b6059068.

commit 737db33f (2023-03-18): [iso] enable menu.lst patching

* Closes #2183
* Also improve safe_strdup() and "update" Grub4DOS to latest
  (NB: grldr.mbr has not changed in the last few years)