pbatard/rufus #2187: feature/ability recomendation: Ignore waht windows sais about active internet connection.

dm8mydog, 2023-02-24:
Checking for Rufus updates...
Updates: Unable to connect to the internet
Downloading https://rufus.ie/files/syslinux-6.03/20171017/ldlinux.sys
Could not open Internet session: Internet is disconnected.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.sys
Could not open Internet session: Internet is disconnected.
Could not download the file - cancelling

Hi, I know this might be a minor issue overall, but hear me out: For my own personal experience: Windows 10 has always concidered itself to be offline. And yes I'm typing this from the same computer that considers itself to be offline. so.. yeah. Windows Lied :)
Please consider not relying on whatever windows system function that checks if there is an active internet connection and instead actually checking it instead. I am aware this is likely due to the weird workplace network setup we have at my current location, but all actual network activity works regardless of what windows itself thinks.
Appologies for taking your time about this really bening random issue.
Cheers

pbatard, 2023-02-24:
Please provide the full log. And please provide data on how you are connected to the internet (e.g. through a proxy, ref of your network adapter).

I hope you can appreciate that it is **IMPOSSIBLE** to look into your issue unless you provide enough data to replicate it. And I'm afraid there's way too much data that's missing here.

dm8mydog, 2023-02-24:
I've already sorted my issue, this was a feature request.
   : Please consider not relying on whatever windows system function that checks if there is an active internet connection and instead actually checking it instead.
cheers

closed 2023-02-24 by dm8mydog

pbatard, 2023-02-24:
> : Please consider not relying on whatever windows system function that checks if there is an active internet connection and instead actually checking it instead.

Again, to be able to do that, I need more data. As I have never experienced this issue, I can't "fix" it by stabbing in the dark and expecting a hit.

You seem to be under the impression that it's common knowledge that there's a "wrong" and "right" Windows API to check if a connection is active, but I'm afraid that this is not the case, and that, as with anything, without being able to replicate the issue, a developer can't just "guess" what they should do differently from a simple _"there must be a better way"_ request.

So, again, if you want your issue addressed you need to provide enough data to address it, which you haven't done at all so far. It's as simple as this.

pbatard, 2023-02-24:
By the way, [we already use two different methods to check if there's an active internet connection](https://github.com/pbatard/rufus/blob/master/src/net.c#L287-L308), precisely because someone came to us and was able to give us enough data to replicate the issue they had (with DHCP not being instantiated) so that we could address it. Therefore, if we need to add a 3rd way to check if there's an active connection, I will expect the same process to happen, with enough data being provided to __understand__ what the underlying issue is.