pbatard/rufus #2188: bypass windows 11 network connection

sarmad84, 2023-02-28:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I wanted to request to add bypassing windows 11 network connection capability.
We can do this by using the Shift + F10 keyboard shortcut and typing OOBE\BYPASSNRO command in Command Prompt, but that would be wonderful if you can make it automatic (like adding a local account).

Log
---
```
<FULL LOG>
```

pineapple63, 2023-02-28:
For Windows 11 22H2, Rufus already has that capability (although you do have to click “I don’t have internet”)

In terms of making it where the prompt to connect is not displayed at all, a different method would be needed (although this can be done by using the HideWirelessSetupInOOBE and HideOnlineAccountScreens entries in an unattend file, athough doing this may cause errors to pop up on computers with slower CPUs)

pbatard, 2023-02-28:
The _Remove requirement for an online Microsoft account_ **already** does what you want, as it [creates the BypassNRO key](https://github.com/pbatard/rufus/blob/master/src/wue.c#L119-L125).

I suggest you play with the options, because what you request is unneeded and I am not planning to add an extra option when all you need to do is, just like 21H2, make sure that a network cable is not plugged in.

The Windows User Experience dialog is intended for elements that are difficult for users to deal with. Skipping the network connection prompts, once you have _Remove requirement for an online Microsoft account_ checked, is not one of these.

closed 2023-02-28 by pbatard