pbatard/rufus #2190: Windows 11 extraction failure

Aayushman210310, 2023-03-01:
I have downloaded Windows 11 iso from microsoft official website and when I tried to flash on my usb pendrive it shows "iso image extraction failure"
[image]

JonnyTech, 2023-03-01:
Please complete the new issue checklist, especially the following:

```
If you are reporting an issue when trying to run Rufus, or when trying to boot a media created by Rufus, you *MUST* provide a log, 
```

No log?

```
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.
```

No verification?

pbatard, 2023-03-01:
Since you did not provide a log and did not produce the checksum, I am going to have to guess corrupted drive or corrupted ISO.

Next time, **if** you actually want people to help you, please follow what's requested of you because we can't "guess" your issue from just a screenshot.

closed 2023-03-01 by pbatard