pbatard/rufus #2193: what is this error?

Jly4, 2023-03-04:
after pressing start i get this error
[image]

pbatard, 2023-03-04:
Can't help you without the log.

I'd really like to know how you got this, because Rufus should fallback to English messages if it can't find a translation for your language, and this seems to hint that your environment is the problem if that hasn't happen (such as, it may be missing some resources or something).

Basically, this is what you should see:

[image]

Also, the fact that you get `user` instead of your account name further seems to hint that your Windows installation has been tweaked in some way.

Finally, `My_19045.2673_Full_x64_by_Flibustier.iso` is very unlikely to be an official Microsoft published Windows ISO, and I can only support **official** Windows ISOs. Please make sure that you use an official image.

But really, unless you provide your FULL log, as was requested of you, I will have to close this issue.

Jly4, 2023-03-04:
This is a custom assembly that I made in vmware, a user has already been created in it. I tried to set the English language, but the error text did not change. I think that the screenshot that you threw off is enough for me to understand what to choose there.

[rufus.log](https://github.com/pbatard/rufus/files/10888347/rufus.log)

pbatard, 2023-03-04:
```
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
```

This may be why you're seeing untranslated messages.

Your Russian version of Windows is interesting, because AFAIK, localised versions of Windows should still come with the English language pack as standard (at least I don't recall seeing that message on French or other localised version, but I need to double check).

What happens if you do [install the English language pack](https://support.microsoft.com/en-us/windows/install-a-language-for-windows-ccd853d3-9ecd-7da7-9ef0-72b4a055410a#WindowsVersion=Windows_10)?

pbatard, 2023-03-04:
Something's weird with your log compared to your screenshot though:

> `Localization set to 'en-US'`

This means that your interface in Rufus should be displayed in English, not in Russian.

Did you switch language in Rufus after you posted the screenshot?
Can your please also provide the log when using Russian for your language in Rufus?

Basically, the fact that you don't get `Localization set to 'ru-RU'` in your log is weird and does not match the screenshot you sent.

Jly4, 2023-03-04:
hmm... i already had english (US) installed

pbatard, 2023-03-04:
According to [`EnumUILanguagesW()`](https://learn.microsoft.com/en-us/windows/win32/api/winnls/nf-winnls-enumuilanguagesw), I'm afraid you don't...

Jly4, 2023-03-04:
https://i.imgur.com/ARtCCbB.png
https://i.imgur.com/N3AFV6G.png

pbatard, 2023-03-04:
I don't doubt that you somehow have the packs installed. But I can tell you that, despite that, `EnumUILanguagesW()` does not list 0x409, which is the code for _English (United States)_ as installed.

Basically, the process is very simple:
1. When running Rufus in a selected locale (here, it appears that you have selected `English (English)`) and this is not the default system locale, Rufus goes through all the Windows language IDs that are relevant to that language (the numbers at the end of [this line](https://github.com/pbatard/rufus/blob/master/res/loc/rufus.loc#L46))
2. For each of these language IDs, Rufus invokes `EnumUILanguagesW()` with parameter `MUI_LANGUAGE_ID` to see if Windows reports that the language pack for that specific ID is installed (NB: And yes, if you're looking at the code, we need to do a string comparison for what are essentially integers &mdash; The Windows APIs are that weird)
3. In your case, `EnumUILanguagesW()` does not ever see a match for `0409` which is the (hexadecimal) Windows language code for `English (United States)`, so, all I can tell is that, the Windows API that is supposed to report what language packs are installed, does not report that you have a language pack for `English (United States)` installed. And no, as far as I can tell, this is not an application bug, like, for instance, comparing `409` vs `0409` as strings. It appears to be a Windows bug.

pbatard, 2023-03-04:
Can you please download and run [`EnumUILanguages.exe`](https://files.akeo.ie/rufus/EnumUILanguages.exe) from a command prompt, and paste your output? I'd like to see what `EnumUILanguagesW()` actually reports on your system.

For reference, the source code for this commandline application is this:
```C
/*
 * base-console - Because sometimes I want to release a win32 console
 * utility in a hurry, and I like to have it set up properly.
 *
 * Copyright © 2020 Pete Batard <pete@akeo.ie>
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 */

#ifdef _DEBUG
#define _CRTDBG_MAP_ALLOC
#include <stdlib.h>
#include <crtdbg.h>
#endif

#define WIN32_LEAN_AND_MEAN
#include <windows.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <stdbool.h>

#include "msapi_utf8.h"

#define _STRINGIFY(x) #x
#define STRINGIFY(x) _STRINGIFY(x)

#ifndef APP_VERSION
#define APP_VERSION_STR "[DEV]"
#else
#define APP_VERSION_STR STRINGIFY(APP_VERSION)
#endif

static __inline char* appname(const char* path)
{
	static char appname[128];
	_splitpath_s(path, NULL, 0, NULL, 0, appname, sizeof(appname), NULL, 0);
	return appname;
}

static BOOL CALLBACK EnumUILanguagesProc(LPWSTR lpUILanguageString, LONG_PTR lParam)
{
	fprintf(stdout, "o %S\n", lpUILanguageString);
	return TRUE;
}

int main_utf8(int argc, char** argv)
{
	fprintf(stdout, "Language Packs installed on this system (seen by EnumUILanguages):\n");
	EnumUILanguages(EnumUILanguagesProc, MUI_LANGUAGE_ID, (LONG_PTR)NULL);

	return 0;
}

int wmain(int argc, wchar_t** argv16)
{
	SetConsoleOutputCP(CP_UTF8);
	char** argv = calloc(argc, sizeof(char*));
	if (argv == NULL)
		return -1;
	for (int i = 0; i < argc; i++)
		argv[i] = wchar_to_utf8(argv16[i]);
	int r = main_utf8(argc, argv);
	for (int i = 0; i < argc; i++)
		free(argv[i]);
	free(argv);
#ifdef _DEBUG
	_CrtDumpMemoryLeaks();
#endif
	return r;
}
```

Jly4, 2023-03-04:
`C:\Users\user\Desktop>C:\Users\user\Desktop\EnumUILanguages.exe
Language Packs installed on this system (seen by EnumUILanguages):
o 0419
o 0409
`

pbatard, 2023-03-04:
Thanks. Then I don't get why Rufus doesn't pick `0409` from your system.

By any chance, do you perhaps have a `rufus.loc` somewhere in the directory where you have the Rufus executable? Because that's the only way I can currently see Rufus not picking your `0409` language pack, because, if you have a `l "en-US" "English (English)"` line in that `rufus.loc` that is missing the language ID codes, that would of course replicate the issue.

For instance, if I have a `rufus.loc` along with the exe, and remove `0x0809` from it, I can get:
```
System locale ID: 0x0809 (en-GB)
NOTE: No Windows Language Pack is installed for English (English) on this system.
This means that some controls may still be displayed using the system locale.
```
But you really have to hack the loc file to get there, because otherwise, and I validated this, it will just pick the `0x0409` from the loc and therefore should have no issue matching it in the comparison call.

If not, I'm probably going to have to install a localized version of Windows somewhere to be able to investigate this further...

One last thing I may also ask you before I do that:
Can you please install and run DebugView from https://learn.microsoft.com/en-us/sysinternals/downloads/debugview and then launch Rufus?

You should see a line starting with `*** Rufus init ***` in Rufus along with some data that is relevant to localization. If you could send me a copy of that DebugView log, as it may contain further clues, I would appreciate it.

Jly4, 2023-03-04:
I understood what the problem is, I had rufus.loc and since I have been using Rufus for a long time, this file was created on 06.2022, after deleting it, I got the same window as you have in the screenshot

pbatard, 2023-03-04:
A-ha! So I wasn't going crazy! 😄 

I'm glad we could resolve this, because I really couldn't figure out how this could happen without a bug in `EnumUILanguages()` or a broken `rufus.loc`.

I guess I'll be more careful about always making sure to ask people who have localization issues if they have a `rufus.loc` somewhere.

And as a matter of fact, I think I definitely want to report if an external `.loc` is being used in the UI log (currently this is only being reported when using DebugView, because we need to pick that information early), so I'm going to keep this issue open and I will close it once I have added code in Rufus that displays if an external log is being used, in the Rufus UI log.

At any rate, thanks a lot for helping troubleshoot this issue!

Jly4, 2023-03-04:
thanks to you too for the quick help, rufus is a great program

commit d78a556e (2023-03-04): [loc] report if an external loc file is used, in the UI log

* Also avoid using LPTSTR in lieu of LPWSTR (even if they do resolve to the exact
  same thing) and force the use of app_dir when looking for a local .loc file.
* Closes #2193
closed 2023-03-04 by pbatard