pbatard/rufus #2194: [Feature request] Less popups before starting the actual writing

Deepcuts, 2023-03-04:
Hello,

Been a long time user of Rufus and I find it very useful. Thank you.
I just have one major gripe with the process of writing an ISO to USB.
Pretty please with sugar on top, give us an option to disable the reduntant and delayed popups.

For example.
I am writing my custom Windows 10 image a lot to USB.
I have zero need or use for this popup:
[image]
After I click OK, it takes several seconds to get another popup:
[image]
A lot of times I forget about the 2nd popup and just leave my PC, only to find the message when I return, without the ISO beeing written to the USB stick.

When I click Start, I expect to get an instant warning popup like:
[image]
and start the writing process and soon as I click OK.
I could very much do without the delay in "checking for conflicting processes" followed by the actual warning that my data on the USB will be destroyed. It serves no purpose.

So, pretty please, consider streamlining the process. Two clicks without the delayed popup would be extra nice.
One to start the operation and one to aknowledge the potential data loss.

Thank you.

pbatard, 2023-03-04:
I'm afraid I can't do that. The vast majority of users will be first time users, and the question of moving the Windows User Experience elsewhere has also been debated to death. I can't just add 7 extra checkboxes to the general interface, especially as I have already had many complaints, from people running on lower resolutions, that it is already occupying too much space.

So, I have no choice but to have an extra dialog when a user selects a Windows ISO and after they press START.

Please also understand that you are in the minority of people who say they have no use for the dialog. A lot of people, including many Windows 10 users, have been asking for the options to be presented to them.

This means that, if I have to displease some users with an extra click, and with a search for conflicting process, that's what's going to happen, because, regardless of your specific scenario, it does actually benefit the vast majority.

Remember that I do not design Rufus for *you* specifically. I do design it for everybody. And I base my design decisions on what reflects the feedback that I get from varied sources (rather than what single individual screams the loudest), as well as what my experience with users leads me to anticipate as the pros and cons of any solution.

If this is not agreeable to you, then I will remind you that Rufus is Open Source and easily modifiable (figuring out how to remove the calls to the WUE dialog or the search for conflicting process isn't exactly difficult, even if you don't know programming). So you can just install Visual Studio 2022 and start removing the stuff you don't like to tailor your own version to your specific liking.

&rarr; Rejected

closed 2023-03-04 by pbatard

Deepcuts, 2023-03-04:
The request is for an option, inside the settings window, not a default.
You seem a bit bitter for some reason with "(rather than what single individual screams the loudest)" and "Remember that I do not design Rufus for you specifically"
If a simple request sounds like a scream to you, no comment.

pbatard, 2023-03-04:
I don't see enough people benefiting from this option, especially in the UI. If I was to satisfy all the requests to tailor the UI towards each person's preference, I'd have about 20 extra checkboxes in the UI (I'm not kidding), which would be very confusing.

You have to understand that I simply cannot satisfy every single request I get on this issue tracker. It has to make sense for the vast majority of Rufus users and, and I'm sorry if this appears to aggravate you, from where I stand, yours doesn't.