pbatard/rufus #2197: Windows To Go - what is the default "Other user" username and password?

jddcef, 2023-03-06:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Installed Windows 10 22h2 as Windows To Go on a usb. It gets to the login screen but it says "Other user" and then wants me to enter a username, and to enter a password also. 
I didn't specify any of these. And I didn't check or enable the "local account" option.

Log
---
Log too large for issue text limit, here's the gist of it:
https://gist.github.com/jddcef/57902e7c6a4a401546069557352150e1

pbatard, 2023-03-06:
1. Unless I'm mistaken, `Windows 10 x64 22h2 en-uk Mar 2023.iso` is not an official Windows ISO as published by Microsoft. Can you please report where you obtained it? For obvious reasons (one of which being precisely that non-official images can create all kind of issues that official ISOs don't have) I can only support ISOs that have been downloaded from Microsoft's servers (such as the ones you can [download from Rufus](https://raw.githubusercontent.com/wiki/pbatard/rufus/images/download.gif)) or created from Microsoft official applications like the MCT utility.
2. `Notice: An extra 1949 files and directories were applied, from the 153427 expected` also seems to hint that there is something non-official with the image you are using.
3. Because _Windows To Go_ basically means applying whatever `.esd`/`.wim` is included in your ISO, and a `.esd`/`.wim` can be customized to force a specific user, or include customisation so that you don't have to go through OOBE (_Out Of the Box Experience_, which is what Microsoft names the process where you go through a fresh install where you create the user accounts, set the region, set the privacy options, etc.), my guess is that the ISO you are using contains a **custom** `.esd` that sets a default account, and you will have to see with its makers, or the publishers of the creation script (if you used a script) what it is. At any rate, when OOBE is active, Rufus will only ever create a local account **with an empty password** if you selected to do so in the Windows User Experience option (but that will only work if OOBE is actually executed, which, if using a custom ISO, is not guaranteed).

In short, unless you can report the same issue after downloading the official retail Windows 10 22H2 ISO, either from within Rufus or from https://www.microsoft.com/en-US/software-download/windows10ISO (NB: If accessing that last URL from Windows, you may have to change your browser's `User-Agent` string to be proposed an ISO download &mdash; At any rate, this will give you the exact same ISO as the one you can download from Rufus), I am going to dismiss it as something that has to do with using a custom unofficial ISO.

jddcef, 2023-03-06:
It's the official retail Windows 10 22H2 ISO, I just named it like that so I know which one it is. Gotten via MediaCreationTool22H2.exe  v10.0.19041.572, stock standard from microsoft.com

pbatard, 2023-03-06:
Please bear in mind that the a _retail_ ISO is one that is published by Microsoft on release date (i.e. November 2022 in this case) and that is never altered afterwards, and therefore one that bears the same SHA-1 for everyone.

The ISO created from MCT is what I consider an official ISO, but it is **not** a _retail_ ISO, so, while I try to generate an MCT ISO to see if I can replicate your issue, I would encourage you to download an actual retail ISO to see what happens.

jddcef, 2023-03-06:
OK I see.  

After I rebooted that windows to go usb, it booted up to a popup that says the setup wasn't good etc and then it just restarts. So it's damaged, so maybe something went wrong during the setup. 
So I will do it all over again to see what happens, but with that Win10_22H2_EnglishInternational_x64.iso rather.
and this time I won't click any of the funny customizations that are offered like the don't share data/privacy one, so that there is no skipping of any OOBE things.

Thanks for the information and care in your responses. 
I didn't know about the MediaCreationTool vs Retail thing!

pbatard, 2023-03-06:
Well, the MCT ISO should still work, and it has the advantage of including updates since the 22H2 release, but the problem compared to retail is that you can't check it for corruption, since it doesn't have a universal SHA-1 that can be validated (unlike what is the case for _retail_ 22H2, which, for `Win10_22H2_EnglishInternational_x64.iso` is `86976675A8900CE0952F779FC57468543218ACF2`).

I'm still in the process of testing the MCT ISO, but if I don't see the same issue as yours, I guess that it indeed seems likely that your media (be it the source ISO or the USB) may have gotten corrupted in some way.

pbatard, 2023-03-06:
Just tested Windows To Go mode from an MCT created ISO of Windows 10 22H2 (English International with Home edition selected. just like you used), and I was prompted to create the user account, which was then used to logon.

So I'm afraid that, since I am unable to replicate your issue under the same conditions of what you did, I'm going to have to dismiss it as an environmental issue. It's possible that, if you attempted to use an online account during OOBE and your computer was unable to connect to the internet properly after you did, the account retrieval failed, which may explain why you were prompted for Other User. But that's just a wild guess.

At any rate, since I don't see much I can do in Rufus, I will close this issue.

closed 2023-03-06 by pbatard