pbatard/rufus #2199: Error windows 11

Spanico1973, 2023-03-07:
Hi I was trying to create a pendrive with windows 11 using the costumize installation menu but when it reaches 99% of the creation it gives me an error with the message: impossible to correct / set the boot files, if instead I don't use that menu modify the pendrive me create normally!!! You can help me???

pbatard, 2023-03-07:
I can't help you without the log.

Spanico1973, 2023-03-07:
Where can I find the log because once it gives me the error I go ok and the program closes without writing anything to me.
Excuse my English but I use a translator to write to you!!!

pbatard, 2023-03-07:
This was explained to you when you opened the issue:

> - [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
> - [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Spanico1973, 2023-03-08:
Rufus x86 v3.21.1949
Windows version: Windows 10 Pro, 64-bit (Build 19045.2673)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0410 (it-IT)
Will use default UI locale 0x0410
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'it-IT'
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
Using 'autorun.inf' label for drive E: 'WinPE11Ktv'
1 device found
Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
Cylinders: 941, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {3C2D0A02-1265-4184-8317-3EE797EF35C7}
Max parts: 128, Start Offset: 17408, Usable = 7747363328 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Microsoft Basic Data'
  Detected File System: FAT32
  ID: {862011F3-752B-4F3A-8291-1421B4B25975}
  Size: 7.2 GB (7746332160 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Unable to send request: The connection with the server has been reset.
Scanning image...
ISO analysis:
  Image is a UDF image
Disk image analysis:
  Image does not have a Boot Marker
ISO label: 'ESD_ISO'
  Detected: Windows 11 ISO (Build 22621.525)
  Size: 4.2 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows 11 22h2.iso (4.2 GB)

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Opened \\.\PhysicalDrive3 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 7.2 GB)
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Writing protective message SBR
Found volume \\?\Volume{d3883a42-b352-11ed-8686-001999f58e06}\
Successfully remounted \\?\Volume{d3883a42-b352-11ed-8686-001999f58e06}\ as E:
Extracting files...
Image is a UDF image
Extracting: E:\TPM.reg (156 bytes)
Extracting: E:\WOW 21.exe (20.9 MB)
Extracting: E:\autorun.inf (128 bytes)
Extracting: E:\boot\bcd (16 KB)
Extracting: E:\boot\boot.sdi (3 MB)
Extracting: E:\boot\bootfix.bin (1 KB)
Extracting: E:\boot\bootsect.exe (105.3 KB)
Extracting: E:\boot\etfsboot.com (4 KB)
Extracting: E:\boot\fonts\chs_boot.ttf (3.5 MB)
[... 940 lines cut ...]
Renamed 'E:\sources\appraiserres.dll' → 'E:\sources\appraiserres.bak'
Created 'E:\sources\appraiserres.dll' placeholder
Mounting 'E:\sources\boot.wim[2]'...
Could not mount 'E:\sources\boot.wim [2]' on 'C:\Users\Spanico\AppData\Local\Temp\Ruf2E37.tmp': [0xC1420127] The specified image in the specified wim is already mounted for read and write access.
Finalizing, please wait...
E:autorun.inf already exists - keeping it

Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0951:1665)
1 device found
Disk type: Removable, Disk size: 7.7GB, Sector size: 512 bytes
Cylinders: 941, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {E0044CF2-8EBD-4FAE-9706-3F3FF10D0818}
Max parts: 128, Start Offset: 17408, Usable = 7747363328 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  Detected File System: FAT32
  ID: {F82B2893-F4B1-4AE6-95E7-C03E581417E7}
  Size: 7.2 GB (7746332160 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Checking for Rufus updates...
Checking release channel...
Unable to send request: The connection with the server has been terminated.
Updates: Unable to access version data

pbatard, 2023-03-08:
> `The specified image in the specified wim is already mounted for read and write access.`

Please reboot your computer (full reboot, not sleep/resume from sleep) and try again.

Spanico1973, 2023-03-08:
I've already restarted the computer a thousand times for three days that I've been doing tests and counter-tests, I've also tried different isos of windows 11 and also different pendrives but nothing always the error!!!

pbatard, 2023-03-08:
This is strange because a proper reboot should have cleaned up the mounted `boot.wim`. I suspect that you have fast-startup enabled, in which case restarting the computer does not actually reboot it.

In that case, please open an **elevated** command prompt and type the following command:
```
dism /get-MountedWiminfo
```
This should list `C:\Users\Spanico\AppData\Local\Temp\Ruf2E37.tmp` as mountpoint.

Then type:
```
dism /unmount-Wim  /MountDir:C:\Users\Spanico\AppData\Local\Temp\Ruf2E37.tmp /discard
```
If needed, replace `C:\Users\Spanico\AppData\Local\Temp\Ruf2E37.tmp` with whatever Mount Dir you got from `dism /get-MountedWiminfo`.

You may also want to try `dism /cleanup-wim`.

pbatard, 2023-03-08:
> the problem, as previously written, occurs to me by myself if I make changes with the customize windows installation menu because if I don't change anything, the pendrive is created without problems 100%!!!

Yes, that is because Rufus needs to mount `boot.wim` to apply changes if you make customization, and doesn't need to do so if you don't, and your issue is that, at once stage you tried to create a drive with customization, but interrupted Rufus before it got a chance to unmount the `boot.wim`.

Normally, and this has been the case with other people, rebooting the computer should take care of unmounting any existing `.wim`, but for some reason, this doesn't work for you so you have to unmount the `.wim` manually.

Spanico1973, 2023-03-08:
so how do i fix it if it can't modify this boot.win???

Spanico1973, 2023-03-08:
Great fixed with DISM /cleanup-wim command, now everything works!!! ^_^

pbatard, 2023-03-08:
Happy to hear that this fixed it. Considering that you're not the first person to run into this issue, and that I was debating adding an enhancement to close stale mounted `.wim` images (which, is **not** as easy to accomplish as it looks), I have now applied an improvement to the Rufus code that should attempt to unmount stale `.wim` images if it detects one.

closed 2023-03-08 by pbatard