pbatard/rufus #2200: Manually select an edition during windows installation

rainniel, 2023-03-08:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Some multi-edition windows installer automatically choose an edition during setup in some device, mostly in laptops, more details here, https://www.makeuseof.com/windows-11-select-edition-during-install/

It would be nice to have an option to manually select an edition on Windows User Experience dialog after clicking the start button so we won't need to do what's in the guide on the link.

Log
---
```
<FULL LOG>
```

pineapple63, 2023-03-08:
I have previously tried forcing the installer to use a specific edition, and all i will say is if the edition does not match the one the embedded product key is for, it will error out and not proceed

rainniel, 2023-03-08:
I used the Win11_22H2_English_x64v1 iso from the rufus download function, after enabling the edition selection, I'm able to install the pro edition smoothly on laptop where a home edition is originally installed, I always do custom installation on setup to erase everything and to partition the disk and use a local account and without connecting to internet, the setup never ask for key during installation, after the complete installation, I'm able to activate it with a pro key in the settings

pineapple63, 2023-03-08:
If you use a file to force the installer to display the edition selection screen, then it will work normally
From what I understand, you are asking for the edition selection to be handed by Rufus, which is where the error comes in (if the laptop has an embedded product key is for Home, but the file is configured to only allow Pro, then it will error out, or if a custom `install.wim` file is used which does not have Home in it, it will also error out)

Regardless of what you are asking for, I believe peter has indicated elsewhere that he has no plans to add a feature to allow you to install a different Windows edition on devices with an embedded product key

pbatard, 2023-03-08:
Yeah, creating a `ei.cfg` is not going to happen, because this creates compounded issues, and also because licensing troubles are out of scope of what the WUE is about.

Again, the WUE is there to address the annoyances that Microsoft shoves into the face of users during a standard Windows installation (such as multiplying privacy popups or requiring specific hardware or an online account). Picking up the existing license key and using that as default so that the user doesn't get prompted to pick their edition is actually something that I consider a welcome feature, because in the vast majority of cases this is exactly what people want and, unless you some people have actually managed to obtain a cheap MSDN subscription (otherwise, your license/edition hopping is unlikely to be done in a legal framework) I don't really see who this will **legally** actually benefit.

AFAIK, the way you change your licensing as a consumer, is, once you have installed Windows, you apply for a new license, and the edition gets updated through Windows Update as needed. Then when if reinstall Windows, your new license/edition will be picked. But you wouldn't (or at least Microsoft support wouldn't advise) scratching your working installation, and switch license/edition during reinstall.

&rarr; Rejected

pbatard, 2023-03-08:
Oh and please also see https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want on how _"Rufus is **NOT** designed as a Sysadmin/Power User-friendly application"_, because what you're asking for is what I consider a Sysadmin/Power User feature.

closed 2023-03-08 by pbatard