pbatard/rufus #2206: [Feature-Request] ReFS Support

xIUPITERx, 2023-03-19:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Feature-Request description
-----------------
Please add support for ReFS for Windows 11 (non enterprise) and the Windows-To-Go version.
>https://pureinfotech.com/install-windows-11-refs-partition/

pbatard, 2023-03-19:
This looks interesting, but it requires a restart, and unless one can enable ReFS support without restart, it's not something I see myself adding to Rufus.

I's also sure like to know more about where exactly the feature id `id:42189933` comes from (yes, I've seen the twitter post, but it doesn't tell me how exactly that specific ID was found).

All in all, the only feature I am really interested here in is re-enabling ReFS formatting on non Workstation/Server versions of Windows without requiring a reboot, because I believe that installing Windows on ReFS is a bit too niche to be worth supporting and people who really want it can use the link you pointed to enable it. However, since it doesn't look like ReFS support cannot be enabled without reboot, and I am lacking data on the potential drawbacks of enabling the A/B feature **experiment** mechanism of Windows (emphasis mine, because, if it's experimental, it means we're at the mercy of Microsoft to remove it from one day to the next), I don't think I want to add that feature.

Still, this is very interesting information...

xIUPITERx, 2023-03-19:
Right, it's probably better to wait for an official release anyway.

pbatard, 2023-03-20:
Okay, I'm going to close this issue and I may review ReFS support when/if it makes it into a formal release of Windows.

closed 2023-03-20 by pbatard