pbatard/rufus #2207: USB CORRUPTED

02tz, 2023-03-20:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
USB unplugged while halfway through formatting. I HAVE NO LOGS AND THE USB IS SHOWING no media - 0 bytes IN DISKPART AND DISK MANAGEMENT, THE USB IS Innostor IS903

Log
---
```
<FULL LOG>
```

pineapple63, 2023-03-20:
Best case scenario the partition table (or whatever it is called) is corrupt (which you would probably need specialised software to fix), worst case the USB is completely dead
I can’t say for certain what is wrong with it (apart from the fact that disconnecting it mid format probably corrupted something major), but I don’t think there will be an easy solution, if it is even possible to revive it

JonnyTech, 2023-03-20:
https://github.com/pbatard/rufus/wiki/FAQ#help-how-do-i-restore-my-usb-to-how-it-was-before-i-created-a-bootable-drive

pbatard, 2023-03-20:
Yes, I'm afraid there's not much that can be done for you as USB corruption issues have nothing to do with Rufus (Even if it had a major bug, [software like Rufus simply does not have the capability to make a USB drive fail](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive)).

Most likely your USB drive failed and you will need to replace it.

As such, I will close this issue.