pbatard/rufus #2209: Cannot unmount drive after creating Windows To Go using newer versions of Windows ISOs

Sunspark-007, 2023-03-23:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I am not sure if it is a bug in the OS or if Rufus should do something different based on how an image was created to make it identify differently.

The Windows To Go process works. The issue being reported here is that with older isos of Windows, e.g. Windows 10 1507, it will create the drive correctly, you can copy some files over from the desktop and unmount it from the icon on the taskbar. With newer image versions, e.g. 21H2, after creating it, you won't be allowed to unmount the drive. Attempting to do so puts up a message stating that the drive is in use and cannot be unmounted. Intermittent disk traffic is visible in the task manager. To do it safely, one has to shutdown or restart the machine. It is strange because one would think that the behaviour would be the same no matter what the version is, but it is not so. Newer will get attached and won't be released, and older will be released no problem despite the disk free space being the same and it is the same drive being used each time.

Log
---
```
<FULL LOG>
```

pineapple63, 2023-03-23:
I suspect its a bug, but if it only occurs with Windows to Go (and not on a standard install), then i would say don’t bother reporting it to Microsoft, as Windows to Go is not (officially) supported on newer versions (Microsoft would likely tell you that you are using an unsupported version of Windows)

pineapple63, 2023-03-23:
I suspect this is probably a Windows bug (athough I can’t be certain)
I seem to recall that I've had something similar happen to me before (although not with Windows to Go drives)

pbatard, 2023-03-24:
Yeah, that's pure Windows behaviour. Rufus does write the drive with exclusive access, but we do need to relinquish that access precisely so that the drive can be ejected (else it would be Rufus itself that would be preventing the drive from being ejected). And the problem is that, as soon as you do that, Windows low level processes are free to access the drive and perform whatever bullshit Microsoft added to their OS, to somehow lock drives that contain Windows content.

There's really not much I can do in Rufus (short of trying to force eject the drive as soon as it's done, but I don't think that's a good idea, and this would be much more trouble than it's worth), so you'll just have to yank the drive, which should be fine anyway, since Windows has really no business writing to that drive and should only perform read-only access (once it's created the `System volume information` directories, but even on a slow drive, it'll be completed before you get a chance to eject the drive.

As such, I will close the issue, since it's really a pure Microsoft problem.

closed 2023-03-24 by pbatard

Sunspark-007, 2023-04-05:
@pbatard I found a possible solution. In disk management, if you select the drive, right click and set the drive to offline, it can then be ejected.