pbatard/rufus #2211: 'Error: Failed to download file' when using Rufus 3.22 portable on Windows 7

mikbuch, 2023-03-31:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.

This question is not in FAQ

- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

I haven't found the issue among the opened issues here at Rufus' GitHub

- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.

I used the portable version, and I didn't know how to show the log. Now I know, I can provide the logs, if necessary.

- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

See above. The affected version: 3.22, portable

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

The issue is described here: https://askubuntu.com/questions/1092864/failed-to-download-file-error-when-trying-to-download-ubuntu-18-10-in-rufus-3

The solution for this issue is shown and described here: https://www.youtube.com/watch?v=m2W6DR3Ns3c

See also the following comment: "Safest solution: install the non-portable version of Rufus. This issue only happens on the portable version of Rufus."

Additionally, I was using Windows 7 Starter, and a downloaded lubuntu ISO image.

Log
---
```
see the explanation above
```

[image]

pbatard, 2023-03-31:
Sorry but I am no longer supporting Windows 7:

https://github.com/pbatard/rufus/wiki/FAQ#user-content-Windows_XP_Windows_Vista_Windows_7_NO_LONGER_SUPPORTED

closed 2023-03-31 by pbatard

mikbuch, 2023-03-31:
A very sad new, but I understand. Thank you for the feedback

pbatard, 2023-03-31:
Yeah. I edited to askubuntu post you linked to indicate that I no longer supported old Windows platforms.

mikbuch, 2023-03-31:
Thank you for editing the post!

If I could suggest something else, please consider changing the description on the official web page (https://rufus.ie/en/), because now it says:

> **System Requirements:**
> 
> Windows 7 or later, 32 or 64 bit doesn't matter.

So I guess that system requirements are now Windows 10, yes?

pbatard, 2023-03-31:
Not really. You _should_ still be able to run Rufus 3.22 on Windows 7. Just don't except all features to work and also don't expect me to support you.

The day I release a version of Rufus that is actually incompatible (which will happen with 3.23), I will edit the page.

mikbuch, 2023-04-02:
Ok, I see! Thank you for all the explanations.

I am a great fan of Rufus, and I use it from time to time for years now, keep the good work going!