pbatard/rufus #2212: Corrupt 2 usb disk while creating bootable usb

yadnesh2907, 2023-04-01:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<I do have two individual USB of HP which both failed to install windows bootable & then comes as disk is write protected now both usb is not able to format as those are detected as Write protected>

Log
---
```
[rufus-issue.txt](https://github.com/pbatard/rufus/files/11128666/rufus-issue.txt)

<FULL LOG>
```

yadnesh2907, 2023-04-01:
Please fix this issue as I already lost 2 pendrive

pbatard, 2023-04-01:
1. Please read [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive).
2. Most likely you have been sold _fake_ drives, which would explain why, while going over the "hidden" real capacity of your drives (could be 2 GB or 4 GB) you got a write error and your drives failed. Or you have other hardware issues. At any rate, I am 100% positive that your issue was not cause by Rufus and therefore that there is nothing that I can fix. Again, I strongly invite you to read the first link I gave you if you are willing to dispute this (and please be prepared to liaise with HP then, since, if you were sold genuine HP drives rather than fake ones, and you think you do have a reproducible case where Rufus will "destroy" a specific model of HP drives, HP flash drive engineers will probably be exceedingly interested to hear from you to help you demonstrate that, as opposed to what I am stating, Rufus is actually liable to damage drives).

closed 2023-04-01 by pbatard

yadnesh2907, 2023-04-06:
> 1. Please read [this](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Help_Rufus_damaged_my_flash_drive).
> 2. Most likely you have been sold _fake_ drives, which would explain why, while going [...]

Dear Sir,
both pendrives came as zenon installtion drive(so it not possible to be fake ) which I am using for more than year now , I never faced this issue untill now. I always use those pendrive to format windows in multiple system as I am technical head in my company. I think this issue is present in latest version update. I am not saying that you have to do anything for my lost pendrive , I simply ask you to fix this issue because I don't want to lose anymore pendrive. I am using Rufus from more than 2-3 years and till this issue it is reliable software for creating Bootable device for me.
One more thing if you check the log it started copying files in between it happen and in that state pendrive stuck so it is got write protected. I thought this issue is due to new update options like those skip privacy question, keep account. I never used those option beforhand but while preparing two time I keep all of them tick on. 
So kindly will you again check new version.
It will be much helpful if this problem won't occur furthermore

Thanks & Regards,
Yadnesh

pbatard, 2023-04-06:
> so it not possible to be fake

Unless you actually saw the components being assembled, it is always possible for something to be fake.

> I never faced this issue untill now. 

I cannot tell how much data you wrote to the drives prior and how, so I'm afraid that doesn't mean much. And things tend to work fine... until the day they break. Outside of fake drives, there's also the possibility that your power supply on USB ports became flaky, or you have grounding issues, or many other elements that could explain the failures you observed.

>  I am technical head in my company.

Good!

Then what I wrote in the link I gave above should __speak__ to you, and tell you that it is simply not possible for an application like Rufus to destroy a drive, period. As technical head, I will expect you to understand what a Hardware Abstraction Layer is, as well as how Rufus does not send direct commands onto the USB bus, and therefore, even if it were to have a bug, it could not send the only possible command that might make a drive inoperable, which is a USB vendor command.

> I think this issue is present in latest version update.

It can no more be present in the latest version as it can in previous version, which is, a software impossibility.

> I thought this issue is due to new update options like those skip privacy question, keep account.

I'm sorry but if this is what you think, then, despite being the technical head in your company, I have to point out that you clearly haven't read the link I gave you where I explained that there are **NO** special blocks on a USB device seen by Windows as far as software applications are concerned. So, it's a logical impossibility that, even if Rufus writes things slightly differently than before, it could cause a failure, because, as far as the drive is concerned, it's completely interchangeable **data-agnostic** disk blocks being written. All these blocks are treated the same by the USB drive, and even if they were to contain complete garbage or random data, it still wouldn't be able to induce a hardware failure.

I would **strongly** encourage you not to jump to conclusion or try to find a scapegoat to your issue _"I had a drive failure with the new version, _THEREFORE_ it must be the new version changes"_ whereas it has been explained to you, time and time again, how the Windows Operating System makes it impossible for even the most buggy application writing random data to a USB drive to actually render that drive inoperable. Again, __EVEN IF I WANTED TO__, I wouldn't be able to kill a USB drive in Rufus by simply writing data to the drive. As a matter of fact, there's really no difference whatsoever in the manner Rufus "talks" to a USB drive or a regular disk like an internal SATA HDD, a VHD, etc, and, because Windows presents and lets software access them exactly the same, we actually have to filter out those drives to only list USB ones, as otherwise, Rufus would be able to format and access any disk drive that the system reports. Which should _prove_ to you that there really is no special code going on when it comes to USB drives. Which should _prove_ to you that all the software can do is write __GENERIC__ data to a drive, which, again, does not have the possibility to cause destruction.

But again, if you want to insist that you have a case and that it's my software that somehow caused your drives to become inoperative, by all means, contact the manufacturer of your drive, because I hope you can appreciate that, if a manufacturer of flash drives is made aware of an application that has the alleged capability to "kill" their drives, they will want to help you take action against it, as, otherwise, they are likely to lose a large amount of money on RMA'ing drives, and they definitely will want to avoid that. So, if you are serious about claiming that you think that Rufus can damage USB drives, you __will__ contact your flash drive manufacturer (whose engineers will of course tell you the same thing as what I am telling you here, in that software like Rufus does not have the capability to destroy drives, but perhaps hearing this from more than one person will be enough to convince you of the veracity of what I am __factually__ reporting).