pbatard/rufus #2216: Feature Request: Unpack archive onto non-bootable device

codefaux, 2023-04-03:
Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, or used [DebugView](https://learn.microsoft.com/en-us/sysinternals/downloads/debugview), and copy/pasted the log into the section that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- They weren't

Issue description
-----------------
I've used Rufus many times to reformat USB/SD devices with various partition layouts from Linux images or game systems or etc. It's my go-to reformatting tool. Super handy that it also allows for making non bootable drives, I really appreciate that.

Often, I just need a blank USB drive with a file unpacked onto it, for eg Steam Link packages (Moonlight), SBC update images (Smoothie, Android, etc) or gaming console recovery/update images, sideloading of apps/packages, etc etc (the list goes on)

Suggestion: 
[image]
If I pick a supported archive using the `Select` button with `Non bootable` selected, Rufus currently complains.
[image]

Instead I believe one of two scenarios would be valuable features to a fair market share who already use Rufus;
First - most simply: I suggest Rufus would show the same message, _but additionally_ ask if the file should be unpacked onto the drive after formatting, yes or no. No further options necessary.

Second - A slightly more radical request would be the option to unpack the image/archive in non-bootable, or even just copy it wholesale onto the drive. Sometimes I actually want a blank drive with a .zip, .iso or .img file on it, for example. Some apps use phasing such as "Would you like Rufus to unpack the file (yes) or copy it (no) to the destination?" with a yes/no/abort dialog, and this would fit with the general schema in place.

Log
---
```Rufus x86 v3.22.2009 (Portable)
Windows version: Windows 10 Pro, 64-bit (Build 19045.2728)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.06
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Kingston DataTraveler G3 USB Device' (0930:6545)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Found USB 2.0 device 'Kingston DataTraveler G3 USB Device' (0930:6545)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Found USB 2.0 device 'Kingston DataTraveler G3 USB Device' (0930:6545)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  'C:\Users\Username\Downloads\Moonlight-SteamLink-4.3.1.zip' doesn't look like an ISO image
Disk image analysis:
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Failed to scan image
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Warning: Could not delete partition(s): [0x80070032] The request is not supported.
Opened \\.\PhysicalDrive2 for shared write access
Requesting logical volume handle...
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.3 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting to FAT32 (using IFS)
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Using Zeroed MBR
Found volume \\?\Volume{280ba281-d101-11ed-aacf-2cfda1582a9d}\
Created: E:\autorun.inf
Created: E:\autorun.ico
Successfully remounted \\?\Volume{280ba281-d101-11ed-aacf-2cfda1582a9d}\ as E:

Found USB 2.0 device 'Kingston DataTraveler G3 USB Device' (0930:6545)
Using 'autorun.inf' label for drive E: '7.8GB'
1 device found
Disk type: Removable, Disk size: 7.8GB, Sector size: 512 bytes
Cylinders: 948, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Detected File System: FAT32
  Size: 7.3 GB (7802126336 bytes)
  Start Sector: 2048, Boot: No
```

pbatard, 2023-04-03:
Thanks for the suggestion, but I'm not planning to add such a feature.

The reason for this is that it is very easy to accomplish what you want without using Rufus and there is therefore little point of adding such a feature. That's the same reason why, despite multiple request, I am not going to add an "eject" button to Rufus, because all it would accomplish is save a couple of clicks.

Furthermore, while Rufus does have compressed archive extraction code, this code is quite limited and not as performing as what you would find with a **dedicated** archive utility like 7-zip, so, unless I were to spend a lot of time improving this part of the code (which I am not seeing as a good investment, because for the most part, usage of compressed image is limited and the lack of top performance in that specific scenario is not a dealbreaker for its users), users of the feature you suggest would soon complain that Rufus is not as fast as dedicated unarchivers on archive uncompression and would probably state that, even if Rufus was able to do it, you are better off **not** using Rufus to uncompress large amount of data.

Then there's the problem of our current archive handler not currently supporting popular formats like `.7z` archives, and that our compression handling is mostly designed to deal with archives that contain a **single** file (usually a DD image) and not a whole set.

I will conclude by adding that, all in all, what you are really asking for is something that I would qualify as [_feature creep_](https://en.wikipedia.org/wiki/Feature_creep), in that, if you are not careful about keeping the design goals of an application focused, you soon end up with someone prompting you to make your application to send mail or connect with their toaster _"because it would be convenient"_.

In short, I would **strongly** invite you to just install 7-zip, which is an application that is designed with archive extraction in mind, and use that in conjunction with Rufus, instead of asking for Rufus to integrate the features of 7-zip. That's the much better way to do it on account that, since the developers of each application will be concentrating on a single feature from this workflow, you can rest assured that, overall, the performance will be better and likelihood or running into issue will be less than if you ask either Rufus or 7-zip to "bloat" itself with the features of the other application.

&rarr; Rejected.

You may also want to see [this relevant entry from the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#user-content-Why_cant_you_just_add_the_feature_I_want).

closed 2023-04-03 by pbatard

pbatard, 2023-04-03:
Oh, and just to elaborate on your first suggestion, which is to ask the user to extract the ISO, my issue with it (besides the fact that this can be accomplished with 7-zip in the same manner as if you worked with a compressed archive) is that I'd be certain to run into people using this to continue on what is essentially an error, and then expecting their drive to boot, and then asking why it doesn't.

So, I'd much rather have people force create a non-bootable drive with data outside of Rufus, as there can be non confusion that their drive will never boot. Because I'm afraid that I am well placed, from answering user requests, to tell you how much you'd be surprised as how users _misinterpret_ what an application is telling them and how, as long as you let them continue, they assume that whatever they __wish__ is what the application is going to accomplish.